Ralim/IronOS #1906: IKEA SJÖSS 30W charger does not work with Pinecil v2

positron96, 2024-05-01:
**Is your feature request related to a problem? Please describe.**

IKEA recently released a 30W USB PD charger for an attractive price, which I bought (and there is also 45W version). 
https://www.ikea.com/nl/en/p/sjoess-30w-1-port-usb-charger-fast-charging-50549412/

The charger works fine on its on (charges a smartphone fine), and Pinecil works fine with other USB-PD devices (a DELL laptop charger) and Xiaomi QC power bank. However, when I power Pinecil from Ikea charger, Pinecil goes into reboot loop. It works for couple seconds, you can even start heating with a button press, but then it resets. It resets even if I do not start to heat it. On average, it lives for around 3-5s before restarting.

Would be nice to troubleshoot it and if possible add support for this charger.

**Additional context**
Pinecil shows 16.1V on LCD when booted in idle (the charger can do 20V 1.5A and PPS)

I tried to go into PD debug menu of IronOS (by holding a "+" button on boot). It goes to "State 12 No VBus", and after that you can see what modes are supported (if you are fast enough):
* 5V 3A
* 9V 3A
* 12V 2.5A
* 15V 2A
* 20V 1.5A
* 5-11V 60w
* 5-11V 40w

Could it be that the reported modes are wrong and pinecil requests too much?

Ralim, 2024-05-02:
Hia,
On the product itself is it printed what its declared output's should be?

positron96, 2024-05-02:
Yes, it is.
[image]

Ralim, 2024-05-10:
Okay cool there is an issue to do with the PPS profiles. Unsure if its charger side or our PD stack side.

positron96, 2024-05-11:
Can I provide any assistance?

Ralim, 2024-05-14:
Specifically the difference is 

[image]

VS

[image]

We are decoding it as the latter due to bits 28/29 being 0b01. When I _think_ they should be 0b00.

Ralim, 2024-05-14:
I think the options are:

1 
Add a workaround that mitigates this if EPR hasn't been negotiated
2
If you have a USB-PD test of some kind and confirm what it decodes it as. If it also reproduces the issue it might one to raise with Ikea support potentially ( No idea)
3
We add a settings option to allow disabling variable voltage options as we are likely to see more of these in the future.

(3) and (1) are easiest.

Ralim, 2024-05-22:
Can you try flashing the firmware build from that pr, and testing when you have a chance?
you can download it from the bottom of https://github.com/Ralim/IronOS/actions/runs/9188593532?pr=1911

positron96, 2024-05-22:
Hi! I flashed the version you provided, and it worked perfectly right from the start.
If I was supposed to disable PPR/EPR from settings menu ("PD VPD0"), then it wasn't needed, it still enabled.
The PD debug menu now shows PPS modes in amperes, not watts. The relevant modes are:
6 5-11V 3.00A
7 5-16V 2.00A
(that matches the print on the charger)

The negotiated voltage is shown as 16.1V (during heating raises to 16.4) and maximum wattage I am seeing is 30.1.

So basically, it works perfectly now.

Ralim, 2024-05-22:
Fantastic to hear. This change shouldn't require you to do anything and should "just work".
I'll test this on a few devices here to sanity test but thank you for helping test 😀

Ralim, 2024-05-26:
This should now be resolved 🙇🏼 
I hope the Iron works well for you 😄

closed 2024-05-26 by Ralim