Ralim/IronOS #1908: Dynamic temperature setting (automatic boost)

KondratokGK, 2024-05-07:
**Is your feature request related to a problem? Please describe.**

When i solder small smd and hard-to-solder components one-by-one. I forsed to press boost button for a long time on heavy components (my thumb hurts) or change temperature every time by component type (it`s really annoing).

**Describe the solution you'd like**
Automatic boost feature: Increase temperature when power bigger than specified.
For example: 10W = 280C, 15W = 280C, 20W = 320C, 30W = 350C, 45W = 380C.
It need 2 user setting: power level and temperature increase per watt C/W or F/W.

**Describe alternatives you've considered**

**Additional context**

ia, 2024-05-09:
> Automatic boost feature: Increase temperature when power bigger than specified.
> For example: 10W = 280C, 15W = 280C, 20W = 320C, 30W = 350C, 45W = 380C.
> It need 2 user setting: power level and temperature increase per watt C/W or F/W.

I'm really sorry, but could you, please, rephrase your vision of the demanded feature so it could be way much more clear, in a step-by-step manner description (like, "_action - result_, _action - result_"). I'm just curious personally about what exactly feature you want to be implemented.

However, I may be wrong (and I hope I will be corrected) but the way how it works in _IronOS_ it's completely the other way round: _**first**_, you regulate temperature (i.e., how much _power_ aka _Watts_ tip needs in a form of _heat_) and only _**then**_ the firmware regulates drainage of _power_ from a _power source_. In short, you don't get _degrees_ based on _watts_ but _watts_ based on _degrees_. So if my humble understanding is correct then your feature _probably_ will demand a lot of coding, testing & debugging.

P.S. As always will be thankful for being corrected by those who is more familiar with the related parts of source code.

Ralim, 2024-05-09:
>When i solder small smd and hard-to-solder components one-by-one. I forsed to press boost button for a long time on heavy components (my thumb hurts) or change temperature every time by component type (it`s really annoing).

This seems to be flawed; what device are you using, what power supply are you using? You should generally never need to hold the device in boost, just set the temperature for the solder you are using and allow the iron to regulate power.

The way power works in these devices is based on how far the measured tip temperature measured is from the set point.

For example if the setpoint is 280 and measured is 250; its going to go at high power. Likewise if the set-point is 400 and the measured is 370 its also going to be high power. 
If the set point is 320 and the measured is 318 its going only add a very small amount of power.

What boost button does is temporarily raise the set-point temperature for when you are heatsinking into a ground-plane with poor thermal contact; while you are heating the solder to then get good thermal contact.

Boost mode does NOT directly tell the iron "increase power output". It only raises the temperature.

This all sounds like you might be using a wrong device+tip+power supply combination for the work you are doing, and you are running into poor thermal transfer to your target (and thus overheating the tip).

I could be wrong here, but I don't see how this would even be able to work?

All that I can see here is that the temperature set-point is going to swing around wildly and be very, very hard to use. Keeping in mind in normal operation the device is going to swing from 0 -> 100% on and back again all the time as the temperature control code does its best to maintain the tip temperature. So we are going to cross a lot of the wattage thresholds whenever it heats up from off/sleep. Also rapidly changing the set-point temperature will cause it to be relatively unstable.

Could you please outline more how this suggestion would work so that we can see if there is a way to implement this?