Ralim/IronOS #1909: miniware MHP30 only showing yellow light and voltage

Piwio, 2024-05-15:
Hi,
I got a MHP30 a few days ago and thought I could try Iron OS.
Flashing was easy on my Macbook, no problem.

I use the following charger and cable combination:

https://de.ugreen.com/collections/ladegerat/products/ugreen-nexode-usb-c-ladegerat-65w-usb-c-netzteil-3-port-gan-pd-charger-pps-kompatibel-mit-macbook-pro-air-iphone-14-pro-max-14-iphone-13-ipad-pro-galaxy-s22-s21-steam-deck-schwarz

https://www.anker.com/eu-de/products/a8552?ref=select_your_product&variant=42667662278819

Before I flashed Iron OS I had zero issues and the hotplate functioned just normal.
After flashing Iron OS I only get a yellow light and some voltage numbers (20v, 18v or 5v) when I connect the charger and cable.
I can access the settings, not problem.
The MHP30 has HW version 1.6 and the original FW was 2.11.

Anything I can do?

Ralim, 2024-05-25:
Hello, please try flashing a `dev` build from the actons tab. This should be fixed but not yet in a release.

MarcusTseng, 2024-06-02:
> Hello, please try flashing a `dev` build from the actions tab. This should be fixed but not yet in a release.
I also bump into the same problem. I'll try the dev build. Thank you.