Ralim/IronOS #1910: L245p usb soldering iron

AbdurRoufBusiness, 2024-05-17:
Recently bought a usb soldering iron that uses c245 style tips i was thinking if it is possible to install iron os on it providing some pictures the mcu has it’s makings removed so didn’t took a close up of it and seems like it has a uart port 
Thank you
[image]
[image]
[image]

ia, 2024-05-20:
> if it is possible to install iron os on it

The short answer is: _unfortunately, no_.

Please, look at [the very first table in this document](https://github.com/Ralim/IronOS/blob/dev/README.md) to check which devices are _supported currently_.

Thanks.

Ralim, 2024-05-20:
[Documented in the main documentation is the requirements for IronOS](https://ralim.github.io/IronOS/PortingToNewDevice/).

Realistically there are so many cheap cost-cut irons coming out of China I dont have the time to look into each one.

closed 2024-05-21 by ia