Ralim/IronOS #1915: Pinecil V2 has erroneous temp spikes in companion

vguttmann, 2024-05-24:
**Describe the bug**

When using IronOS companion, sometimes the tip temperature spikes up to 450C for one time point (see screenshot). I'm not sure whether this is an IronOS or a companion bug, therefore I'm opening an issue in both repos
[image]

**To Reproduce**

**Details of your device:**
 
 - Device: Pinecil V2
 - Release: V2.22E.C308FE8C
 - Power adapter being used: Lenovo 65W USB PD laptop charger
 - IronOS companion version: 1.0.0

**Additional context**

vguttmann, 2024-05-28:
Thanks to the help of the Companion devs, I was able to figure out that it's indeed an issue with IronOS. Interestingly, the spikes only appear when the iron is actively heating - not when it's in cooldown or idle.

xam100, 2024-05-31:
Hi I have encountered a similar issue with a similar setup
- lenovo 20v 65w usb c
- pinecil v2 running v2.22
- via usb bluetooth dongle
- monitored using https://joric.github.io/pinecil/

I wasn't able to reproduce it in the same manner but I did see it happen during normal usage.

I did manage to reproduce temperature spikes by playing (contacting) with the solder tip using tweezers, and that spike did not reflect on the pinecil display. 
[image]

I did also notice random temp spikes when the iron is just laying on the table cooling down. 
[image]

Ralim, 2024-06-02:
@xam100 Did you test with a build from #1916 as mentioned above?

xam100, 2024-06-02:
@Ralim just tried using [this build](https://github.com/Ralim/IronOS/actions/runs/9336146774/artifacts/1559818820) , and yes it seems to be reproduceable. 

Firstly a spike appears during the first power on, right after flashing, resetting settings and re-paring.
[image]

Afterwards, the issue comes back occasionally. Strangely, the spikes don't always reflect on the bluetooth web page but seem to always show on the display, i.e. the pinecil display can be seen briefly temp spiking to 300-310 but the graph doesn't always have time to show that spike
[image]

Ralim, 2024-06-04:
Hia,
If you check a chance you can test this firmware image?

[Pinecilv2_EN.zip](https://github.com/user-attachments/files/15551329/Pinecilv2_EN.zip)

xam100, 2024-06-04:
Hi, tested 1069e461 04-06-24, three good findings.

Random temp spikes did not show up this time.

Temp spike after pairing is still noticeable and is corelated with high (45W) sustained power output.
[image]

Very similar behavior is observed when re-heating the iron after an idle period (no re-pairing)
[image]

Those spikes max out at around 350C on the pinecil display
[image]

xam100, 2024-06-04:
Whoops, another weird behavior showed up right after posting.

When the iron is passively staying at the requested temp and I request an extra 20C (by doing +10 +10 on the pinecil), there will be a temp spike.
https://github.com/Ralim/IronOS/assets/33205747/a53bd17c-b947-4801-827e-07488b568509

Another instance where you see me go from 206 to 260 and the BLE will update to 209->240->260 and the temp will spike. In that scenario, the pinecil display showed something like 209->270->220->270->260
https://github.com/Ralim/IronOS/assets/33205747/bcd49afa-9603-4aaa-b3c9-c472861b2e79

discip, 2024-06-04:
@xam100
Sorry, I did not mean to delete your post. 😯

Unfortunately on my device it showed up as if this post was entered twice, so I removed the latter. But for some reason both posts vanished. 

I’m so sorry. Please forgive me.

Hope you will find the references you used again.

Ralim, 2024-06-04:
Okay this is good. These temperature spikes are likely something else (PID) can you confirm what power source and what tip your using so I can replicate these ?

xam100, 2024-06-04:
pinecil B2 86mm Tip
lenovo 65w 20v charger but sometimes maxes out at 45w 2.25A 
[image]

The post accidentally deleted by @discip was showcasing another temp spike issue resulting from a sudden request of power
here are two examples when it happened

https://github.com/Ralim/IronOS/assets/33205747/f1ba5fdf-bc2d-4a6d-8a71-226f71386881

https://github.com/Ralim/IronOS/assets/33205747/7ad8637d-ed79-4f36-a3a5-f7bed7073094

Ralim, 2024-06-04:
>pinecil B2 86mm Tip
lenovo 65w 20v charger but sometimes only 45w is used

Aha, okay I'll have a look at tuning. The wattages displayed are only estimated so take them with a grain of salt.

The system is tuned to overshoot slightly on the readback when you do a large temp change like that but it shouldn't overshoot that much. (Some overshoot is desireable to help bring the outside tip temp up faster since the temp sensor is inside the tip and leads the outside).

Ralim, 2024-06-08:
Just to test; could you try the build from here and compare for you: https://github.com/Ralim/IronOS/actions/runs/9426474334?pr=1925

discip, 2024-07-24:
@vguttmann
@xam100
Is this still an issue after the latest update [#1925](https://github.com/Ralim/IronOS/actions/runs/10076411126)?
If not, please consider closing this issue. 😊

thanks in advance