Ralim/IronOS #1930: Support request: G3061 Hot-plate support

darknessblade, 2024-06-10:
**Support request**: G3061

**Description:**
The G3061 is a small hotplate with a size of 50x50x35mm with a what looks to be a standard 0.91"128X32 Oled Lcd, that cost no more than 10-20 euro on aliexpress [depending on sale price/coupons] [Trough the "bundle deal" you can get it for roughly 12-14 euro]

**Example link:** [non affiliate]
https://aliexpress.com/item/1005007009784987.html?

**Aliexpress Item ID**
1005007009784987

**Images**
[image]
[image]
[image]
[image]

**Chip/IC codes:**
The main controller chip looks to be a CMS8S5897 234901T
https://www.mcu.com.cn/en/Products/258

Spec-wise it may not be as powerful as a STM32. so Many features might not be usable due the limited RAM/ROM space of the small hotplate, As its just a 8bit microcontroller compared to the 32bit a pinecil uses.

the small chip is a: H002 A2329

**requested features:** in case features need to be cut to support the device.
-Automatic off [More or less the most important one, since based on what I can find it does not have it by default]
-Advanced display settings [that shows more info like on the Pinecil/TS100, compared to just heat, the current layout is a bit cluttered even though its a basic display]

Ralim, 2024-06-10:
Hello,
At this time I don't believe supporting an 8051 core is going to happen anytime soon.

I would happily accept a PR, but I can't maintain it. I haven't worked on 8051's in a very long time, and while I'm not against supporting them; a working freertos stack is currently a requirement of IronOS.

Please see [the porting docs](https://ralim.github.io/IronOS/PortingToNewDevice/) for more information.

Additionally, this hotplate has already been discussed  in the discussions https://github.com/Ralim/IronOS/discussions/1698

Thank you for the information dense ticket though, highly appreciated

darknessblade, 2024-06-11:
> Hello, At this time I don't believe supporting an 8051 core is going to happen anytime soon.
> 
> I would happily accept a PR, but I can't maintain it. I haven't worked on 8051's in a very long time, and while I'm not against supporting them; a working freertos stack is currently a requirement of IronOS.
> 
> Please see [the porting docs](https://ralim.github.io/IronOS/PortingToNewDevice/) for more information.
> 
> Additionally, this hotplate has already been discussed in the discussions #1698
> 
> Thank you for the information dense ticket though, highly appreciated
> 
> For IronOS support I would suggest the secure T55 as it uses a chipset we can support. Or if other devices come into the market that use modern chipsets adding support should be possible.

I totally understand that Reliance is more important than support.

I didn't notice that there was already a ticket, mostly because searching "IN-ticket" for names isn't possible. and I couldn't find it based on the title alone.

For the hotplate, I will check out the "Secure/sequre T55". Maybe I'll buy it to check it out.

The G3061 is still a fun small hotplate to use for small basic jobs, and I got for like 7 euro [after all discounts], so it wasn't the biggest expense for a small hotplate.

closed 2024-07-05 by Ralim

RadioactivePotato, 2025-09-03:
https://oshwlab.com/oshwhub.com/pd-protocol-high-definition-mini
Looks like this entire thing is open source