Ralim/IronOS #1931: Sequre S60P USB-PD selection

kabel42, 2024-06-11:
**Describe the bug**

Software does not select higher Voltage after PD negotiation

**To Reproduce**
Plug Iron in USB-PD Supply
-> PD-Debug list Voltages but Iron only selects 5V

**Expected behavior**
Iron selects Voltage that allows for max Power

**Details of your device:**
 - Device: Sequre S60P
 - Release: v2.22D.442DBD98
 - Power adapter being used: Lenovo 65W

**Additional context**
The check in https://github.com/Ralim/IronOS/blob/dev/source/Core/Drivers/FS2711.cpp#L180 can never be true for this combination.
tip_resistance is fixed at 2 Ohm
PD-Debug reports:
5V 3A -> 1 Ohm -> Ok
9V 3A -> 3 Ohm > 2 Ohm -> not used
15V 3A -> 5 Ohm > 2 Ohm -> not used
20V 3A -> 6 Ohm > 2 Ohm -> not used

kabel42, 2024-06-12:
If i remove the check `if (min_resistance_omhsx10 <= tip_resistance)` the device negotiates 20 V and heats up faster.

jonasius, 2024-06-24:
I fixed this in PR https://github.com/Ralim/IronOS/issues/1920

kabel42, 2024-06-24:
works for me

closed 2024-07-24 by discip