Ralim/IronOS #1933: Cannot see custom logo on TS101

johncohn, 2024-06-13:
**Describe the bug**

**To Reproduce**

**Details of your device:**
 
 - Device: [TS101]
 - Release: [eg 2.15.40087E6]
 - Power adapter being used: Apple USB-C charger
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

johncohnvt, 2024-08-11:
Any ideas on this?… I have successfully created logos using the python code in the iOS meta-repo… But when I load them, I only get .ERR

Any help would be much appreciated 
-jc

StevenSeifried, 2024-08-11:
> Any ideas on this?… I have successfully created logos using the python code in the iOS meta-repo… But when I load them, I only get .ERR
> 
> Any help would be much appreciated -jc

Are you uploading under Windows or Linux? If Linux, try Windows

johncohn, 2024-08-12:
Thanks Steven - I'm using Windows 11...  (though I haveve both Mac and Ubuntu as well)

Arbut-himself, 2024-12-11:
Can confirm same exact issue on V2.22E.C308FE8C.

Ralim, 2024-12-12:
Eugh, getting `.ERR` means the TS101 bootloader is being a pain.
What DFU version does it say when you put the device into bootloader?

Arbut-himself, 2024-12-12:
1.05 DFU version. Just to be clear it's flashing correctly and showing `.RDY` not `.ERR` after flashing. Not sure if related, but the default Miniware logo that shows up is not affected by boot logo duration setting and always shows up for <1s even if set to Off.

reversr, 2024-12-26:
> Eugh, getting `.ERR` means the TS101 bootloader is being a pain. What DFU version does it say when you put the device into bootloader?

I have the same issue. DFU is 1.05, I've downloaded the latest version of IronOS from Github. I tried different commands which lead to the same result - the firmware is being created and it works - everything except the image:
python img2logo.py -M TS101_RU.hex 1.bmp TS101_Merged.hex -m ts101
or
python img2logo.py 1.bmp i -m ts101 --merge TS101_RU.hex

I tried different languages, and seem to try almost everything I could only think of.
Every command creates 2 files with .hex extension and 2 files with .dfu extension. I tried copy each file to the TS101 while pressing A button before connecting the cable. The result is the same it remounts the drive with .RDY file. IronOS works perfectly - but no image. This is a present, and I would really appreciate if someone can help or point what am I doing wrong, please? Thank you in advance!

ia, 2024-12-28:
Here is the very important thing to understand - **first of all**, _technically_ this is **not _IronOS_ bug**: obviously, there is _some problem_ in built-in _dfu loader_ _made by Miniware_, flashed on a factory by _Miniware_ (or their subcontractors) for a soldering iron _produced by Miniware_ (or what's their another name for today). There is no any way we could even possibly fix it.

Second, I have older model myself - _TS80P_ - and I heard a lot of flashing problems there too, but **I did not have not a single one myself** (_crossing fingers_), whether it's gif or png. Why? Because when you flash, you should _do everything by the book_. Or at least, please, report **detailed report**. But some of these reports do not contain even the name of the operating system (I'm talking in general, not even about this particular issue). Which OS do you use? If it's *nix/mac-based, what is `mount` output when _dfu storage_ is mounted (yes, that's important to), how do you copy - drag'n'drop or `cp image.hex /mount/point`, I even noticed that in some comments here hex/dfu files are mentioned in plural...

Anyway, here is the rules which I follow and do not have problems with flashing logos:
- **DO NOT COPY MORE THAN ONE FILE**;
- **DO NOT COPY RANDOM FILE ONLY BECAUSE PREVIOUS DOESN'T WORK**;
- after you generated **files**, pick only one which you really need (let's use the names from the original report): `outsmurphysiron_L.png.hex` or `outsmurphysiron.png.hex`, let it be the latter;
- **RENAME IT TO HAVE THE SHORTEST NAME POSSIBLE WITH .HEX EXTENTION, ALL CAPS**: `cp  outsmurphysiron.png.hex  FW.HEX`
- if you're on `gnulinux`:
  - **disable any automount** - in gnome or mate it's only these two commands: `$ gsettings set org.gnome.desktop.media-handling automount false; gsettings set org.mate.media-handling automount false;`
  - plug-in _iron_ holding the button for _dfu mode_;
  - detect attached block device using `dmesg`;
  - mount the _dfu storage_ **USING `msdos` file system module, not _`vfat`_**: `$ sudo mkdir -p /mnt/dfu && sudo mount -t msdos /dev/sdXN /mnt/dfu`;
  - copy hex file to the directory: `$ cp -f FW.HEX /mnt/dfu/`
  - flush the cache, wait for a couple of seconds, unmount disk: `$ sudo sync && sleep 3 && sudo umount /mnt/dfu`
  - **DO NOT UNPLUG NOR TOUCH ANYTHING**, now _iron_ tries to flash itself & reboot, wait for ~5 seconds;
  - verify the flashing: find block device in dmesg again, mount it again, and check `/mnt/dfu` for `FW.RDY` file (if ok) or for `FW.ERR` (if not ok).
  
I flash in `gnulinux` environment, works for me. Flashed in `Windows 8` once, it went good either. Sorry, but we can't help with the detailed instruction for every OS or environment... not a sarcasm, I really wish to help but not sure how. I guess the main takeaways:
- **do not copy more than one file at a time**;
- **you need hex file, nothing else**;
- **rename file before uploading to two-three letters name + .HEX, ALL CAPS LOCK**;
- **try to use legacy msdos file system module in your system if possible**;
- **do not trust automount/autoplay mechanics - they mess with dfu FAT storage badly** (for example, `macOS` just loves to create `.DS_Store` meta file **on every single attached & discovered storage and volume**... where is the chance that it's not messing with _DFU_? Sorry, but we don't have so much time to debug all these 3rd party _madness_ nuances.

Oh, and of course, do not forget to turn the logo feature on in the settings. ;)

ia, 2024-12-28:
Dear @johncohn, @Arbut-himself, and @reversr. Please, try to follow the advises & recommendations from [the previous comment](https://github.com/Ralim/IronOS/issues/1933#issuecomment-2564140310), and read it very carefully.

If it will be helpful - good, if not - it's really hard and nearly impossible to tell what else you should try, sorry.

johncohnvt, 2024-12-28:
Ok thanks ! I will try this method 

-jc

resistancelion, 2025-01-25:
> * where is the chance that it's not messing with _DFU_? Sorry, but we don't have so much time to debug all these 3rd party _madness_ nuances

Android's one loves to mess to, while i was succesfull to flash TS101 from my phone, i cannot longer do it, and now it have folders : .fseventsd, .Trashes and a 4kb file ._.Trashes. & I can no longer flash or open it from phone for whatever reason.
Flashing under Windows 11 (x64, usb 3.2 controller; DFU 3.50) works no problem, same with TS100(DFU 3.45)

ia, 2025-01-25:
> Android's one loves to mess to
> now it have folders : .fseventsd, .Trashes and a 4kb file ._.Trashes

I'm not sure about all of them, and this is quite off-topic, but `.fseventsd` is leftover from _MacOS_ `FSEvent API` daemon (background service) `fseventsd`, so it's not from _Android_.