Ralim/IronOS #1936: Pinecil V2: randomly leaves PD3.1,  switches to 5 V

Nickduino, 2024-06-24:
**Describe the bug**
I have a Chinese somewhat generic power supply that looks like an "updated PinePower Desktop": it has two USB-A and three USB-C. The last USB-C port supports PD3.1 140 W

With the Pine64 silicone cable, my Pinecil negociates 20 V and works perfectly.

With the Miniware PD3.1 240 W silicone cable, my Pinecil negotiates 28 V and works perfectly...until it doesn't

**To Reproduce**
1. plug Pinecil V2 with a Miniware USB cable into the PD3.1 port of the aforementioned power supply;
2. enjoy a few minutes of 130 W soldering;
3. wonder why solder doesn't flow anymore;
4. notice the Pinecil screen now displays 5.1 V and is unable to keep temperature (it will try but won't succeed, ending up with a thermal runaway error);
5. pull the plug, insert it back and hello 28 V!

**Expected behavior**
The Pinecil should not switch back to 5 V

**Details of your device:**
 - Device: Pinecil V2
 - Release: V2.22 (not the RC3 or whatever, the production V2.22)
 - Power adapter being used: https://www.aliexpress.com/item/1005006990964723.html
 - If this is an accelerometer related issue, please include its model number here: I don't think so

**Additional context**
Sometimes it happens (the switch from 28 V to 5 V) when the Pinecil is resting (probably sleeping) on my desk, sometimes it happens when I have it in my hand. It doesn't seem to be because the power supply trips the output or overheat or whatever (I did tests with a block of copper, it didn't fail then, and it sometimes while with light use). I tried increasing the PD timeout setting and changing the power pulse pulse settings (higher wattage and more frequent pulses), even though they're meant for power banks rather than power supplies...to no avail.

What's infuriating is that it's a random issue.

When I enter PD debug mode by pressing the button near the tip while plugging the iron in, I get:
- with the Miniware (240 W) silicone cable in a 100 W port: State 12 no Vbus and the power mode it reports are the ones you'd expect
- with the Pine64 silicone cable in a 100 W port: State 12 no Vbus and the power mode it reports are the ones you'd expect
- with the Miniware (240 W) silicone cable in a 140 W port: State 12 no Vbus and the power mode it reports are the ones you'd expect
- with the Pine64 silicone cable in a 140 W port: State 12 (it doesn't say anything about Vbus) and the power mode it reports are the ones you'd expect. After you've displayed all power modes, it will show  State 12 no Vbus (like in the 3 other scenarios)

I have a Fnirsi FNB58 (with a PD listener function) but I'm not sure what would be helpful to you (if anything at all).

Ralim, 2024-06-24:
Hmmmmmmmmm 🤔

Yeah this is going to be fun.

If you can setup the USB tester to capture PD traffic, first step would be to try to get a capture over the time when it drops from EPR to 5V. 

This issue is likely a bug in PD, so this should catch what's going on. Once I have a trace of what went on I can try to figure out where things went off the tracks. This could be as simple as a missing heartbeat message or messy.

I haven't used that USB tester but reviews indicate it can capture the traffic so that would be ideal if you can.

Nickduino, 2024-06-24:
> I haven't used that USB tester but reviews indicate it can capture the traffic so that would be ideal if you can.

That's the sort of data I get:

[image]

Ralim, 2024-06-24:
Hia,

For normal 20V profiles it's set and forget. But for PPS and EPR you have to chatter ever so often to confirm you're still there.

Can possibly work with the screen dump, is there any way to record to a computer ? (So it can be shared easier)

Nickduino, 2024-06-24:
> Can possibly work with the screen dump, is there any way to record to a computer ? (So it can be shared easier)

I'll see what I can do tomorrow, I just wanted to confirm that was the data you were after. I hope the thing keeps listening and recording over time.

Any idea of what I should expect to see when it drops down to 5 V?