Ralim/IronOS #1939: Profile mode: target temperature never settles

grinningmosfet, 2024-07-02:
**Describe the bug**
In profile mode (for hotplates), the target temperature rises beyond the phase temperature and never stops to increase.

**To Reproduce**
1. Example scenario: set `Profile Phases` to `1`, `Preheat Temp` to `40`, `Preheat Speed` to `1`, `Phase 1 Temp` to `50`, `Phase 1 Duration` to `30`.
2. Go back to the start screen and start the profile mode by long pressing the `A` key.
3. See the target temperature rising beyond the phase temperature.
4. By increasing the `Preheat Speed` to `10`, the target temperature can rise to >500°C during the phase.

**Expected behavior**
The target temperature rises until it matches the phase temperature, then settles until the end of the phase.

**Details of your device:**
 
 - Device: Miniware MHP30
 - Release: `v2.22D.1B042FCF` ([latest dev build for MHP30](https://github.com/Ralim/IronOS/actions/runs/9439985072))
 - Power adapter being used: xtorm XVC2140 (140W USB-PD EPR and 65W USB-PD)
 - If this is an accelerometer related issue, please include its model number here: not applicable

**Additional context**
The issue resides in [SolderingProfile.cpp, lines 115-124](https://github.com/Ralim/IronOS/blob/baf2f26e59da8408ce49cf072441ea0ee69a41c6/source/Core/Threads/OperatingModes/SolderingProfile.cpp#L115), /* · */ comments are mine:

```
// determine current target temp
if (cxt->scratch_state.state6 < cxt->scratch_state.state5) { /* Last phase temp less than current phase temp */
  if (profileCurrentTargetTemp < cxt->scratch_state.state5) { /* current target temp less than current phase temp */
    profileCurrentTargetTemp = cxt->scratch_state.state6 + ((xTaskGetTickCount() - cxt->viewEnterTime) / phaseTicksPerDegree); /* increase current target temp with respect to time spent */
  }
} else {
  if (profileCurrentTargetTemp > cxt->scratch_state.state5) {
    profileCurrentTargetTemp = cxt->scratch_state.state6 - ((xTaskGetTickCount() - cxt->viewEnterTime) / phaseTicksPerDegree);
  }
}
```

From my understanding,

- `cxt->scratch_state.state6` is the last phase's temperature
- `cxt->scratch_state.state6` is the current phase's temperature

`profileCurrentTargetTemp` is not saved as a `scratch_state`'s `stateX` and always fulfills the `profileCurrentTargetTemp` increase condition.

Another observation:

 - The phase termination condition requires the target temperature to be *equal* to the tip temperature. This rarely happens in practice, a range would be more appropriate.

```
if (phaseElapsedSeconds >= cxt->scratch_state.state2 && tipTemp == cxt->scratch_state.state5) {
```

I'm willing to help fix this issue.

ReimuHakurei, 2024-07-17:
Just submitted #1948

I don't own an MHP30, but I do own a Sequre T55. The T55 is not yet supported in the dev branch so I can't fully test the PR (since it is against the dev branch).

Care to test if the attached build from that PR fixes it on your MHP30?

[MHP30.zip](https://github.com/user-attachments/files/16260371/MHP30.zip)

grinningmosfet, 2024-07-17:
Hi @ReimuHakurei,

Just tested your build, it works fine on my MHP30 and fixes the profile mode. Thanks for the PR :-D

PR #1948 (2024-07-17): Fix profile mode
* **Please check if the PR fulfills these requirements**
- [ ] The changes have been tested locally
- [X] There are no breaking changes

* **What kind of change does this PR introduce?**
Fixes #1939 -- this appears to most likely be a regression from baf2f26e59da8408ce49cf072441ea0ee69a41c6
In addition to the bugs noted in that issue, I observed a couple of others that are also fixed here:
  - If two phases had the same target temperature, it would set the target to 0 degrees when hitting that phase.
  - The target temperature could underflow when cooling down.
  - The comment [... cut]