Ralim/IronOS #1945: Alientek T80/T80P

vulpes2, 2024-07-09:
Looks like Alientek released their own irons that use JBC cartridges as well, the pricing seems reasonable too. The T80 uses C210 cartridges, and the T80P uses C245 cartridges. I'm creating an issue to keep track of these two devices in case anyone is interested.

### Porting requirements

1. ✅ Supported processor: CH571F, RISC-V
2. ✅ 192KB of flash 
3. ✅ 16KB RAM
4. ✅ Device has one or more heating elements that can be controlled by a main temperature sensor 
5. ❔ I'm not sure if the board has a cold junction sensor or not
6. ✅ Means of the user updating the device without opening 
7. ❌ Known pinmap for the microcontroller

### Soft requirements

Ralim, 2024-08-23:
These should be doable for support; but to be honest I'm unlikely to prioritise them just due to time.
If someone would want to pick up the work for porting to these I'm happy to provide help.

gammeflamme, 2024-10-16:
[image]
[image]
[image]
[image]
https://github.com/user-attachments/assets/402540d0-a60c-483f-b0f4-71ba28cceb6a
https://github.com/user-attachments/assets/b15057de-8180-4012-ab60-3363d3df7512

Some scans from a T80P to help with schematic reverse engineering. Would love to have some open source firmware on this thing, I miss having live wattage readout.

hattesen, 2024-11-16:
I have just received a T60P and have ordered a T60, which seems to be virtually identical hardware.
I will prioritize to try to reverse engineer and create a schematic diagram (in KiCAD) and make that available in a GitHub repository along with a generated PDF version.
I can assist with testing new firmware, but I don't have the resources to help with the required IronOS changes.

Would the Alientek T60(P) development happen in a git branch of the IronOS repository, with pull requests, until officially supported?

vlad-mod, 2025-09-21:
After some looking around i found out that .atk file is just plane firmware with attached binary header of 32 bytes, Sniffing with wireshark while updating firmware shows that first 32 bytes of .atk file are not send to the device, while rest of file is sent without any modifications. It looks like first 16 bytes are name.
I managed to modify .data section of firmware and modified one of used sprites.
Provided update tool did not complain when uploading modified firmware.
[image]

vlad-mod, 2025-09-24:
I've managed to decompile upgrade app and found header definition:
```
|41 54 4B 2D 50 54 54 38 30 50 00 00 00 00 00 00|     3A   |      28     |  2D 01   |  64 00  | 61 3F |       4C 40 01 00       | E9 07 |   04   | 1D  |firmware...|
|                 name 16 bytes                 | data_len | encrypt_pos | app_addr | app_ver |  crc  | size of firmware 4bytes | year  | month  | day |firmware...|
```
I've also managed to compile custom firmware that just sets one of the pins to high, calculated correct crc and flashed it to the device. But it seems like firmware must be somehow encrypted, original firmware can be partially decompiled, but some parts are corrupted or dont make any sense, and flashed firmware does not in fact work.

For example, here is part of vector table from original firmware(decompiled with [radare2](https://github.com/radareorg/radare2))
```
0x00000020      6f7f2f35       jal t5, 0xf7372     
0x00000024      6f6f6f6f       jal t5, 0xf671a  
0x00000028      6f6f6f6f       jal t5, 0xf671e        
0x0000002c      006f           ld s0, 24(a4)                
0x0000002e      ef67007f       jal a5, fcn.0000681e    
0x00000032      6f796f6f       jal s2, 0xf7728        
0x00000036      6f6f6f6f       jal t5, 0xf672c     
0x0000003a      6f6f6f6f       jal t5, 0xf6730          
0x0000003e      6f6f6f6f       jal t5, 0xf6734         
0x00000042      6f6f6f6f       jal t5, 0xf6738          
0x00000046      6f6f006f       jal t5, fcn.00006736      
```
first entry must be jump to the reset handler, but instead it jumps to some strange address outside of flash ( entire flash size, including built in bootloader, is 0x7FFFF)

One of the solutions is to rewrite the entire firmware with a custom bootloader and IronOS, but this requires use of [wchisp](https://github.com/ch32-rs/wchisp) and shorting one of internal pins to the ground, which isnt that bad, and seems to be only solution. Moreover i accidentally cleared the entire flash while playing with wchisp, so for me this is the only solution.

dozed-dev, 2025-11-30:
Hi, I managed to reverse-engineer the crypto algorithm used to encrypt OEM firmware. Also, I found a way to pack arbitrary code to .atk and flash it using Alientek's bootloader.

I wrote a simple .atk converter/decryptor for this, here's my repo: https://github.com/dozed-dev/atk-fw-util

The process of flashing the resulting .atk firmware is the same as flashing OEM firmware - it doesn't erase Alientek's bootloader and it's completely reversible!

I also have a dump of Alientek's bootloader from my T80P. If you accidentally erased your flash completely, you can flash my dump via WCH-Link, restore the bootloader, and flash OEM firmware a usual.

This also makes it possible to enable debugging and debug the original firmware with openocd and gdb!

silicagel777, 2025-12-30:
> Hello mate and happy holidays!!

Hi, and happy holidays for you as well!

Here a few better photos:

[image]
[image]

Yanitwei, 2025-12-30:
>
> Maybe [RY8310](https://datasheet.lcsc.com/lcsc/1912111437_RYCHIP-Semiconductor-Inc-RY8310_C370876.pdf)? At least the marking seems to match:
> 
How did you find it so fast lol ? I am spending couple of hours now with no luck whatsoever.
Nevertheless, I am very thankful to you, you made my day! 

I'll try to do a continuity test and check pin 2 with ground, that should be a hint I guess.

[image]

0BlueNachid, 2026-01-16:
I used the original updater and bricked my device. It will just start in update mode. Is someone able to give me the original firmware dump of the T80 ? Would have flashed this alternative firmware but it doesnt support the model for now :(

xlanor, 2026-11-07:
> [@hsy70862026-png](https://github.com/hsy70862026-png) Sorry to hear that! Depends on what you were trying to recover; your procedure should have been correct.
> 
> I gave it my best shot dumping the whole flash; I don't think we have any success stories of restoring the original bootloader yet (although I believe it's possible with the right dump). Restoring the main "app" with a working bootloader definitely works. Did you overwrite the original bootloader?
> 
> Happy to try and help to restore the iron:) But I think the different display controller (SSD1312 vs Alientek's SSD1306) means you need a different intialisation/firmware. Would you mind starting a new "Discussion" and we walk through it there?
> 
> I did port IronOS to the T80p but had issues with display refresh rates; I might look at it again this weekend. That port can definitely be adapted to your iron.

Do you plan to release the t80p port? Just picked up a T80p this weekend and the stock firmware is... not doing it for me.