Ralim/IronOS #1950: Sequre S99 Support (C245 tips)

MmaxX09, 2024-07-23:
After the recent Sequre S60 support I think supporting the S99 would only be logical

It seems like the better iron overall for most people with the bigger, often more accessible C245 JBC tips compared to the C210 of the S60 and S60P which should make it more powerful and more applicable to solder bigger stuff like wires and connectors. With the small I als ILS style tips SMD down to 0402 should easily be possible as well.

Looking at some pictures I've found [online](https://www.reddit.com/r/soldering/comments/1ciozfn/jbc_style_usb_soldering_iron_roundup_teardown/) the PCBs and components used on both devices are nearly identical so implementing a custom firmware like on the S60P shouldnt be too difficult (altho I'm really not the one to judge here)

Definitely looks like the most promising portable Iron to be at this moment on the market, surely would be even better with IronOS ;)

Ralim, 2024-07-23:
Closing as its a duplicate of #1920

closed 2024-07-23 by Ralim

MmaxX09, 2024-07-23:
Sorry, I see now. Still getting the Hang of Github structures. 
Thanks for the quick reply and all your phantastic work!

Ralim, 2024-07-23:
No drama at all :) its a bit messy since you have Issues/Pull-Requests/Discussions.

TL;DR on them is:

1. Issue are "work to be done" so bugs or feature improvements
2. Pull-Requests are "code in progress"
3. Discussions are for anything that is not 1/2