Ralim/IronOS #1951: Change Date Format to YYYY-MM-DD

BAM5, 2024-07-23:
**Is your feature request related to a problem? Please describe.**
Viewing the debug menu fw date and saw 05-11-23 which being American that looks like May 11 2023, but after checking github I realized it was intended to represent Nov 5 2023

**Describe the solution you'd like**
Represent dates as 2023-11-5
I'm of the opinion it's pretty well understood when you start a date with a full 4 number year that the following digits are month, then day (going from most significate date unit to least)
This should be the least misunderstood date format globally.

Just a bit of polish for a global community without needing to do localization on formatted dates.

discip, 2024-07-23:
Sounds reasonable.

@Ralim
What do you think?

Ralim, 2024-07-23:
Yeah absolutely no issue by me; happy for either y/m/d or d/m/y format. 
It's set here https://github.com/Ralim/IronOS/blob/a855a68ca3f9df9373d25ed6c94dd7942e5295b2/Translations/make_translation.py#L155 if you have time to poke 😁

closed 2024-07-23 by discip

BAM5, 2024-07-23:
Heya @discip, looked at your code and I think you need to change it to %Y to output 4 digit year as per https://docs.python.org/3/library/datetime.html#strftime-and-strptime-format-codes

discip, 2024-07-23:
Yep, you are right! 😅
Will open another PR.

PR #1952 (2024-07-23): changed date format to YYYY-MM-DD
closes #1951
commit a4f070f4 (2024-07-23): changed date format to YYYY-MM-DD

closes #1951