Ralim/IronOS #1953: Language Pack in Original Firmware

wsedits01, 2024-07-23:
Is it possible to use the language pack in the original firmware instead of ironOs?

I would like to put the Portuguese language on my soldering iron...

wsedits01, 2024-07-23:
> Hello, I'm not sure what you mean here; we can't extract text from the vendor firmwares. You could contrive a translation for ironos if you are willing?

O ironOs vem com pacotes de linguagem na pasta,correto? Eu gostaria de saber se posso usar esses pacotes de linguagem com o firmware original,sem instalar o ironOS,apenas o pacote de linguagem

discip, 2024-07-23:
> I would like to know if I can use these language packages with the original firmware, without installing ironOS, only the language package

IronOS is a completely different firmware compared to the original one, hence the _'language pack'_ wouldn't be compatible with it, even if the firmware and the _'language pack'_ could be separated somehow.

discip, 2024-07-24:
Since this is neither an issue that could be solved here nor a feature request I'm closing this. 😊

closed 2024-07-24 by discip