Ralim/IronOS #1958: MHP30: Temperature stability

Ralim, 2024-08-01:
Sure, here's a revised version for better legibility:

I am having a similar problem as the OP.

I flashed the latest version of IronOS, and I consistently experienced overshoots in the temperature target. This issue improves when I limit the power, but that solution doesn't work for my application. I need to adjust the temperature in steps of ~10°C, ranging from approximately 35°C to 350°C, and I require a relatively stable temperature. Overshoots are unacceptable, and limiting the power prevents me from reaching high temperatures.

Does anyone have a possible solution for this? I also tried compiling IronOS myself. Although I managed to compile it, I'm not sure I did everything correctly. I experimented with changing the `TIP_THERMAL_MASS` value by increasing and decreasing it as @Ralim suggested, but it only worsened the problem.

Any help would be greatly appreciated.

Best regards,
Tomás

_Originally posted by @tomasrojasc in https://github.com/Ralim/IronOS/issues/1461#issuecomment-2262160133_

Ralim, 2024-08-01:
@tomasrojasc

I suspect the newer "tips" for these units have different thermal properties to the older ones.
If changing the thermal mass does not give enough calibration range, you can instead swap to using the PID controller and tuning P/I/D independently.

This is currently how the Pinecilv2 is tuned for example; you will need to add [this line](https://github.com/Ralim/IronOS/blob/2bce5b710e181a46bdcb49a7fa5d9023f9e0a81f/source/Core/BSP/Pinecilv2/configuration.h#L172) and the following tuning numbers to the configuration.h file. Then you can adjust the P/I/D response using the tuning numbers.

tomasrojasc, 2024-08-01:
Well, that was a super quick response. Thank you very much! I'll try that tomorrow since today is quite late already.

I'll keep you up to date.

Best regards,
Tomás

Ralim, 2024-08-01:
I don't have one of the newer tops so I can't really tune much, but if your willing to test tuning and try and get a working set of settings that would be an immense help.

tomasrojasc, 2024-08-01:
Hi, here is an update. I compiled the OS on my own but something is off... when I set the temperature, instead of having a solid red light, I have an alternating flash between red and green. I think my hex is not right. Can you help me compile correctly this? maybe a docker? I am committed to tune the PID controller but I need to be sure that my compilation works as expected.

Ralim, 2024-08-02:
If the PID is really unstable you could run into that. I would suggest starting with just a P value (make the others 0).

If you open a draft-PR the github actions system will build firmware for you as another methof.

tomasrojasc, 2024-08-02:
But I haven't done anything yet, I just compiled the software as is and the LEDs are not doing the same as in the published hex in the repository. My concern is that clearly there is something different hence idk if my running will work for everyone.

tomasrojasc, 2024-08-02:
Ohh ok I'll do it with GitHub, that is super useful, thanks. I'll let you know how it goes.

tomasrojasc, 2024-08-02:
Hi,

I believe I need approval for this to compile. I accidentally changed the draft to an actual PR (sorry, I'm not very familiar with PRs) and then created another one, which is a draft.

Do I need authorization every time I want GitHub to compile the project? Is there a way to do it independently, like creating my own private repository with GitHub workflows for compilation? This way, I won't bother anyone, especially since tuning a PID involves a lot of trial and error.

Thank you!

corndog2000, 2024-11-13:
@tomasrojasc You can get the docker container for local development so you can guild your own versions of the firmware.

[https://ralim.github.io/IronOS/Development/](https://ralim.github.io/IronOS/Development/)

tomasrojasc, 2024-11-13:
Thank you @corndog2000, althought we moved the issue [here](https://github.com/Ralim/IronOS/pull/1961#issuecomment-2453559066)