Ralim/IronOS #1959: Pinecil v2 won't power on with anything but 5v

greenwoodma, 2024-08-02:
**Describe the bug**
I've had a pinecil v2 for roughly a year although I've only used it a handful of times, all with the same USB-C charger without any issues. Today when I tried to use it, nothing happened when I plugged it in. I've tried a couple of different USB-C chargers all with the same effect; nothing. On a whim I also tried a standard 5v USB charger and with that it turns on, but of course I get a "DC LOW" warning and it won't heat up. I've looked online and can't find any reports for anything similar (usually it's either completely dead or it's not).

I also can't seem to get it to connect to my laptop for flashing; I'd not tried that before so don't know if it used to work or not.

I'm not sure if this is a hardware (and outside the scope for here) or a software issue, and from the debugging menu when running on 5v I can't see anything that jumps out at me as abnormal but I'm happy to admit I'm not an expert (I'm a software dev not an electronics engineer)

**To Reproduce**
Plugin in a known good USB-C power supply and nothing happens. Connect to a standard 5v USB charger and it boots but reports "DC LOW"

**Expected behavior**
Pinecil v2 should boot 

**Details of your device:**
 - Device: Pinecil v2
 - Release: v2.20.14AF70
 - Power adapter being used: have tried numerous and as I say it was previously working
 - If this is an accelerometer related issue, please include its model number here: N/A

**Additional context**
I don't currently have a suitable size barrel jack power adapter to test and see if it powers on from that. I have one on order which should arrive tomorrow so will update this issue once it arrives with details if it works or not.

greenwoodma, 2024-08-02:
Reading through a few other issues I've tried two more things with interesting results

- the 5v USB source I tried previously was using a USB-A to USB-C cable and giving the low voltage warning. Plugging the pinecil into my Samsung phone with a USB-C to USB-C cable still reports 5v although it does now heat the tip, although once it got to about 140C it switched to saying thermal runaway so I unplugged it.
- I then tried the USB-C socket on my laptop (where I'd previously tried it while holding down - to be able to flash the firmware) and it powered on but seemed to keep rebooting with the voltage switching between 3v and 5v.

Unfortunately I don't know what voltage my phone (a Samsung S24) should be able to provide so I've no idea if this is normal behaviour or if it points to a particular problem.

I didn't list a specific charger in the initial report given that none seemed to work, but if it helps this is one of the ones I've tried with no luck although it fast chargers my phone etc. with no problem: https://www.amazon.co.uk/dp/B0BXWLW8ZT?psc=1&ref=ppx_yo2ov_dt_b_product_details

greenwoodma, 2024-08-02:
Apologies for three comments in quick succession but..... it now works although I'm still confused.

Having seen it heat up when powered from my phone and the reboots when plugged into the laptop I thought I'd have one more try with the USB-C charger. I expected it not to work but instead it booted straight up, initially showing 5v but then quickly showed 15v and then 20v and it now seems to be functioning properly.

The only thing I can think is that it got confused somehow and either the reboots from the laptop or the 5v power from the phone has "reset" something.

There is still something weird going on though. I've tried it with three different USB-C chargers and it works with all of them but....... only when the cable is inserted in one orientation. If I flip the USB-C cable over I get nothing. Given that it happens with three different chargers using three different cables I'm guessing it must be something on the pinecil end that's the issue. I don't know enough about the USB-C spec to know how power is routed so that the plug can be inserted either way up, but this seems rather odd behaviour to me.

Again apologies for the rash of comments, and if you think this is likely to be hardware related rather than the software, but thanks for the great work on IronOS!

Ralim, 2024-08-02:
Hello,

> I'm not sure if this is a hardware (and outside the scope for here) or a software issue

Both are welcome here for the Pine64 devices since I'm basically responsible for both.

Okay so I think you have covered off a lot here.

1. You are not getting the USB-PD IC not being detected warning, so its likely the FUSB302 is fine (as confirmed later by it working in one direction)
2. The "only works in one orientation" is a very strong indicator that the issue is one of the CC pins in the usb port.

My guess at this point would be one of the CC pins in the USB-C port is not connected properly. 

The easiest to fix / most common here is sadly slightly poor soldering of the USB-C jack + movement, if you want to investigate further you can take the board out of the case and inspect these. 

If its not the connector it could be the FUSB302 chip itself but that is rare.

greenwoodma, 2024-08-02:
> Both are welcome here for the Pine64 devices since I'm basically responsible for both.

Ah cool, useful to know (I hate reporting problems to the wrong people as it just wastes everyone's time).

Given it now seems to be working I'm loathe to try and take it apart,as knowing my luck I'd make it worse, but if it fails again at least I have an idea of where to start.

One thing I did just try was the USB-A to USB-C cable that worked initially but gave the "DC LOW" error. The interesting thing is that I get the same behaviour there regardless of which way up the plug goes in. Again, I don't know enough about the USB-C socket to know if that's expected or if it would help narrow down which pin is likely to be the one with the poor joint.

Thanks again!

Ralim, 2024-08-03:
Yeah I 100% respect not touching something that works.
If you decide to dig in the future though I'm definitely happy to help; though in this case I think you figured it out before I got to my emails for the day 😂

So a USb A -> USB C cable is "dumb" and doesn't check the CC pins at all; it just provides 5V straight through. So it will always work even if the USB PD stuff is broken.

So sadly it doesn't narrow it down 😅

rkolbaskin, 2024-12-05:
> 1. You are not getting the USB-PD IC not being detected warning, so its likely the FUSB302 is fine (as confirmed later by it working in one direction)

I have similar issue with my v1 pinecil. PD IC was heating up when connected via USB A, but I am pretty sure I didn't get PD IC not detected warning until I desoldered it.