Ralim/IronOS #1981: Piezoelectric buzzer/beeper support

FleXoft, 2024-10-02:
**Describe the solution you'd like**

A support for a piezoelectric buzzer would be useful! Some sound/beep feedback when you press the button, reach the temperature would be very nice feature. 
**Don't you think?**

StevenSeifried, 2024-10-02:
For this, the devices must have a buzzer. afaik, no supported device has a buzzer.

The only device I know which has a buzzer is the Sequre T55 (which isn't supported by a stable release yet)

FleXoft, 2024-10-02:
Thnx.

StevenSeifried, 2024-10-05:
May @Ralim (and @discip) can tell more about which devices have a buzzer and which not and if it's possible in general (I guess it is).
In my opinion it's a good idea for devices, which have a buzzer, but it should be optional and selectable.
e.g.: no beep for button press but beep when the temperature was reached or vice versa

Ralim, 2024-10-06:
Buzzer is fitted in the MHP30, and the Sequre devices iirc.
On the MHP30 it is used to alert when at temperature. On the Sequre units its not currently used (mostly as I'm a person who cuts buzzers out of devices to stop them making noise, so I didn't yet have a use for it).

An option could be added for button feedback if thats something people want.

FleXoft, 2024-10-06:
Of course, I also think that this feature should be optional on those devices that have such hardware support, i.e. have a a buzzer device built-in.

Another option could be to look at how the existing devices (Pinecil v2) could be extended/upgraded with such a sound device.

Nickduino, 2024-11-18:
> mostly as I'm a person who cuts buzzers out of devices to stop them making noise, so I didn't yet have a use for it

+1
I never understand the buzzer on button presses: I know when I've pressed a button. And as far as buzzing when the right temp is reached: it's so quick that I don't need to know

FleXoft, 2024-11-18:
Thnks for your feedback with your opinion and I understand your views.

Nickduino, 2024-11-18:
> Thnks for your feedback with your opinion and I understand your views.

I'm glad you took it for what it is: only **my** opinion

FleXoft, 2024-11-19:
My idea for this improvement would be to somehow allow the introduction of some kind of beep capable option for the devices like the Hall Effect Sensor (HES) is for Pinecil v2.

I also wonder if there is a free I/O port at all that can be used for this on the HW? Has Pinecil v2 this possibility by itself?

[image]

https://www.piezoelements.com/piezo-buzzer/piezo-buzzer-external/10mm-small-size-piezo-buzzer.html

**Such beeps could be:**

- when temperature is reached
- when using burst mode  
- when the button is pressed
- when the button is pressed and hold for larger scale changes in value (+5C) could be a different tone
- when standby / sleep mode reached or switched back to normal mode

Of course all these options could be switched on or off from the menu.