Ralim/IronOS #1983: TS80P auto orientation is reversed

emlyn, 2024-10-04:
**Describe the bug**
When I set the display orientation to A, the screen is always upside down

**To Reproduce**
1. Go to User Interface settings
2. Select Display Orientation
3. Set it to A (auto)
4. Notice screen is upside down
5. Turn TS80P over
6. Screen flips so that it's upside down again

**Expected behavior**
Screen should be the right way up

**Details of your device:**
 - Device: TS80P
 - Release: 2.22.C308FE8C (05-11-23)
 - Power adapter being used: Supplied  PD-65W
 - ACC: LIS2 CLONE

Ralim, 2024-10-06:
Please try using the latest `dev` build. I believe this is already fixed
https://github.com/Ralim/IronOS/actions/runs/11185500339
(Download links for models at the bottom of the page)

emlyn, 2024-10-06:
Yes, that works. Thanks!

Gamerx007, 2024-12-07:
I just got a TS80P, and I have installed version 2.22

C308FE8C and the Automatic orientation is reversed as well 

do you have an alternate version of the IronOS for the TS80P to fix this issue?

emlyn, 2024-12-10:
In fact there hasn't been a release for over a year, any chance we could get a release with the latest fixes in?

gamelaster, 2024-12-11:
@emlyn hopefully new release should be soon. https://github.com/Ralim/IronOS/commit/5f96309f9c5ace2814d07a82c0e69427fc1d331d

ia, 2024-12-16:
Hello, @Gamerx007 

> I just got a TS80P, and I have installed version 2.22
> C308FE8C and the Automatic orientation is reversed as well

Well, just like stated in [the report above](https://github.com/Ralim/IronOS/issues/1983#issue-2567449866), the version you did try **is affected** by the issue, unfortunately.

> do you have an alternate version of the IronOS for the TS80P to fix this issue?

As the alternative, you can try _the latest successful build_ for your _TS80P_ from `dev` branch.

> I wanted to try the dev, but the link is expired

For exactly this purpose, now we added recently _dynamic_ links to the latest `dev` builds right into the main `README.md` [documentation file](https://github.com/Ralim/IronOS#Builds) for convenience. Just download and try _Development Build_ for _TS80P_, [here is the direct link from the table](https://nightly.link/Ralim/IronOS/workflows/push/dev/TS80P.zip) if you trust me. ;)

> Any help would be greatly appreciated
> Thank you 😎 👍

Hope this will help. And, please, if you will be having time to test the `dev` build, then come back with reporting the result. Thanks.

Gamerx007, 2024-12-16:
I was able to use my TS80P yesterday to do some soldering and it is from my usage working flawlessly

I currently do not have any way of testing if the temperature displayed is accurate ( but I was able to solder using 60/40 solder without a hitch at 280°C - 320 °C I was soldering a high power led that is on an  Aluminum substrate so I ended up needing a bit more heat 😉 ) display was in the proper orientation the whole time when set to auto 😎👍

ia, 2024-12-17:
> display was in the proper orientation the whole time when set to auto

@Gamerx007, thanks for the feedback!

closed 2024-12-17 by ia