Ralim/IronOS #1986: TS80P reading room temperature too high

nikolasgd, 2024-10-21:
My TS80P with IronOS (2.22 latest release as of today) reports a room temperature of around 170 degrees Celsius in idle mode.

For reproducing both handle and tips are at room temperature, around 21 degrees C. I have calibrated the iron and tried multiple tips (and power adapters just to make sure), all the same.

Values reported in debug menu are: RTip at ~ 1500 and CHan at ~ 170.

So from my understanding, CHan is the PCB temperature sensor reading in Celsius x 10? Seems reasonable in tolerance.

Reported temperature in idle main menu seems to be exactly the CHan value (without the decimal fraction part). So a CHan value of 170.1 will report tip temperature of 170 degrees celcius.

Since both Rtip and CHan values are in expected range, this must a software bug?

Thanks for looking into it. I really appreciate the work and dedication you put into this.

nikolasgd, 2024-10-21:
Just to confirm. This does not occur in the previous release IronOS V2.21. I have switched back to this version for now, as PD is also not working anymore in the latest release.

discip, 2024-10-29:
Have you tried calibrating the device according to this documentation?
Please follow the instructions carefully, especially those stating **_to wait_**. 😊

nikolasgd, 2024-10-29:
Definitely not a calibration issue. Temperature is reported correctly after switching back to IronOS 2.21

discip, 2024-10-29:
Oh, sorry, I forgot to put in the link to the documentation and missed the part in the initial post that **literally** states that you already calibrated the iron. 🤦🏻 😅

> Definitely not a calibration issue.

> I followed these instructions:

That's interesting.
I also own a **TS80P** and it works as expected with the latest firmware.

Maybe your device uses a **GD** chip instead of a **STM** one?

nikolasgd, 2024-10-29:
I will check tomorrow which chip is used. Can I find this info in the debug menu or only by looking at the pcb?

discip, 2024-11-01:
Unfortunately you would need to open the device.

nikolasgd, 2024-11-04:
[image]

So the chip on the upper side of the PCB is an STM32.

discip, 2024-11-04:
@nikolasgd
Have you tried cleaning the contacts of the jack and plug with **_isopropyl alcohol_**?

@Ralim
Any idea what could be causing this behavior?

nikolasgd, 2024-11-04:
> @nikolasgd Have you tried cleaning the contacts of the jack and plug with **_isopropyl alcohol_**?

I haven't tried that, as the problem can be reproduced with all of my 5 or so tips. And the temperature reads perfectly fine in IronOS 2.21.

Ralim, 2024-11-05:
Eugh what a mess.

I have a suspicion of the cause on this, we had some issues with analog mode on clone chips.
Possibly this is causing an issue on this legit STM32.

I'll try and make a test image for you when I get time (nag me in 24 hours if I dont post one)

nikolasgd, 2024-11-06:
Ready to test the image whenever you are! @Ralim

Ralim, 2024-11-07:
Apologies on the delay,
Turned out to not be what I was first suspect of (turns out that change was only done for MHP30)

But can try the same change.
Can you try from the TS80P files [from here](https://github.com/Ralim/IronOS/actions/runs/11720606218) (download at the bottom of the page)

nikolasgd, 2024-11-07:
Can confirm! Both tip temperature as well as PD do seem to be working alright in my quick test.

discip, 2024-11-07:
> Can you try from the TS80P files [from here](https://github.com/Ralim/IronOS/actions/runs/11720606218) (download at the bottom of the page)

This seems to have solved the issue as my iron still reads the correct temperature and OP states that his iron does the same. 😊

So, I hereby declare that the issue is solved.

PR #1990 (2024-11-07): Change Input mode of the tip sense signal for Miniware

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

* **What is the current behaviour?**

Closes #1986
* **What is the new behaviour (if this is a feature change)?**
Closes #1986

* **Other information**:
closed 2024-11-07 by discip