Ralim/IronOS #1988: TS80P only negotiating 5V on QC3

AmazingSpanoMan, 2024-11-02:
**Describe the bug**
Not sure if this is a bug but my new TS80P only negotiates 5V and around 4.6ish watts using QC3. Tried a number of wall plugs from no name brands to Anker with QC3. Advanced display shows it switching between PD and QC for some reason. It heats very slow and hitting around 300C displays "thermal runaway" and starts cooling. This is using a USB A to USB C cable. Tried multiple cables just in case. Only thing that works is using a USB C to USB C on an Anker plug but then it's using PD and gets a full 30 watts at 12v.

**To Reproduce**

 - Device: 
V2.22E.C308FE8C
05-11-23

TS80P miniware e-Design

**Additional context**

Ralim, 2025-02-22:
QC is a wild west and very hard to debug without a USB Meter that can show how the voltages on the USB data pins are reacting.

Please, please, just use PD it's a significantly better protocol if you at all can.

Being stuck at 5V often comes down to the charger wanting a slightly different setup to enter into the QC mode, or even different timing. You can disable PD by setting the timeout to 0 to force QC to be tried earlier if it's a power supply that needs you to negotiate it faster. If this doesn't work then without more information on what the charger is doing it's hard to debug 😢.

MonoAnji, 2025-02-22:
I get you, I'd use QC whenever I shouldn't have PD available, the TS80P is a bit of a travel mate of mine, so I noticed that it works fine with the original firmware, but not on IronOS, which even says PD even if actually QC is used, PD not being available. I might try to do some debugging myself, since I have a device to look into the negotiation. Thank you anyway for the information on how one might force QC if in deed necessary. :)

Ralim, 2025-12-16:
Hia  @DaverDavids  I cant view the video on imgur sorry.

So QC2 support is still in the IronOS code and last I tested it, it worked on all of my adaptors here. Are you able to post the video in your github post?