Ralim/IronOS #1992: Add support for Sequre HT140

Invictaz, 2024-11-09:
The Sequre HT140 soldering tweezers has an OLED screen and is one of the newer models. 
I hope it gets support on IronOS to improve reliability.

https://sequremall.com/products/sequre-ht140-2in1-hot-tweezers-soldering-iron-desoldering-repair-tool-for-smd?variant=44204748275900

Invictaz, 2024-11-22:
@Ralim I got mine today. If there is an option to dump the firmware over USB without opening it I'm interested to provide it. 

Mine came on firmware 1.03

Especially the lack of intuitive controls is quite a steep learning curve. There are short button presses. Long. Button hold and two buttons at the same time needed to control the iron. 

[image]

PetteriAimonen, 2024-12-05:
I opened mine up and dumped the firmware using Johannes Obermaier's [method D1-A](https://github.com/JohannesObermaier/f103-analysis/blob/master/d1-a/arm-cks32f/cksFirmwareExtractor.py).

Top of the PCB has testpads labeled `R`, `V`, `C`, `D`, `G`. These are reset, voltage, swclk, swdat, ground. I powered the device from USB 5V during the dump.

The default firmware has a USB bootloader that shows up as mass storage drive. To enter the bootloader, hold down front button when connecting USB cable.

Having support for this would be great, as the original firmware does not have support for heat conduction compensation. This would be quite useful with the thin tips.

* [sequre_ht140_pcb_bottom.jpg](https://github.com/user-attachments/assets/327a9e7c-3997-42e5-9179-fdcb4f9b0490)
* [sequre_ht140_pcb_top.jpg](https://github.com/user-attachments/assets/83fcb03a-2f98-4c45-b163-0d454d5ed7ac)
* [sequre_ht140_firmware_v103.zip](https://github.com/user-attachments/files/18020536/sequre_ht140_firmware_v103.zip)

Rudimentary firmware analysis:
* 0x0000 to 0x47FF: Bootloader (18 kB)
* 0x4800 to 0x13F8F: Main firmware (64 kB)
* 0x13F90 to 0x1FBFF: Empty flash
* 0x1FC00 to 0x1FFFF: User settings and calibration (1 kB)

Invictaz, 2024-12-05:
Thanks a lot. 

1. Will custom firmware burn a e-fuse and thus void warranty?
2. What is heat conduction compensation?
3. Can you downgrade or only upgrade the firmware?

PetteriAimonen, 2024-12-05:
1. Probably not. Though with just 90 day warranty, does it really make much of a difference?
2. Basically increase temperature at heater when there is a high power draw, as the tip temperature would drop otherwise. But I don't really know if it would work well for this device, it just feels that the tips cool off quite a lot when trying to heat larger contacts.
3. I don't think it checks.

Invictaz, 2024-12-05:
The 90 day warranty is quite strange as PayPal and Mastercard have a 180 day warranty on the Sequre website and EU consumer law is 2 years. Although this product isn't bought in the EU but imported. But I'm no lawyer. 

What for now is important is running RTOS on the iron. That paves the way for IronOS as per the wiki.

ThorstenWirth, 2024-12-08:
As I understand it - if you buy it directly from china - you would be responsible for the warranty as the importeur.
As a private person imports from china are usually considered doa.

Invictaz, 2024-12-09:
> As I understand it - if you buy it directly from china - you would be responsible for the warranty as the importeur. As a private person imports from china are usually considered doa.

But I didn't sign any import papers. A bill isn't a declaration of conformity to EU regulations. Sequre did when they handled customs on my behalf. 

Back on topic: some user on the EEVBlog forums requested heat conduction compensation as the 1.03 firmware is missing it.

Ralim, 2024-12-09:
Thank you @PetteriAimonen  for the images & firmware dump.

Given its a `CKS32f103CBT6`; the main OS features are already done.
Will require a modest amount of re-factoring to support two tips (aka two independent temperature controls). 

Main blockers at this point would be
1. Pinout and reverse engineering the PCB
2. A fair amount of code massaging to add two tips being controlled.

I dont see any blockers, other than the unit being fairly pricy 🤣

Invictaz, 2024-12-09:
Yes but as stated I didn't waive them from EU warranty with my purchase receipt.

Invictaz, 2025-04-20:
@PetteriAimonen Official firmware 1.04 was released yesterday. Can you dump and upload it?

PetteriAimonen, 2025-12-17:
I'm not working on this, and after getting some config settings in order, I find the original firmware quite satisfactory.

Apparently I also understood the circuit wrong and it is using the thermocouple after all.