Ralim/IronOS #2010: SI012 Pro Max Support

BenGitGud, 2024-12-03:
Hi there, 
Would it at all be possible to implement support for the Sequre SI012 Pro Max iron, I know there was an older issue posted but it hasnt been updated in a while. 

I am willing to try and email sequre about getting schematics and possible firmware. Or is there already a method of using the SI012 with IronOS, there was mention made of it in the Sequre S60 #1546 but I am not sure what came of it. 

Any help on how I can help make this possible would be appreciated!

msy021, 2025-02-13:
I would also really appreciate if SI012 Pro Max would be supported by IronOS! As @Gamerx007 stated, the original firmware is terrible in many different ways.

jemarro12, 2025-03-23:
I could help in testing or anything I could help... Any testing firmware which boots up??

Ralim, 2025-03-23:
We would need hardware details to see if its even feasible.
Then usually next steps is a schematic or someone reverse engineering how all the hardware connections are made. 

Then if both of those look feasible, someone needs to make the PR to add support.
IIRC none of these steps have happened yet.

jemarro12, 2025-03-24:
@Ralim I think what you need is all the wiring to the MCU, which pin is connected to which component?
I can do that reverse engineering over a Si012 Pro (pink PCB) v1.5.
MCU used is a CKS32F103C8T6, I think is doable, looks like a good replace to STM32F103
I'll share my progress in a week or so.

jemarro12, 2025-03-28:
I´ve made some progress: 
As said in a previous response Sequre Si012 Pro V1.5 (pink PCB) uses a CKS32F103C8T6.
This iron is compatbile with 2 different kind of tips T12 and TS/SI so a 2 channel P-Mosfet as well as a dobule opamp is used.
The PD chip used is HUSB238 (0x08 I2C addr). 0.69 inch  OLED is used in the same I2C bus.
For now this is the pinout i have, if you need anything else say to me.

[image]

Ralim, 2025-03-29:
Ah yes the `CKS32F103C8T6`, aka the STM32F103 with 10x the bugs (I'm only slightly salty).

But yeah this is basically the key thing, is that we need to know what pin goes to what function in hardware, and how the hardware works at a high level (what pin does tip pwm, if we have multiple analog sensors or mosfets etc how that is wired up).

Once we have these we can fork off the S60 firmware to set these up and start to test.

jemarro12, 2025-03-29:
Components are really small, I"m trying my best with my 10bucks usb microscope. I"m almost certain they use separate PWM signals, one for each MOSFET. PB8 gates first MOSFET while PB9 gates the second.
I'll try to reverse the opampm setup. Not really sure what are they using as reference for the cold junction.

I can't see any accelerometer or vibration switch, any idea how is this made on other irons?

jemarro12, 2025-04-07:
After exchanging several emails with Sequre, they are asking if both Ralim (the project owner) and I would be willing to sign an NDA in order to share the schematics of the Si012 Pro Max, so it can be included in the IronOS project.
@Ralim if you agree please let me know through which channel I can tell them to contact you to send you the document (if you're okay with it).

mail copy:

[image]

Ralim, 2025-04-08:
I'm happy to do an NDA, ralim@ralimtek.com.
If they send it to me I can review and check.

jemarro12, 2025-04-22:
@Ralim after several mails and corrections Sequre team finally signed the NDA, i´ve forwarded it to you for your approval. 
Last week i´ve tried to reverse on my own the opamp setup but with no microscope it has been too difficult to me.

Ralim, 2025-04-23:
@jemarro12  I haven't received the revised copy?

jemarro12, 2025-04-24:
@Ralim I forwarded it to you yesterday. You must have received an email from jemarro12@gmail.com yesterday at 14:18h. Attached is a PDF signed by sequre. If it has not been sent correctly let me know and I will try again

Ralim, 2025-04-26:
Sent you back my signed copy 🙇🏼

jemarro12, 2025-04-30:
After several mails back and forth with Sequre finally we have Si012 Pro schematics. 
@Ralim if you need a copy i can send to you.

CKS32F103C8T6 pinout:
PB9 - T13_PWM
PB8 - T12_PWM
PB7 - SDA
PB6 - SCL
PB3 - LED10
PA15 - LED9
PA14 - SWCLK
PA13 - SWDIO
PA12 - D+
PA11 - D-
PA8 - (gates D+ MOSFET)
PB15 - LED 6
PB14 - LED 5
PB13 - LED 4
PB12 - BUZZER
PB11 - LED8
PB10 - LED7
PB1 - B_switch (Pulled up)
PB0 - A_switch (Pulled up)
PA6 - T13_TEMP
PA5 - NTC
PA4 - Voltage reading (VBUS = PA4 reading * 0.1176)
PA3 - MOTION SENSOR (Pulled down)
PA2 - Ambient Temp sensor (TC1047)
PA0 - T12_TEMP
PC15 - LED3
PC14 - LED2
PC13 - LED1

This iron uses double tip support (T12 and TS/SI), this is the T12 and T13 temp.

jemarro12, 2025-04-30:
@Ralim if you see it convinient i can create new PR to provide this info

goldtyson, 2025-06-22:
Any news guys?

jemarro12, 2025-06-22:
A Pull Request is opened with the first build working.
I appreciate if you can test it and share feedback in the PR.

Ray-Alfy, 2025-10-26:
Is this compatible with SI012 Pro (not SI012 Pro **Max**)?