Ralim/IronOS #2014: Ts101 unable to load firmware

fcortes626, 2024-12-07:
**Describe the bug**
I am unable to upload the firmware on a ts101 with usbc. I keep getting the ERR files after it reboots. This is with the EN version of the file. I tried the ES versions and it worked fine. Tried both builds and 2 different cables and computers

StevenSeifried, 2024-12-07:
Under which operating system do you have the problem?

alexkingkz, 2024-12-07:
I have the same problem, only none of the languages ​​work for me :( I have Windows 11

fcortes626, 2024-12-07:
> Under which operating system do you have the problem?

It was on windows 10

StevenSeifried, 2024-12-07:
I can confirm the problem with Windows 11

alexkingkz, 2024-12-07:
We really hope and wait for corrections!

Mel0ne, 2024-12-17:
I confirm, I encountered the same problem. Neither stable nor dev are flashed.

Mel0ne, 2024-12-17:
UPD.
Windows 10 Pro 22H2 flashed OK.

cliobrando, 2024-12-18:
I can flash the TS101_ES firmware, but the EN firmware gives ERR every time.

Blong2023, 2025-01-04:
Same. I can flash the TS101_ES firmware, but the EN firmware always gives ERR.

Windows 11 24H2

cliobrando, 2025-01-17:
I think I found a solution (that worked for me at least):
I'm using Windows 11 24H2 btw.
- Launch an Ubuntu 24.04 Virtual Machine (I use VMWare Workstation)
- If you have USB 3. X ports, change the USB controller of the VM to USB 3.1
- Connect the TS101 in DFU Mode
- Copy the hex firmware (you can use any language)
- If it gives ERR just try again, for me, it always worked the second time.

I use the stable version (2.22), most dev versions over that puts the TS101 on reboot loop (even the 2.23 RC1).

Thanks.

ia, 2025-01-30:
If anyone still have this problem, **please**, could you read & try advises and recommendations from these two links:
- [general set of hints how to flash _"more properly"_](https://github.com/Ralim/IronOS/issues/1933#issuecomment-2564140310);
- [confirmation that **all files you have on DFU drive must be deleted before copying HEX firmware**](https://github.com/Ralim/IronOS/discussions/2061).

If nothing will help, there is not so much we can do. :(

paulfertser, 2025-01-30:
On Thu, Jan 30, 2025 at 03:19:30AM -0800, Ivan Zorin wrote:
> If anyone still have this problem, please, could you read & try advises and
> recommendations from these two links:
> 
>   • [1]general set of hints how to flash "more properly";
>   • [2]confirmation that all files you have on DFU drive must be deleted before
>     copying HEX firmware.
> 
> If nothing will help, there is not so much we can do. :(

One can always connect SWD (without a debug adapter it can be
bitbanged with a raspberrypi or other SBC) and flash a sane bootloader
(implementing real standard DFU class), right?

This constant wrestling with broken mass storage emulation tricks
feels like a rather unnatural and joyless way to spend time.

ia, 2025-01-30:
> One can always connect SWD (without a debug adapter it can be bitbanged with a raspberrypi or other SBC) and flash a sane bootloader (implementing real standard DFU class)

This is out of _IronOS_ scope. But since we're here - do you know any 3rd party DFU bootloader with _TS101_ support?

> right?

Not quite. Not when users just want to have an iron with _IronOS_ without a need in learning how to use hardware debugging protocols to flash their irons. _IronOS_ users may have very different backgrounds and interests, so getting a separate equipment and figuring out the wiring to connect to an iron to flash bootloader may be by far not ones of them.

> This constant wrestling with broken mass storage emulation tricks feels like a rather unnatural and joyless way to spend time.

Not when a user at least tries to follow the recommendations. :/

paulfertser, 2025-01-30:
On Thu, Jan 30, 2025 at 09:44:23AM -0800, Ivan Zorin wrote:
>   One can always connect SWD (without a debug adapter it can be bitbanged with a
>   raspberrypi or other SBC) and flash a sane bootloader (implementing real
>   standard DFU class)
> 
> This is out of IronOS scope. But since we're here - do you know any 3rd party
> DFU bootloader with TS101 support?

Judging by
https://github.com/Ralim/IronOS-dfu/commit/c157229e1e8fb465a8abebd13ad7c7489885f7c9
I would expect it to be nearly trivial to add support for TS101 as
well. Or am I missing some important detail here?

> right?
>
> Not quite. Not when users just want to have an iron with IronOS without a need
> in learning how to use hardware debugging protocols to flash their irons. IronOS
> users may have very different [...]

So probably what would be _in_ scope for the project is developing a
userspace libusb-based UMS+FAT implementation to reliably flash even
when the extremely buggy Miniware bootloader has to be used? But even
with that windows users will complain because nothing is easy there...

IDK, I think it's great when users get educated about things,
connecting 4 wires and running a pre-made script shouldn't be that
much of a burden, we're not talking about any exotic hardware or
actual required knowledge here.

>   This constant wrestling with broken mass storage emulation tricks feels like a
>   rather unnatural and joyless way to spend time.
> 
> Not when a user at least tries to follow the recommendations. :/

Over the time there were many conflicting recommendations (mount as
"msdos", mount as "vfat" copy with this command, copy with that
command, use Nautilus, use windows xp, use another port etc etc), none
were really working reliably and there's no decent way to troubleshoot
because it involves many layers of different kernels (different
versions too) and userspace and that proprietary bootloader without
any additional feedback.

I'd say the recommendation should be: if you try it and it works to
flash IronOS-dfu "Runtime", great, lucky you, proceed to flashing
everything proper. If it doesn't don't waste too much time, just
connect SWD.

ia, 2025-01-31:
> Judging by [Ralim/IronOS-dfu@c157229](https://github.com/Ralim/IronOS-dfu/commit/c157229e1e8fb465a8abebd13ad7c7489885f7c9) I would expect it to be nearly trivial to add support for TS101 as well. Or am I missing some important detail here?

Your link to the patch is about _S99_. [_TS101_ support is on the way for a long time](https://github.com/Ralim/IronOS-dfu/pull/16), but apparently as it turned out there are not so many _TS101_ owners who would like to help with testing & debugging, especially if something will go wrong.

> developing a userspace libusb-based UMS+FAT implementation to reliably flash even when the extremely buggy Miniware bootloader has to be used?

This is one of those cases when `effort < result`.

> IDK, I think it's great when users get educated about things, connecting 4 wires and running a pre-made script shouldn't be that much of a burden

Do you understand that your set of skills & interests is not a set of skills & interests of _a user_ who just want to use a soldering iron with _Iron OS_, right? The world doesn't begin nor end in IT world, while soldering can be used in very different ways & areas.

I say this calmly and politely, because this is observable fact related to any project: in fact, modern users don't even check bug trackers for already opened & existed similar bug reports anymore at all, while some of the reports are looks like nothing more than "feature X doesn't work, please, fix it!" (literally). And you're talking about using hardware debug interfaces. The era when kids could make their own radio or calculator just by reading local magazines & books about electronics & hardware are over. For the record: I'm not complaining, just sharing observation from my perspective.

> Over the time there were many conflicting recommendations [...]

I just provided what works for me & for other users (like that great discovery about deleting all the existed files first).

Anyway, it starts to look more like a general discussion on usenet :)
Ironically, I understand that I'm part of the problem. So I'm posting this so you could read my reply, but in a little bit I will have to mark the replies from both of us as hidden here, since they do not have any relation to this #2014 issue at all. Thank you for understanding.

abjugard, 2025-08-31:
Same happened to me from Windows 11 24H2 (build 26100.4946), worked from latest macOS.

Riebart, 2025-09-22:
Can confirm, I can't flash anything on my TS101 from Windows 1124H2, on two different computers. Or a Raspbery Pi 3b+ running Debian 12.

But I could flash just fine from WIndows 10.

This is insane. 😓

MisterFixx, 2026-10-30:
I also can't flash my TS-101 with the recent 2.24 RC update, i'm on windows 11 25H2 and the TS-101 says DFU 1.05, every attempt ends with .ERR.

cliobrando, 2026-10-30:
I recently used an  Ubuntu VM to replace the bootloader with ironos-dfu, it worked perfectly.
I can flash it every it using dfu-utils without problem.
TS-101 with 1.05 to ironos-dfu

sorz, 2026-10-30:
https://github.com/Ralim/IronOS/blob/dev/scripts/flash_ts10X_linux.sh

This script works for me. Basically it do `mount -t msdos` then `dd oflag=direct` to write, umount, then repeat.