Ralim/IronOS #2015: TS80P Automatic orientation is upside down

Gamerx007, 2024-12-08:
Describe the bug

When I set the display orientation to A, the screen is always upside down

To Reproduce

Go to User Interface settings
Select Display Orientation

Set it to A (auto)
Notice screen is upside down

Turn TS80P over
Screen flips so that it's upside down again

Expected behavior
Screen should be the right way up

Details of your device:

Device: TS80P
Release: 2.22E.C308FE8C (05-11-23)

Power adapter being used: USB C cable plugged into 12v usb power supply 
Bought from ToolSage on Amazon on December 3rd 2024

Any help would be greatly appreciated 

Thank you 😎 👍

Ralim, 2024-12-09:
Can you please check the ci build of the latest Dev

Gamerx007, 2024-12-09:
where is that located?

I also have a weird issue with my ts101 display brightness when at 1 the screen appears to be off only at brightness level 2 dose the display show anything all the way 2 - 4 ( is this a bug for the TS101 version 2.22 ? or have you already heard about it? )

discip, 2024-12-09:
So this is solved I assume. 😊

closed 2024-12-09 by discip