Ralim/IronOS #2016: TS101 Screen Brightness Control

Gamerx007, 2024-12-09:
**Describe the bug**
 a weird issue with my ts101 display brightness when at 1 the screen appears to be off only at brightness level 2 then the screen lights up 

Brightness setting 2 - 5  work with no issues 

**To Reproduce**

 - Device: [e.g. TS101 ]
>  - Release: [v2.22.DED422AA4 ( 2024-12-07 ) & v2.22E.C308FE8C ( 05-11-23 )]
 - Power adapter being used: 5v PC USB 

**Additional context**

Ralim, 2024-12-09:
This is down to the specific OLED panel used. Better quality ones will work at lower drive levels.

Gamerx007, 2024-12-09:
well that's sad

thanks for the info 

The animations in the menus are working very well on my TS101 

thanks again 😎👍

PR #2017 (2024-12-12): adjusted lowest brightness threshold for TS101
should fix #2016
closed 2024-12-12 by discip