Ralim/IronOS #2020: Pinecil V2 measures spikes in temperature when heater is on.

rkolbaskin, 2024-12-10:
**Describe the bug**
Whenever heater is on temperature reading behaves erratically with what seems like a lag followed by a jump to above 400°C. It also causes it to overshoot when heating. Temperature doesn't jump while it cools down. 

https://github.com/user-attachments/assets/28a17088-033e-4560-af7d-f570de731d90

**To Reproduce**
I don't know what is causing it.

**Expected behavior**
No measured temperature spikes.

**Details of your device:**
 
 - Device: Pinecil V2
 - Release: 2.22D.ED422AA4
 - Power adapter being used: any

**Additional context**
This happens regardless of charger used. I have TS100 and clone V1 pinecil that work as expected with same chargers. USB PD chip is removed because it got fried the day I received it (broken wire in the cable might have been the cause). I am not sure if it was behaving like this before that. On the video it is powered with DC using PD trigger and 65W charger, but same happens with QC charger. Power limit is set to 45W.

I've seen Ralim asking someone with similar issue on Discord for scope captures so here are mine:

[image]
Yellow - `Tip_Control`, Blue - `Tip_Sense`

[image]
Yellow - `A`, Blue - `Tip_Sense`

[image]
Yellow - `Tip_Control`, Blue - `A`

[image]
Yellow - `Tip_Sense`, Blue - `A`

rkolbaskin, 2024-12-10:
Correction. Power limit wasn't set during video recording, but setting it doesn't help.

rkolbaskin, 2024-12-11:
So... I put a new USB PD chip on the board and the issue seems to be gone.

Ralim, 2024-12-16:
Thank you for confirming. I2C _shouldn't_ cause this. Sounds like IRQ fighting somehow, will dig when I have some time

utaxiu, 2024-12-28:
Can confirm this wierd behavior. Yesterday I updated my Pinecil V2 to [v2.23-rc1](https://github.com/Ralim/IronOS/tree/v2.23-rc1) and after pressing "+" to begin the heating the temperature spiked and it destroyed my TS-KU tip (stuck at 450˚C, infinite resistance). I was using PD 3.1 EPR 140W power supply.

utaxiu, 2025-03-04:
I can test these new RCs in the next few days. IAC thanks for your effort.

discip, 2025-03-04:
In my case those spikes are gone.
Thank you! 😊

utaxiu, 2025-03-14:
I can also confirm that the problem was fixed on my device. Thank you!

closed 2025-03-17 by Ralim