Ralim/IronOS #2033: Last temp & sleep temp messed up in Z

CakeIsYummy21, 2024-12-25:
Last used temperature used to heat while start-up behavior is on "Z", instead of sleep temperature selected in settings, if last used temp < sleep temp
**To Reproduce**
1. Go to "Start-up Behavior"
3. Change to "Z"
4. Go to "Sleep temp"
5. Change to "60"
6. Change iron temperature to "50"

**Expected behavior**
Sleep temperature used to heat in "Z" start-up behavior, even if the sleep temperature is greater then the last used 

 - TS101
 - v2.23E.DD4A5500

CakeIsYummy21, 2024-12-26:
Maybe it's a feature, not a bug

ia, 2024-12-26:
Hello. Thanks for the report. I will look through your steps in a couple of days (i.e., as soon as possible) trying to reproduce the issue, and see what's wrong and how it could be fixed. Just be patient, _ETA_ here (estimated time of answer) is usually ~2-3 days average.

Ralim, 2025-03-04:
Hia, sorry for delay on this one @CakeIsYummy21 
I've tried to replicate this but wasn't able to, for me its sleeping to the lower of last used temp or sleep temp? am I missing anything 😓

ia, 2025-03-04:
It's getting a bit confusing, so let's clear this in simple English together. ;)

> Last used temperature used to heat while start-up behavior is on "Z", instead of sleep temperature selected in settings, if last used temp < sleep temp

> Expected behavior
> Sleep temperature used to heat in "Z" start-up behavior, even if the sleep temperature is greater then the last used

With the latest `dev` build I did this:
- set sleeping temp to 100;
- set soldering temp to 60;
- wait for sleep timeout.
The result: `Sleeping... / Tip: 60`.

@CakeIsYummy21 , what you ask is this: there must be `Sleeping... / Tip: 100` instead of 60 [because a _user_ **did set it self manually**], is this what your original report implies?

P.S. I'm sincerely apologizing that I promised to look into this earlier and didn't.

ia, 2025-03-04:
> I think it should always use the sleep temp in sleep mode.

So, to answer my question is: yes, it must be `100` instead of `60`, right?

> In other words:
> sleepHeatingTemp = min(lastUsedTemp, sleepTempFromSettings)

Well, you basically quoted the snippet from [the code](https://github.com/Ralim/IronOS/blob/c6bdc28081820a8c3f0a3802e9893f5c65435da9/source/Core/Threads/UI/logic/Sleep.cpp#L22-L26). Git history shows, that this has been implemented with [this huge change](#1749), but the trace of it may be deeper.

Interesting finding, @CakeIsYummy21. I never saw this, because I set sleep relatively low (~120), and there is no point to first set soldering temp below that point, and then wait for sleep mode: if I want sleep mode, then I just leave the iron untouched and in 40-50 seconds it's cooling off by itself.

@Ralim, should `min` be removed from `Sleep.c`, how do you think? _**Was there some case or incident which made you to put this _minimal_ condition**_?

On one hand, I like that safety net, on the other hand, I guess that if a user set manually some setting, then we should _respect_ the value of the custom user' setting.

ia, 2025-03-04:
When our builds will be [up & running again](2096), a test build to try & to check will be ready.

Basically, [this change](https://github.com/ia/IronOSf/commit/efb010256976c1417d7e042e5fb6e3c08a664ea7) should be more than enough. Except I still would like to hear from @Ralim, why exactly this safety approach has been implemented, and maybe we shouldn't change here anything.

ia, 2025-03-04:
Okay, since _we're back in business_, could you, @CakeIsYummy21, please, pick a firmware for your device from [these builds](https://github.com/ia/IronOSf/actions/runs/13661312625#artifacts) and to test it, to make sure that _**this behaves exactly like you do expect**_. Thanks.

discip, 2025-03-04:
Sorry guys, but this doesn’t make any sense.
What would be the purpose of this behavior?

The current behavior is correct.

Imagine the following:

For regular soldering, the soldering iron needs to be at around 320°C, but when melting things like plastic without burning it, 120°C is usually more than enough.

Now imagine the current sleep temperature is set to, say, 170°C.
As long as you're soldering, everything works as expected.

But on another occasion, you need to melt plastic, so you set the temperature to 120°C, work with it, lay the soldering iron down, pick it up again, continue, and only realize after the part has been touched by the iron that it's ruined because the soldering iron heated up to 170°C while "sleeping." 🤦🏻

Please leave it as it is.

discip, 2025-03-04:
> * Go to "Start-up Behavior"
> * Change to "Z"

Sorry I missed that part as I started to read the post towards the end. 😅

In this case we would need 2 separate modes:

If the **`Start-up behavior`** is set to **`Z`** use the sleep temperature.
Else use the soldering temperature.

ia, 2025-03-04:
> But on another occasion, you need to melt plastic, so you set the temperature to 120°C, work with it, lay the soldering iron down, pick it up again, continue, and only realize after the part has [...]

Aha! Very interesting and very reasonable explanation of this feature. Thanks for the clarification, @discip.

P.S. Every single tutorial I saw about soldering usually says to never ever use a regular soldering iron to melt plastic since it's not good for a tip and the result is not good anyway. So just out of curiosity - do you really use an iron to melt plastic sometimes or do you just use it as an vivid example for the current behavior? Just asking, since those results of melted-by-iron plastic, which I observed, looked pretty bad.

ia, 2025-03-04:
> Sorry I missed that part as I started to read the post towards the end. 😅

I was too fast too with my reply. :)

> In this case we would need 2 separate modes:
> If the Start-up behavior is set to Z use the sleep temperature.
> Else use the soldering temperature.

Ok, I will make additional tests with different _Start-up_ settings to see _how it works now_, and I will try to figure out _how it should work_. But as far as I understand, in the provided builds above, removal of `min` check should affect "z" start-up setting only, but I'm not 100% sure.

discip, 2025-03-04:
Regarding you question about melting plastic:

I indeed do use an old tip to occasionally melt plastic, for example, to structurally repair things that don't necessarily need to look good. 😊

discip, 2025-03-04:
> and I will try to figure out _how it should work_.

I could help with that, as I was partially involved while this was implemented.

mode | expected behavior
:-: | -
**`Off`** | Just like a regular soldering iron.
**`S`** | Should start heating up as soon as the iron is powered up.
**`Z`** | Starts in sleep mode while heating up to the previously set sleep temperature. As soon as the iron gets moved it heats up to the soldering temperature.
**`R`** | Starts in sleep mode without heating at all. As soon as the iron gets moved it heats up to the soldering temperature.

ia, 2025-03-04:
> R | **Starts in sleep mode** without heating at all. As soon as the iron gets moved it heats up to the soldering temperature.

Technically, it's _hibernating mode_, right?

> I could help with that, as I was partially involved while this was implemented.

Thanks. Funny enough, I started to draw a similar table but with more columns for every major test case. :)
This may be not accurate and not true anymore, but I remember clearly that while experimenting with these modes something was a little bit broken right after one of those huge refactoring commits long time ago, but it wasn't so important to me back then, but now it's just another good reason to double-check everything.

But now I have a question: the table you made reflects the behavior during _the initial power on_, right? And those settings shouldn't affect what is going on when sleep/hibernate/shutdown timer is triggered, right?

discip, 2025-03-04:
> And those settings shouldn't affect what is going on when sleep/hibernate/shutdown timer is triggered, right?

Correct.

discip, 2025-03-04:
> And those settings shouldn't affect what is going on when sleep/hibernate/shutdown timer is triggered, right?

As far as I know this is correct.

ia, 2025-03-04:
Basically, that's what I would like to check myself and to fill in, just for the sake of the experiment.

## dev

|standby option|soldering temp|sleep temp|in soldering mode|in sleep mode|on power on|on returning to soldering from off/z/s/r|notes|
|:------------:|:------------:|:--------:|:---------------:|:-----------:|:-------:|:-----:|:------:|
| off          | 80           |   140    |                 |             |         |       |        |
| z            | 80           |   140    |                 |             |         |       |        |
| s            | 80           |   140    |                 |             |         |       |        |
| r            | 80           |   140    |                 |             |         |       |        |
| off          | 140          |    80    |                 |             |         |       |        |
| z            | 140          |    80    |                 |             |         |       |        |
| s            | 140          |    80    |                 |             |         |       |        |
| r            | 140          |    80    |                 |             |         |       |        |

ia, 2025-03-04:
UPD: the table is fully ready.

So, I tend to agree with @discip that it's better to make as less changes as possible.
And just to make sure that we all want the same thing...

@discip and @CakeIsYummy21,
**if a user set _startup_ to _z_ AND soldering temp < sleep temp**, then  
**_on power on_ OR when _sleep mode is triggered_ after soldering during power on**
**the result state must be: sleeping mode with sleeping temp**,

**_is this correct?_**

~~... and people ask me why I drink~~ :D

discip, 2025-03-05:
@ia
Testing has to be delayed until I'm home again.

> **_Last used temperature_** ...

If I get this part right, he's referring to some kind of temp-memory from the last job.

What I imagine:
He plugs in the iron, it heats up to 140° in sleep mode. Then, he starts working, notices that the temperature starts to drop, so he adjusts the soldering temp and proceeds.

> But I must admit, that this is very strange indeed to see, how, while the iron goes to sleep, its heating up instead of cooling off and vice versa.

Exactly.

So, for what I understand from OP's request, it should look like this:

|standby option|soldering temp|sleep temp|in soldering mode|in sleep mode|on power on|on returning to soldering from off/z/s/r|notes|
|:------------:|:------------:|:--------:|:---------------:|:-----------:|:-------:|:-----:|:------:|
| z            | 80           |   140    |80 (solder.t)    |80 (sleep t)|mode:slp.,140 (sleep t)| 80 (solder.t)     | patch   |

In fact it doesn't make any sense otherwise.

Maybe OP could clarify the specific case in which he would need this behavior, ideally with a detailed example. 😊

I'm inclined to dropping this entirely, if there is no real use-case, as the way it works now is at least safer.

ia, 2025-03-05:
> So, for what I understand from OP's request, it should look like this:
> [...]

Aha. Well, as far as I understand, there is no routines to help to detect: are we going to sleep on power on or are we going to sleep from soldering. _It should be relatively easy to implement such check_, but I may be wrong.

> Maybe OP could clarify the specific case in which he would need this behavior, ideally with a detailed example.

I second this.

@CakeIsYummy21, could you, please, provide short but clear use case in simplified English, what exactly behavior you do expect? As you can see we all are getting more and more confused here. Thanks.

CakeIsYummy21, 2025-03-10:
Sorry for the long wait, I received so many comments via email that I didn't know which ones were important.

A niche use would be if you were using a low-voltage power supply to solder something heat-sensitive at a temperature lower than your sleeping temperature. The next day you plug in the soldering iron to heat it up, eat your quick breakfast, and the soldering iron would have heated up to your standard soldering temperature (sleep temperature) and you wouldn't have to wait long for the iron to heat up again.

Other than that, I can't think of a use case. 
It was an unexpected behavior and the https://ralim.github.io/IronOS/ page didn't mention anything about it, so I assumed it was a bug.

Thanks for your time and patience.

ia, 2025-03-10:
> A niche use would be if you were using a low-voltage power supply to solder something heat-sensitive at a temperature lower than your sleeping temperature. The next day you plug in the soldering [...]

Aha, now it finally starts to make sense... although I agree with @discip that the way how it works now is way much more safer.

As of for me (if anyone curious), all of these automatic features looks good and definitely it was a lot of effort to implement them, but the only thing that I use is ~40-50 sleep timeout with ~150-200 C sleep temp, and _autostart_ is fully off (it's not even heating on power-on for safety concerns). Besides that, **SAFETY FIRST**: I don't want my _"smart"_ iron to be _too smart_, i.e., not smarter than me at least... :D so if I don't press a button asking a device to do something, I prefer a device doesn't do anything, especially when I have to deal with high temperatures or with high voltages.

@discip & @Ralim, the explanation from @CakeIsYummy21 since a bit reasonable for the first power on, so how about this compromise? :
- if a user set _autostart_ to _Z_ and `soldering temp < sleeping temp`, then:
  - if it's first power-on, then go to _actual sleep temp_ (without `min(soldering, sleeping)`);
  - if it's sleep mode after soldering session, then we keep `min(soldering, sleeping)` for safety concern.

This approach so far looks the most reasonable for me, **BUT** I'm not sure how many users are out there already, who get used to _current_ behavior, but won't be aware about this new update, because then there may be unpredictable consequences.

CakeIsYummy21, 2025-03-10:
Or, maybe, you could write somewhere in the manual, how it behaves, I was kinda just confused with this

discip, 2025-03-10:
Sorry you lost me down the road guys.
To achieve what @CakeIsYummy21, if I got him right, is asking for, he would just need to set the **`Soldering Temperature`**, turn **`Shutdown timeout`** & **`Sleep timeout`** off and set **`Start-up behavior`** to **`S`**.
No need to first over heat and than cool down again. Just the right temperature at all times. 😊

discip, 2025-03-10:
Sorry, you lost me down the road guys.

To achieve what @CakeIsYummy21 seems to be asking for: Set the Soldering Temperature to the desired level, disable both **`Shutdown timeout`** and **`Sleep timeout`**, and set **`Start-up behavior`** to **`S`**. Then, simply plug the iron in before breakfast the next morning, and you're good to go for soldering when you're done with breakfast.

No need to worry about overheating and cooling down—it'll be at the right temperature right from the start. 😊

discip, 2025-03-10:
> Or, maybe, you could write somewhere in the manual, how it behaves,

That'll do.

CakeIsYummy21, 2025-03-10:
I was just confused with the current behavior and did't find any mention of this, so I thought it was a bug.

ia, 2025-03-10:
> I thought it was a bug.

Well, in short, there is the intentional line of code to pick minimal temp between _soldering temp_ and _sleeping temp_, so it's not a bug but more like safety feature.

ia, 2025-03-12:
Since we all did figure out that this is not a bug, but a safety measure, then I'm closing this issue then.

closed 2025-03-12 by ia