Ralim/IronOS #2034: Option to switch A and B button for settings

CakeIsYummy21, 2024-12-25:
**Is your feature request related to a problem? Please describe.**
I always mess up A & B button when in settings menu, the B button feels more like a set button to me and the A more like a selection button, because the B button is closer to the display.

**Describe the solution you'd like**
An option in the settings menu to switch A and B button for the settings menu, in the user interface tab, next to + - button switch.

Ralim, 2024-12-26:
Fair enough, should be reasonable to do.

One question; in your mind, how should this play with the existing +/- swap option?.
Does this make a flip in the settings menu separate or do they stack?

I.e if that option is on but the new one is off, are they still swapped in the menu?
And if both are on, is the settings menu then "normal".

CakeIsYummy21, 2024-12-26:
I would say they could't interfere with each other, because the +/- swap is only when changing temperature while soldering and the A/B should be **only** in the settings menu, neither in the main menu, nor the soldering menu, nor the debug menu, they all make sense to my brain right now.
Map:
`                          Debug Menu`
`TeMpErAtUrE   Soldering   Main Menu   SETTINGS 1   SETTINGS 2`
Normal
+/- SwApPeD
A/B SWAPPED

Thank you very much!

ia, 2024-12-26:
> I always mess up A & B button when in settings menu, the B button feels more like a set button to me and the A more like a selection button, because the B button is closer to the display.

TL;DR: in fact, I could agree with you that this could be useful option for some people, especially those who are new ones to use _IronOS_: for some reason in the beginning of using _IronOS_ my brain always did try to use "_B_" as _"the enter"_  (probably you're right - since it's closer to a display), and "_A_" as more like "_change/back/scroll_" (to the opposite defaults).

@CakeIsYummy21, your report is very funny (in a good way)! Because to be fully honest, this was the very first thing which was bothering me when I booted up _IronOS_ and ended up in the settings for the first time. I even was starting to work on a patch doing exactly what you suggest now, but, first of all, we were running out of free flash space back then for some supported hardware, and, second, by the time I started to make other improvements, I just did get use to the controls, LOL. :)

CakeIsYummy21, 2024-12-26:
LOL
Is there now enough free flash space?
Thx

ia, 2024-12-26:
> One question; in your mind, how should this play with the existing +/- swap option?. Does this make a flip in the settings menu separate or do they stack?
> 
> I.e if that option is on but the new one is off, are they still swapped in the menu? And if both are on, is the settings menu then "normal".

@Ralim, if you don't mind I would like to work on this one. And I fully agree with @CakeIsYummy21 - the new option should be fully independent from anything else and affects buttons' behavior **only in menu**.

ia, 2024-12-26:
> Is there now enough free flash space?

Yes-yes, now it shouldn't be a problem, thanks to the huge refactoring by @Ralim.

Ralim, 2024-12-26:
@ia Always feel free to grab things if you have an interest to do them 😁 
I'm definitely 100% onboard with this as a feature.

ia, 2025-03-10:
Dear @CakeIsYummy21, it seems you're very busy lately, and this is totally normal, since we all have our lives... but I sincerely hope that everything is ok.

And if you will be having some time and if you still interested in this setting, could you, please, test [these builds](https://github.com/Ralim/IronOS/actions/runs/13755708153#artifacts) and provide feedback. New option is located in `Settings - User interface - Swap A B buttons`. Thanks.

closed 2025-03-19 by ia