Ralim/IronOS #2038: Tip connected detection not working in v2.23 RC1

tr4nt0r, 2025-01-05:
**Describe the bug**

**To Reproduce**

Insert/Remove tip.

**Expected behavior**

Display previously displayed an icon of a crossed tip when the tip was removed. Now it displays a temperature of ~440°C when the tip is removed. The tip connected detection is done by comparing the current temperature with the max tip temperature. If the current tip temp has an invalid value (was around 475° before) that is higher than the max temp, it is assumed the tip is disconnected. I used the same logic in the Home Assistant integration for a tip connected sensor and deactivating some other sensors instead of displaying invalid values.  

**Details of your device:**
 
 - Device: Pinecil V2
 - Release:  v2.23 RC1
 - Power adapter being used: PinePower GaN 65W adapter
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2025-01-05:
Hia,
This is not expected. Can you confirm what handle temp, raw tip, and tip temp you see in the debug menu with tip connected and disconnected?

tr4nt0r, 2025-01-06:
Tip disconnected:
- Tip C: 437
- Han C: 27.0
- Max C: 445
- Tip R: 11130 uV

Tip connected:
- Tip C: 26
- Han C: 26.8
- Max C: 444
- Tip R: ~ 630 uV

Same values as those i receive via Bluetooth

colonel-lp, 2025-01-11:
Im going to confirm this one too. tip removed shows 437 or 438 (2.23C.B9BBF585 2025-01-11)
Tip disconnected:

Tip C: 436
Han C: 24.7
Max C: 452
Tip R: 11149

Tip connected:

Tip C: 24
Han C: 24.1
Max C: 452
Tip R: 575

NovaViper, 2025-02-21:
Confirming the same issue on my Pinecil V2 with v2.23E.DD4A5500 2024-12-22
Tip connected:
Tip C: 31
Han C: 31.2
Max C: 454
Tip R: 610-635 (jumps between this range)

Tip disconnected:
Tip C: 443
Han C: 31.0
Max C: 454
Tip R: 11149

Ralim, 2025-02-22:
Hia, can someone test that [these builds](https://github.com/Ralim/IronOS/actions/runs/13469527117) fix this for you

PR #2082 (2025-02-22): Dont adjust pinecilv2 max temp by Coe

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Fixes #2038 ideally

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:
closed 2025-02-22 by Ralim