Ralim/IronOS #2039: Inconsistend IDs for current operating mode over BLE in v.2.23RC1

tr4nt0r, 2025-01-05:
**Describe the bug**

The IDs for current operating mode returned over BLE have changed and are different from the IDs used in v.2.22.

I use this mapping in the Home Assistant integration:

```
    IDLE = 0
    SOLDERING = 1
    BOOST = 2
    SLEEPING = 3
    SETTINGS = 4
    DEBUG = 5
```

In v.2.23rc1 for example idle is now 4 (previously settings), settings is now 9, debug is now 10, soldering is now 5 (previously debug)

**Expected behavior**

The IDs should not be changed and new options should get different IDs that don't collide with previous IronOS versions.

**Details of your device:**
 
 - Device: Pinecil V2
 - Release: v.2.23rc1
 - Power adapter being used: PinePower GaN 65W adapter
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2025-01-05:
Ah this is my bad, they should have been hard coded rather than auto incremented.

Ralim, 2025-01-05:
@tr4nt0r Can you please test the builds on the linked PR's actions page 🙇🏼

PR #2042 (2025-01-06): Allocate fixed ID's for Operating Modes

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Fix the OperatingModes struct to match the ID's of the older 2.22 release

* **What is the current behavior?**

Closes #2039 
* **What is the new behavior (if this is a feature change)?**

* **Other information**:
closed 2025-01-06 by Ralim