Ralim/IronOS #2041: TS-101 enter boot loop on USB-C PD (QC-92)

PaulKissinger, 2025-01-05:
**Describe the bug**
TS-101 enters bootloop after 3 seconds (PD-Debug mode 6) with connected tip. Without tip, TS-101 boot regularly.  

**To Reproduce**

1. Plug USB-C  cable into TS-101
2. Wait 3 Seconds
3. See error --> Enters reboot loop

**Expected behavior**
Boot without reboot loop

**Details of your device:**
 
 - Device: TS101
 - Release: 2.23D.000EBA7B 2024-12-26
 - Power adapter being used: USB-C PD (QC-92) (Part of MHP-30 delivery)

**Additional context**

PD Debug State 3 
-> 
PD Debug State 3 No VBUS 
->
PD Debug State 6
-> 
Boot loop

Ralim, 2025-01-05:
Eugh; I dont see this here on my PD supplies. 
Do you have any means of capturing the PD traffic?

PaulKissinger, 2025-01-05:
Sorry, i have no idea how to capturing PD traffic. 

I'll test prebuilds, if you have an idea.

DarkOperation, 2025-01-13:
same with `Baseus CCGAN2P-L01`. On charger w/o PD ts101 boots normally

<sub><sub>sorry for my english</sub></sub>

stestagg, 2025-01-26:
I'm seeing something similar using a ugreen power brick (brick does 130W max).

With the default firmware, everything works, but ironOS reboots after about 3 seconds on.

I've captured the PD negotiation (I think, assuming my dodgy cable assy is working).

Side-by-side logs of the PD negotiation are here: https://docs.google.com/spreadsheets/d/1SvuCWFnzG3vurIfFmuho2cohFiriwjAlfsj0Mu-qEQo/edit?usp=sharing

'Combined' sheet, the iron os starts column J.

Iron is the TS-101 flashed with 2.23rc1.

stestagg, 2025-01-30:
Looking at the logs, I can see this for IronOS:
```
(r3) SRC[0]: SOURCE CAP [+]
   [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained]
   [2] [Fixed] 9V 3A (27W)
   [3] [Fixed] 12V 2.5A (30W)
   [4] [PPS] 5/11V 3A (33W)
...
(r3) SRC[0]: SOURCE CAP [+116477 us]
   [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained]
   [2] [Fixed] 9V 3A (27W)
   [3] [Fixed] 12V 2.5A (30W)
   [4] [PPS] 5/11V 3A (33W)
(r2) SNK[0]: GOOD CRC [+1087 us]
(r3) SNK[0]: REQUEST [+2224 us]
   [1] (PDO #3: Fixed 12V) 2.5A (operating) / 2.5A (max) [no_suspend]
(r3) SNK[0]: REQUEST [+1611 us]
   [1] (PDO #3: Fixed 12V) 2.5A (operating) / 2.5A (max) [no_suspend]
(r3) SNK[0]: REQUEST [+1611 us]
   [1] (PDO #3: Fixed 12V) 2.5A (operating) / 2.5A (max) [no_suspend]
(r3) SNK[0]: REQUEST [+1611 us]
   [1] (PDO #3: Fixed 12V) 2.5A (operating) / 2.5A (max) [no_suspend]
#25   (3113.798000ms): HRST [+22314 us]
```

Whereas the original firmware has:
```
(r3) SRC[0]: SOURCE CAP [+]
   [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained]
   [2] [Fixed] 9V 3A (27W)
   [3] [Fixed] 12V 2.5A (30W)
   [4] [PPS] 5/11V 3A (33W)
...
(r3) SRC[0]: SOURCE CAP [+116431 us]
   [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained]
   [2] [Fixed] 9V 3A (27W)
   [3] [Fixed] 12V 2.5A (30W)
   [4] [PPS] 5/11V 3A (33W)
(r2) SNK[0]: GOOD CRC [+1087 us]
(r3) SNK[0]: REQUEST [+5097 us]
   [1] (PDO #4: PPS 5/11V) 5V 3A
(r3) SRC[0]: GOOD CRC [+744 us]
(r3) SRC[1]: ACCEPT [+551 us]
(r2) SNK[1]: GOOD CRC [+569 us]
(r3) SRC[2]: PS RDY [+191422 us]
(r2) SNK[2]: GOOD CRC [+569 us]
(r3) SNK[1]: REQUEST [+2335897 us]
   [1] (PDO #4: PPS 5/11V) 5V 3A
(r3) SRC[1]: GOOD CRC [+744 us]
(r3) SRC[3]: ACCEPT [+551 us]
(r2) SNK[3]: GOOD CRC [+570 us]
(r3) SRC[4]: PS RDY [+190962 us]
(r2) SNK[4]: GOOD CRC [+570 us]
```

DieMetRik, 2025-02-28:
TS101
update to v2.23-rc2 boot loop wo any chance enter in any menu
update to  v2.23-rc2 boot normaly, but after press button "A" reboot
back to 2.22 all works!

Achduka, 2025-03-01:
Yes same with rc3. With 2.22 all works.

juhotauriainen, 2025-03-03:
Same boot loop problem for me with every release after 2.22. (Using USB-PD protocol)

Device: TS-101
Power supplies tested:
 Official Apple 61w USB-C (A1947)
Apple macbook air m1 usb port
Apple 20w charger (
Cable: Mophie 240w Usb-c from official Apple store online order, Apple 60w basic usb-c cable, genuine, came with apple product

Bootloop does not happen for some reason if the tip is not in the device. But after inserting the welding tip it starts to happen again.

Ralim, 2025-03-03:
@stestagg 
Any chance you can do a capture on the later 2.23 rc's by any chance? I haven't been able to reproduce a capture to match yours on my on unit here but want to check if anything has changed. I'm fairly suspect something about the IRQ/Status bits isn't working on TS101 but unsure what and hard to tap the unit for proper debugging at the moment.

stestagg, 2025-03-10:
Here's the captures for 2.22 and 2.23rc3:

<details><summary>2.22-100w</summary>
<code>
#1    (143.678800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#2    (145.575000ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#3    (147.799200ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#4    (150.322600ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#5    (152.218600ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#6    (154.095400ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#7    (155.998000ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#8    (158.569800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#9    (160.726600ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#10   (162.648800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#11   (164.535400ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#12   (166.791800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#13   (169.286600ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#14   (171.173000ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#15   (173.082200ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#16   (174.994800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#17   (177.576200ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#18   (179.739400ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#19   (181.600200ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#20   (183.515600ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#21   (185.765800ms): (r3) SNK[0]: VDM - [1] REQ Disc Ident  SVID:ff00
#22   (188.270200ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#23   (304.011200ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#24   (419.766400ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#25   (535.512000ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#26   (650.597400ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#27   (766.028800ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#28   (881.758600ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#29   (997.521400ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#30   (1113.248000ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#31   (1229.304800ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#32   (1344.687000ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#33   (1460.411600ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#34   (1576.142200ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#35   (1691.862200ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#36   (1807.898000ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
#37   (1923.306400ms): (r3) SRC[0]: SOURCE CAP - [1] [Fixed] 5V 3A (15W) [dual_role_power] [unconstrained] - [2] [Fixed] 9V 3A (27W) - [3] [Fixed] 12V 3A (36W) - [4] [Fixed] 15V 3A (45W) - [5] [Fixed] 20V 3A (60W) - [6] [PPS] 3.3/21V 3A (63W)
[... cut ...]
#42   (2845.938600ms): (r3) SNK[0]: REQUEST - [1] (PDO #5: Fixed 20V) 3A (operating) / 3A (max) [no_suspend]
#43   (2847.551600ms): (r3) SNK[0]: REQUEST - [1] (PDO #5: Fixed 20V) 3A (operating) / 3A (max) [no_suspend]
#44   (2849.164400ms): (r3) SNK[0]: REQUEST - [1] (PDO #5: Fixed 20V) 3A (operating) / 3A (max) [no_suspend]
#45   (2850.777400ms): (r3) SNK[0]: REQUEST - [1] (PDO #5: Fixed 20V) 3A (operating) / 3A (max) [no_suspend]
#46   (2869.656400ms): HRST
#47   (2948.825200ms): Junk???
</code>
</details>

stestagg, 2025-03-11:
I found the issue that mentioned swd over usb, and hooked that up injected alongside the battery power, so can now debug the cpu, but that doesn't actually help, because when I flashed a local build from dev or from the v2.23-rc3 tag, the issue goes away.

I tried with the debugger in the loop and direct unmutilated usb-c to the power block and couldn't get it to reboot at all.

Only time I can get the boot issue is when I flash the hex from the releases zip.

I'm having issues with getting docker build working (windows environment) right now, so can't check that build yet.

Ralim, 2025-03-11:
Interesting, compiler optimisation would affect some of this depending how it handles things. Sadly I can only work with the docker setup to ensure things are built with the same compilers.

Note that having a debugger attached can affect this, so have to flash and disconnect before testing.