Ralim/IronOS #2043: MPH-30 shows shows wrong temperature

PaulKissinger, 2025-01-05:
**Describe the bug**
Shown temperature is 100 °C to high. MPH-30 shows 120 °C instead of 20 °C

**To Reproduce**
1. Flash dev build 
2. Plug USB-C power into cold MHP-30
3. See error --> 120 °C  instead of 20 °C

**Expected behavior**
correct temperature measurement 

**Details of your device:**
 - Device: MPH-30
 - Release: 2.23D.000EBA7B 2024-12-26
 - Power adapter being used: USB-C PD (QC-92) (Part of MHP-30 delivery)

**Additional context**
Correct temp measurment with V2.22E.C308FE8C

Ralim, 2025-01-05:
Oh yay, this could be a regression from trying to fix the temperature reading for the newer models of the MHP-30. Will likely need someone to dig into more what its going on with the temp sense hardware.

PaulKissinger, 2025-01-05:
If you have an hint for me how where to start, i'll do this :D

wu9226, 2025-02-03:
Had the same issue on v2.23

closed 2025-03-05 by Ralim

Sandman224, 2025-03-06:
This seems to have the same issue as before. The latest pr #2094 does fix the text issues on the power menu but I am still seeing 120 degrees at room temperature with the latest dev build.

Ralim, 2025-03-06:
@Sandman224 and to confirm your unit reads correctly on 2.22 release?

Ralim, 2025-03-10:
If it reads high on 2.22 and the linked fixed build; then the issue you have is likely not this but related to the other MHP30 issues from when Miniware changed to clone STM32 chips and caused a few readings to break.

Sandman224, 2025-03-10:
I see. I do know that my MHP30 contains one of the clone chips so that may be the case. I did have this working on a previous dev build however but that was likely from August of last year. Did something else change that breaks the temperature functionality for clone chips in this same way?

Ragarnoy, 2025-08-08:
@Ralim Could we perhaps re-open this issue as it seems it's still not fixed ? I'm having the same issue on a newly bought MHP30

commit 1030a491 (2025-03-04): Fixes for ADC mode init

Fixes #2043
commit b6109cae (2025-03-05): Fixes for ADC mode init

Fixes #2043
PR #2094 (2025-03-05): Mhp 30 fixes

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Closes #2036
Closes #2092
Closes #2043
Helps with #1739

* **Other information**:
Tested on my triplet of MHP30 units
commit 5e8ab279 (2025-03-05): Mhp 30 fixes (#2094)

* Fix LED control for MHP30

* Refactor the draw rect function to fix heating

* FIXUP to include correct menu title for MHP30

Closes #2036

* Fixes for ADC mode init

Fixes #2043