Ralim/IronOS #2044: Expose new `HallEffectSleepTime` and `SolderingTipType` via BLE in v.2.23

tr4nt0r, 2025-01-06:
**Is your feature request related to a problem? Please describe.**

2 new setting options were introduced in v.2.23rc1, these are not exposed via Bluetooth characteristic and as so cannot be changed remotely

**Describe the solution you'd like**

Add new characteristics to allow setting the new options via BLE

**Describe alternatives you've considered**

**Additional context**

Ralim, 2025-03-04:
Hia @tr4nt0r Any chance you can test [this build ](https://github.com/Ralim/IronOS/actions/runs/13650834751?pr=2095)

tr4nt0r, 2025-03-04:
Sure 👍🏼

tr4nt0r, 2025-03-04:
Works great! I can read and write values over BLE. 
Just to be sure I got everything correct:
- Hall effect sleep time is 0-60s with a step size of 5s. The raw values are 0-12
- Values for tip types are 0: Auto-sense, 1: TS100 long, 2: Pine short, 3: PTS200

I will start implementing it in the python library and in Home Assistant :)

PR #2095 (2025-03-04): Add missing BLE settings

* **Please check if the PR fulfills these requirements**
- [] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Fixes #2044

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:
commit 50413263 (2025-03-04): Add missing BLE settings (#2095)

Fixes #2044
closed 2025-03-04 by Ralim