Ralim/IronOS #2046: My pinecil v2 isnt working with pd charger i think it should work with

Mzter-Pink, 2025-01-07:
I've tried the Pinecil V2 with two different ASUS ROG power bricks to no avail.... I flashed it with 2.23e.dd4a5500 tried again still nothing so I tried 2.22e.c308fe8c and still nothing... The led on the Pinecil breakout board does illuminate the entire duration when plugged in and it does not heat with the screen off (i did try)

pd debug does not report anything while connected the device is compleatly unresponsive

[ASUS ROG 65W HDMI CHARGER DOCK - Best Buy](https://www.bestbuy.com/site/asus-rog-65w-charger-dock-supports-hdmi-2-0-with-usb-type-a-and-usb-type-c-for-rog-ally-black/6542054.p?skuId=6542054) tried it with the included wire and an Anker wire) along with the powersupply that is included with the rog ally (with the unremovable wire)..

the device does in fact heat with a different 65w laptop charger so I know its not hardware!!

Mzter-Pink, 2025-01-08:
the dock is capible of PD3.0, PD3.0 PPS and PD2.0 at 65W

Ralim, 2025-01-09:
Can you try testing with a USB 2.0 USB-C cable being used to connect to the dock to rule out it interfering with the debug signals that are on the USB-C port by default.

rkolbaskin, 2025-01-25:
Try to disable bluetooth. Mine doesn't work with some chargers as long as it is on and latest stable release has bluetooth on by default.

Ralim, 2025-03-04:
Hia, can you please when you have time re-test with the latest RC?

Mzter-Pink, 2025-03-04:
I just flashed it with V2.23-rc3 and it gives me the same results. Theres a
good chance it has something to do with the way the block is trying to
communicate(its a USB HDMI splitter but they had to put a reset button on
the back of it because how unreliable there protocol is so at this point i
blame rog)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ralim, 2025-03-05:
By default the Pinecil exposes some extra buses on the USB3 pins to allow for people to tinker with it easier. I'd be suspect that the dock is also using these pins before negotiation. And so they are conflicting.

Its your call if you want to test this, it would require removing the 0 ohm link resistors in the pinecil to disconnect these pins.

Mzter-Pink, 2025-03-05:
My other soldering iron the element popped(bought the Pinecil because it
was easier and the same price as sourcing the element) so I don't have a
way to solder the Pinecil but I do have the breakout board if i can modify
that to achieve the same effect!!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ralim, 2025-03-05:
Hmm, only way I can think of with the debug board would be if you cut the traces on the data signals which is a bit messy.

By any chance do you have a USB 2.0 C-C cable? and can you use it with these? (as those cables dont connect these pins)