Ralim/IronOS #2047: Pinecil V2 randomly enters cooldown mode

ebertek, 2025-01-10:
**Describe the bug**
After heating up, the Pinecil randomly goes back to cooldown mode.

**To Reproduce**
1. Press '+' to heat up
2. Wait (or try to solder)

**Expected behavior**
The device should not enter cooldown mode before idle delay.

**Details of your device:**
 - Device: Pinecil V2
 - Release: 2.23
 - Power adapter being used: Ugreen Nexode GaN (UFCS, PD3.1@140W, PD3.0@100W, PPS, QC5), HP TPN-CA06 (3A@5V, 3A@9V, 5A@10V, 5A@12V, 4.33A@15V, 3.25A@20V)
 - If this is an accelerometer related issue, please include its model number here: N/A

**Additional context**
Setting motion sensitivity to zero (disabling sleep) does not change this behavior.
The issue is not present in IronOS 2.22.
More people reported the issue on Reddit: https://www.reddit.com/r/PINE64official/comments/1hp9mhc/issue_with_pinecil_not_staying_heated/

Ralim, 2025-01-10:
Hia,
Can you confirm what temperature you are running at?
This does feel like a bug in the beta, just need to figure out the cause.

Unfortunately as Reddit killed off third party apps I'm no longer able to read/monitor it.

ebertek, 2025-01-10:
Hi @Ralim, thank you very much for looking into this!
I was using 320°C with the original ST-B2 tip. I haven't found a way to trigger (or not to trigger) this behavior, but I can run some test cases if you have anything on your mind. I think it never stays at the set temperature for over 2 minutes, but sometimes it goes back to the main menu even before reaching the target temperature.

Ralim, 2025-01-10:
So the list of reasons why the Iron should exit from soldering mode back to the home screen:
1. Both buttons pressed at once
2. B button long-hold
3. Undervoltage
4. Shutdown timeout with sleep disabled
5. Thermal runaway is detected

To rule out if its is fixed by the tweaks already made to the thermal runaway code; could you test the latest dev build ([can download from the bottom of this page](https://github.com/Ralim/IronOS/actions/runs/12718581075))

colonel-lp, 2025-01-11:
Can confirm same behaviour on Pinecil V2. Heats to 320 as set, voltage dips to 19.9v and a blank screen slides across then goes into cool down. Happens in less than 10 seconds after reaching 320, if you try to heat again by pressing plus, it happens quicker and quicker. Doesn't happen on Ts101 running 2.23, but that's also reading Pd at 20.8v rather than 20v on P V2.

V2.23,  pine short tip, and gvda tips

Pd 20.1V from Anker 167W 12V car charger/adapter. (100w port)
Pd 20.2v from MacBook pro 61v AC charger

Tried with multiple cables with 100+w rating, Anker, MacBook pro, etc.

Ralim, 2025-01-11:
Can you confirm this with the above linked dve build just to be sure?
Can you grab a short video of it so I can see exactly what the screen does?

colonel-lp, 2025-01-11:
lastest build posted above appears to fix the problem, but screen is messed up and glitching, top half of the screen is replaced by a glitchy copy of the bottom half, then it gets more and more messed up

colonel-lp, 2025-01-11:
https://github.com/user-attachments/assets/98a8b4f5-52af-43c1-8a65-e38c362f8e52

Ralim, 2025-01-11:
Thank you

Can you test this version:
https://github.com/Ralim/IronOS/actions/runs/12720132310

And this version:

https://github.com/Ralim/IronOS/actions/runs/12720229392?pr=2049

colonel-lp, 2025-01-11:
From a short period of testing It looks like both versions have fixed the cooldown issue. The screen glitching I think is to do with carried over settings 

Flashing:
V2.22> 1st version, no screen glitch
1st version to 2nd version, glitching screen
2nd version to 1st version, glitching screen,

resetting to default in device clears screen glitches

aka13-404, 2025-01-11:
Can confirm as well, thought I was going crazy. Turning off all auto shutoff features seems to mitigate the problem.

aka13-404, 2025-01-11:
Just tried out https://github.com/Ralim/IronOS/actions/runs/12720132310, and now sleep and everything else works as expected. Thanks for the lightning speed, appreciate your efforts.

Ralim, 2025-01-12:
Eugh yeah ok. Been trying to work around a 3-way problem, Bluetooth interferes with interrupts that can cause screen glitching; so then I can disable BLE's interrupts during updating the screen, but then that causes issues with temp readings.

I'm sadly going to be away from my PC for ~3 weeks :(

colonel-lp, 2025-01-12:
Hopefully it's joyfully away from your pc for 3 weeks. Enjoy. 😁

tatokis, 2025-01-15:
I just installed the latest build (fd91cf9) and the issue still occurs. b401dc8 seems to work fine.

I don't use bluetooth, so I can't comment on its functionality.

Pukesmiley, 2025-02-02:
> Can confirm same behaviour on Pinecil V2. Heats to 320 as set, voltage dips to 19.9v and a blank screen slides across then goes into cool down. Happens in less than 10 seconds after reaching 320, if you try to heat again by pressing plus, it happens quicker and quicker. Doesn't happen on Ts101 running 08/01 dev build, but that's also reading Pd at 20.8v rather than 20v on P V2.
> 
> 08/01 dev build, pine short tip, and gvda tips
> 
> Pd 20.1V from Anker 167W 12V car charger/adapter. (100w port) Pd 20.2v from MacBook pro 61v AC charger
> 
> Tried with multiple cables with 100+w rating, Anker, MacBook pro, etc.

i also had the same issue with the anker20,000mAh Powerbakn 200W but not with my Switch on 45W fast charger.
I didnt see the voltage dip when the problem occured today, but saw it yesterday The shown voltage is 18.8V

vavanbonus, 2025-02-09:
Same issue on two pinecil v2. ebdff5929385f5b6260b6e036462d2b97d39197f and b401dc8264e5dda826a18a8be5c3827368e7808f have same randon turn off behavior.

esotericode, 2025-02-13:
Having the same issue on mine, only on 2.23 with any charger I use to power it. Works perfectly with every charger I try on 2.22. I'm so glad I found this thread because I suspected I had a defective unit, very happy to find out it's a known bug.

tatokis, 2025-02-14:
A month later, and https://github.com/Ralim/IronOS/commit/b401dc8264e5dda826a18a8be5c3827368e7808f has been working perfectly for me.

I tried the latest code and the issue still occurs. If anyone wants to ru/testn that commit in the meantime, I've built it since the artifacts seem to have expired.

[Pinecilv2_EN_b401dc8.zip](https://github.com/user-attachments/files/18805115/Pinecilv2_EN_b401dc8.zip)

vavanbonus, 2025-02-16:
> Same issue on two pinecil v2. [ebdff59](https://github.com/Ralim/IronOS/commit/ebdff5929385f5b6260b6e036462d2b97d39197f) has same random turn off behavior (unfortunately don't gnow how to get firmware from other that dev branch). Voltme revo 100 charger PD [@20V](https://github.com/20V). I've also tried with old xiaomi 2s powerbank. It has qc2 and 12v output. Pinecil triggers only 9v (as expected, because no qc3 support). I've taken a bad cable to test intentionally. So, pinecil sometimes displays "Undervoltage Input V: 8.9V" and goes into standby. So maybe when PD is enabled we have same undervoltage issue?.. At least 0.1v drop is very small value to switch into error state.

this problem is confirmed using DC power. so PD is probably not the reason

Ralim, 2025-02-19:
I have rebased #2049 which you can download the builds [at the bottom of this page](https://github.com/Ralim/IronOS/actions/runs/13406623527).

This has a different method of doing the I2C mutex that doesn't break the screen (instead of [b401dc8](https://github.com/Ralim/IronOS/commit/b401dc8264e5dda826a18a8be5c3827368e7808f)).
It also requires multiple trips of the runaway trigger before tripping and exiting soldering.
Would be good if someone can try this.

colonel-lp, 2025-02-19:
I tried the build at the link you provided, I afraid there are still screen issues when Bluetooth is enabled and connected. It worked for about 5 minutes but then had a few glitches and the screen went off completely (iron still visible and working in the app), had to unplug to get the screen back, no button pressed brought it back.

Ralim, 2025-02-19:
Thanks for testing 😄 
Guess I will need to figure out a different option to make the BLE blob behave.

Totoro-spec, 2025-02-21:
also instability Pinecil V2 increases when the hall sensor is installed

Ralim, 2025-02-22:
[New build](https://github.com/Ralim/IronOS/actions/runs/13469418268) that I'd love it if someone can test please. Also have updated assets to stick around for 90 days on these actions runs.

colonel-lp, 2025-02-22:
Hey. I'm afraid with Bluetooth enabled the glitches start after a few minutes.  Not as bad as the last build. Top half of screen is functional but flickering,  bottom half is not and is frozen. 

[image]

Ralim, 2025-02-22:
@colonel-lp 
How are you reproducing this? Are you using bluetooth with an app or HA?

@RobThree I designed the pinecilv2, I have 10 of them in a box and I haven't yet reproduced this on _any of them_. Which is why I'm asking for testing so much as its probably situational

RobThree, 2025-02-22:
@Ralim 🤦‍♂️Boy do I feel stupid now 😅

colonel-lp, 2025-02-22:
@Ralim so, previous versions had screen glitches just with Bluetooth turned on and seemed to be independent from any other settings and the iron heating. This version as far as I can tell doesn't glitch with Bluetooth turned on alone, but as soon as I connect to the android ironOS companion app, the glitches start, only tried this on the last two versions as was curious. It's a little hard to pin down now as the glitches have never been consistent with particular button presses or time passed, unlike the earlier versions when just turning on Bluetooth created the problem quickly.

colonel-lp, 2025-02-22:
@Ralim sorry, I just tried again and what I said about it only happening with the app isn't the case. It's doing it now from boot with ble on but not connected to anything. It's really random as to when it starts happening. I'd be wondering if I have a faulty device but it never did this on V2.22

Ralim, 2025-02-22:
@colonel-lp It's not a fault of hardware, don't panic. Just this is caused when the BLE "steals" the cpu for too long, causing data to get dropped when updating the screen.

I used to protect against this by disabling interrupts and locking out the BLE blob while updating the screen, but that also caused the over-reads on temperature that were false tripping the runaway detection.

So I've reworked the screen driver somewhat, and I'll have another look at it today to try and improve things further. Ideally I'd use DMA to do this as that wont use the CPU but I dont have it working (tried for a few nights after work).

So mostly trying to figure out a way to reproduce here :)

@RobThree No stress, the offer is appreciated, though any design flaw can be blamed on me 😁

RobThree, 2025-02-22:
> @RobThree No stress, the offer is appreciated, though any design flaw can be blamed on me 😁

In that case: Get on with it! 😜
No, but seriously: thanks! I love my Pinecil V2 ❤

colonel-lp, 2025-02-23:
Thanks for all the effort, the Pinecil V2 is indeed a wonderful device.  Back to v2.22 for now. 

The only other settings I've changed are:
Cool down flashing, and enabled the two  detailed screens, which I'm guessing is asking more of the screen driver and CPU. 

Something I also just noticed with this build is it's now negotiating 15v with my Anker pd charger where on previous builds it was 20v, so not sure if you are working on other code too.

Ralim, 2025-02-23:
@colonel-lp 

[New build](https://github.com/Ralim/IronOS/actions/runs/13480906384?pr=2049), if you could test please. Managed to replicate it here using the companion app and this so _far_ works for me. Does come with slight tradeoff that very rarely BLE has a slightly longer pause than normal to apply but didn't seem a bit deal. (Open to feedback if you see it, and if so what you think of it)

@RobThree May it work well 😁

vavanbonus, 2025-02-23:
> [New build](https://github.com/Ralim/IronOS/actions/runs/13480906384?pr=2049), if you could test please. 
tried it with companion app. playet like 10 minutes wit settings, heating etc - everything works perfectly so far.

Ralim, 2025-02-23:
Fantastic,
I'll leave this for a bit to see if anyone yells, but it has worked for me too here for a while now so will merge it if nobody yells soon.

>Something I also just noticed with this build is it's now negotiating 15v with my Anker pd charger where on previous builds it was 20v, so not sure if you are working on other code too.

Forgot to answer this earlier, it was talked about in the pinecil chat and decided to default to PPS/EPR modes turned off by default (you can just turn them back on in the settings) Which should be why this is happening.

Totoro-spec, 2025-02-23:
hmmm in last firmware only 20V in my Apple 140W power supply, i try - power setting/pd mode
1. Default mode
2. Safe mode
3. No dynamic 
nothing change!

Ralim, 2025-02-23:
@Totoro-spec Can you check if this is caused by this build by trying a [build from dev](https://github.com/Ralim/IronOS/actions/runs/13476947753) as well

Totoro-spec, 2025-02-24:
0515832
PD work, but randomly goes back to cooldown mode return ((

Ralim, 2025-02-24:
Does EPR work (28V?) that is what I want to confirm, the cooldown issue will occur as that linked commit doesnt have the fixes in this branch

Totoro-spec, 2025-02-24:
yep, 28V return, but cooldown return too

Ralim, 2025-02-24:
@Totoro-spec  Thank you for finding this.
Can you test [new builds](https://github.com/Ralim/IronOS/actions/runs/13494866241/job/37699777535) when you have a chance.

Totoro-spec, 2025-02-24:
this link? https://github.com/Ralim/IronOS/actions/runs/13494866241/ 

PD 28V, cooldown return too

Ralim, 2025-02-24:
What do you do to trigger cool down exit on this?

Totoro-spec, 2025-02-24:
Press '+' to heat up
after seconds screen off/on

ericus256, 2025-02-24:
Just my own experience- had the same issue, Pinecil v2 at 2.23-rc1, Anker A1340 250W Power Bank (28V EPR) and UGREEN 300W (also 28V EPR on first port). Multiple certified cables; would always jump itself back to the "home" screen as it started ramping up to temp (not Hot-idle/countdown, but the "cooldown" page).

Just tried https://github.com/Ralim/IronOS/actions/runs/13495017303 build back-and-forth a couple of times and confirm the issue is resolved there.

Totoro-spec, 2025-02-25:
https://github.com/Ralim/IronOS/actions/runs/13495017303 cooldown.

Ralim, 2025-02-25:
@ericus256 This is the main thing I'm waiting to solve for 2.23 RC2 tbqh

@Totoro-spec It sounds like your seeing a power reset, not a cooldown mode exit if the screen is going off and back on. This is often triggered when the power supply can't keep up with the power demand (OR the PD timer expires for the message communication). What exact model power adaptor, cable and tip are you using 😓 Also if possible, can you take a video of the issue just so I can see what the screen shows to see if its a power reset or just a cooldown exit.

Not trying to dismiss, just trying to figure out how I can reproduce :)

Totoro-spec, 2025-02-25:
i use original Pine 65W & original Apple 140W model A2452

vavanbonus, 2025-02-26:
so, I've bought Voltme revo 140 charger. testing it with https://github.com/Ralim/IronOS/actions/runs/13495017303
using essager 100W cable:  uses 19.9V, pd debug max mode is pd6 3-21 3.00A
using red cable has 3-21 3.00A mode, but stays at 15.1V 🤦‍♂️ no random turnoffs or something like that. but low power with red cable 😥 Probably I'll buy 240W cable just to be sure, but i don't think that will help to get 100W mode back....

Ralim, 2025-02-26:
140W needs an EPR e-marked cable (for >20V, and 5A).
100W needs an e-marked cable (for 5A).
un-marked cables will limit you to 20V 3A (60W).

Depending which PD setting you chose, the device will select from
1. Default mode (uses slight resistance padding when selecting voltage, so that you can fudge 20V on a 3A limited cable, since average power is < 60W).
2. "safe" mode, which doesnt use the padding resistance, so it will limit you to 15V on a 60W supply. This means you never push anything super close to its limit, ergo "safer", especially on power banks.
3. No PPS/EPR + safe. This is using the safer profile and also has PPS (dynamic voltage) and EPR (>100W modes) disabled. 

There is no "100W" mode. At 20V, you are using 20V/6.2 ohms = 3.22A. Which is 3.22A*20V = 65W.

Totoro-spec, 2025-02-26:
@Ralim 

~470sec

colonel-lp, 2025-02-26:
@Ralim sorry late to the party, your last build looks like its working fine re screen & BLE. great work.

Ralim, 2025-02-27:
@Totoro-spec 
Please break this out into a new issue. I dont think this is related to this issue, this sounds more like a power supply/cable incompatibility

PR #2049 (2025-02-27): Rev2 reworking I2C to avoid IRQ stalling && adjusting thermal runaway for PinecilV2

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [?] There are no breaking changes

* **What kind of change does this PR introduce?**

This fixes #2047.
This speeds up I2C.
This improves I2C reliability on PinecilV2
This also updates FreeRTOS, and alpine (due to merging branches).

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:
closed 2025-02-27 by Ralim