Ralim/IronOS #2051: Issue with flashing TS80P with DFU 3.51 ?

miodzicho, 2025-01-15:
It seems I cannot flash TS80P with DFU 3.51, I have tried with two PC's , and always getting .ERR after copying hex file into iron.
Anyone can add something to this issue? Any workaround ?

ia, 2025-01-15:
Hello.

> I have tried with two PC's

What operating systems did you use on these PCs exactly?

It seems recently there is some problem with _Miniware_'s bootloaders, since we have too many reports of the same kind. Please, read very carefully [this comment](https://github.com/pine64/blisp/pull/73#pullrequestreview-2541320237) and try to follow advises from there.

Unfortunately, there is not much we can do about built-in _DFU_ bootloader, because we do not have any relation to that.

miodzicho, 2025-01-15:
It was Windows 11 on both.

ia, 2025-01-16:
> It was Windows 11 on both.

Do you have a linux box nearby? If yes, then I would try flashing on linux.

> BTW, did you attached correct link ? (Fix missing simpleicons-based windows logo by standalone version [#73](https://github.com/Ralim/IronOS/issues/73))

Ooops, sorry - I did mess with my copy-paste buffer between tabs in a browser. [Here is the correct link](https://github.com/Ralim/IronOS/issues/1933#issuecomment-2564140310).

cliobrando, 2025-01-17:
I had the same issue with a TS101, you can use a Linux virtual machine to flash the device, under windows (even different pcs)  I had 0% success rate, under Linux works almost all the time.
https://github.com/Ralim/IronOS/issues/2014#issuecomment-2597606332