Ralim/IronOS #2063: Corrupted display on TS101

showier-drastic, 2025-01-31:
**Describe the bug**

After flashing 2.23-RC1 on a newly-purchased TS101, the display is not working properly.

**To Reproduce**
1. Flash the firmware on it

**Expected behavior**
Displays correctly

**Details of your device:**
 
 - Device: TS101
 - Release: 2.23-RC1
 - Power adapter being used: Irrelevant, issue exists regardless of adapter
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

ia, 2025-02-01:
> After flashing 2.23-RC1 on a newly-purchased TS101, the display is not working properly.

Could you, please, honestly remember: did you use the device with the original _Miniware_ firmware **before** flashing _IronOS_?

> Power adapter being used: Irrelevant, issue exists regardless of adapter

No, it's not irrelevant. Let this decide to people who may have time & knowledge to help without telepathic skills. ;)

Although some details during report may seem insignificant, for any good bug report the more information you provide, the better (even if it won't be needed in the end).

So... could you tell, please, what power adapter did you use _exactly_ **right** before it happened? Did you attach power cable over _**DC barrel connector**_ or over _**USB-C port**_?

showier-drastic, 2025-02-01:
@ia Thanks for the tips.

> > After flashing 2.23-RC1 on a newly-purchased TS101, the display is not working properly.
> 
> Could you, please, honestly remember: did you use the device with the original _Miniware_ firmware **before** flashing _IronOS_?

Yes, and the original firmware works normally. I think this might indicate some hardware revision that changes the screen model?

> No, it's not irrelevant. Let this decide to people who may have time & knowledge to help without telepathic skills. ;)
>
> Although some details during report may seem insignificant, for any good bug [...]

I've tried all possible options (DC connector, dumb 5V USB-A to USB-C, and 30W Apple Adapter that comes with my MacBook), and they all have the same issue. Therefore I say that it's irrelevant.

ia, 2025-02-01:
> I've tried all possible options (DC connector

I had to ask, because recently there was a [very similar comment](https://github.com/Ralim/IronOS/issues/583#issuecomment-2615434966). Here is [the statement that it did happen only because of unlucky power source through DC barrel](https://github.com/Ralim/IronOS/issues/583#issuecomment-2618461839). I thought it may be related.

> Also it happens right after I flashed the firmware

Could you try to flash the original firmware back? Even if screen doesn't work, if it's only OLED issue with _IronOS_, then you could try go to _DFU mode_ and flash the _Miniware_ firmware to see if the screen works again.

oliverpool, 2025-02-01:
I also purchased a TS101 recently.
The stock firmware worked fine (forgot to write down the version), DFU indicates 1.05.

I upgraded first to v2.22 without problems (the display was small, as expected).
Then I upgraded to v2.23-rc1 without problem (display is now fullscreen), when connected to my computer (05) or to a proper USB adapter (21).

(it does not likely help the OP, since everything seems to work fine for me - in my case only a big "thank you" to the maintainers for the great work!)

showier-drastic, 2025-02-02:
@ia Sorry for the late reply.

Just flashed back to original firmware (TS101AppV210.zip). It works without issue, no display corruptions. All function works properly.

ia, 2025-02-02:
> Just flashed back to original firmware (TS101AppV210.zip). It works without issue, no display corruptions. All function works properly.

So we just eliminated the theory about OLED corruption by bad power source, in your case. I'm sincerely happy to know that _physically_ your screen works and that it hasn't been damaged.

Now, could you, please, try the [stable release](https://github.com/Ralim/IronOS/releases/download/v2.22/TS101.zip) of _v2.22_ build on your _TS101_ and report results back? Thanks.

showier-drastic, 2025-02-02:
It is still corrupted with v2.22.

https://github.com/user-attachments/assets/f81c1806-8dab-40e2-b4b6-0e3553ec9fb7

oliverpool, 2025-02-02:
Stupid question: which file did you put on the device? (language and extension)

showier-drastic, 2025-02-02:
> Stupid question: which file did you put on the device? (language and extension)

TS101_EN.hex

oliverpool, 2025-02-02:
ts101_en.hex sounds good. 

When flashing, what is displayed as DFU version?
You get the .rdy file after flashing, right?
(sorry if those questions sound stupid, but I am looking to find any visible difference between our devices)

showier-drastic, 2025-02-02:
> ts101_en.hex sounds good.
> 
> When flashing, what is displayed as DFU version?

DFU:1.06

> You get the .rdy file after flashing, right?

Yes.

> (sorry if those questions sound stupid, but I am looking to find any visible difference between our devices)

No these are not stupid questions; I very appreciate your help.

ia, 2025-02-02:
@Ralim, could you look at this when you have time, please?

> I think this might indicate some hardware revision that changes the screen model?

Probably this is the most reasonable assumption so far: _it seems_ _Miniware_ did start to change _OLED_ screen model in _TS101_, so hardware-related code to control the screen doesn't work properly anymore.

And, @Ralim, is there any way to get OLED model like we do it with accelerometers, for example?

@showier-drastic, I'm not familiar with a hardware construction & case of _TS101_, but, **at your own risk**, you could very carefully disassemble your _TS101_ to see if there is any ID on or around OLED screen (to get the exact model). Another suggestion (a bit crazy though) - you could play a social engineering card a bit: write to _Miniware_ official support email/contact, tell them that you damaged OLED, but you did replace it yourself by re-soldering, but now it doesn't work (you could attach photo that you showed us here) and ask them something like: "could you, tell me, please, which OLED screen exactly (manufacturer/model/id) I should buy to replace it successfully?" (believe it or not, but _Miniware_ is very unfriendly about sharing any information about their devices & hardware willingfully).

showier-drastic, 2025-02-03:
> [@Ralim](https://github.com/Ralim), could you look at this when you have time, please?
>
>
> Probably this is the most reasonable assumption so far: _it seems_ _Miniware_ did start to change _OLED_ [...]

Is it possible that we can try some code with similar OLED models? I'm willing to collaborate on trying to flash and debug.

> And, [@Ralim](https://github.com/Ralim), is there any way to get OLED model like we do it with accelerometers, for example?
>
> [@showier-drastic](https://github.com/showier-drastic), I'm not familiar [...]

Unfortunately it seems that it's impossible to see the screen model without doing damage to the iron: https://www.ifixit.com/Guide/Disassembling+Miniware+TS101+soldering+iron/169989

Maybe I can try some social engineering as you mentioned, let's see if that works out.

showier-drastic, 2025-02-03:
Here's an internal picture of the iron, in case that helps:

[image]

ia, 2025-02-03:
> Unfortunately it seems that it's impossible to see the screen model without doing damage to the iron: https://www.ifixit.com/Guide/Disassembling+Miniware+TS101+soldering+iron/169989

But what about [this _TS101_ disassemble howto](https://github.com/Ralim/IronOS/issues/1420#issuecomment-1574356410)? It looks _"less damaging"_ as far as I understand, and as the title says. Sorry, I'm shooting in the dark here, since I don't own _TS101_ myself, but still trying to help with everything I can.

Oh, and another question: where did you get your _TS101_? Is there at least smallest chance that it may be some kind of clone, but not originally produced by _Miniware_?

showier-drastic, 2025-02-04:
> But what about [this _TS101_ disassemble howto](https://github.com/Ralim/IronOS/issues/1420#issuecomment-1574356410)? It looks _"less damaging"_ as far as I understand, and as the title says. Sorry, [...]

Thanks for this, but I do not feel that I can remove the glass without breaking it, so unless it's absolutely necessary, I prefer not to try this way.

To me the screen issue sounds like it might be resolved by tweaking OLED registers, I will probably try to debug the firmware when I have some time.

> 
> Oh, and another question: where did you get your _TS101_? Is there at least smallest chance that it may be some kind of clone, but not originally produced by _Miniware_?

On a taobao shop (购百应科技). It seems to have a high sales volume.

ia, 2025-02-04:
> I do not feel that I can remove the glass without breaking it, so unless it's absolutely necessary, I prefer not to try this way.

No problem, thanks for checking out anyway.

> it might be resolved by tweaking OLED registers

Yeah, as far as I understand, this is the exact root cause - if they changed hardware revision of _OLED model_, then the calls to support new model should be added as well. However, without knowing the exact model and its datasheet, there is not so much we can do, I guess. :(

Oh, I have another theory: they could keep the same _OLED_ model, but **may change** _PCB design_ a bit with modification to traces between, let's say, _MCU_ and _OLED_, so in this case a new pinout _"map"_ _could be a good start_. Visual inspection (even with cheapest usb microscope) and multimeter (in signal tracing mode with very thin probes) are the best friends here (take-out of the main PCB from the main shell is still required though, and **there is no any guarantee that even that will help right away**, so just leaving this advice for the bravest ones).

Another question! If you have opportunity and mood to keep experiments with (re-)flashing, or maybe you have this information from the previous _run_ of _IronOS_ on your device: even without _OLED_ displaying info properly, did you notice that _IronOS_ **works** in general, anyway? I.e., if you flash _IronOS_, and on the first power-on you press `+/A` button to go to _Soldering Mode_ & moving the tip to the solder, does the tip melt the solder? And does the tip cool off after long `-/B` press? Just use measures of precaution, please, since there is no any feedback from the screen, if you decide to check.

> On a taobao shop (购百应科技). It seems to have a high sales volume.

Just one more thing on that - what was the exact date of you placing the order? And date of you receiving your _TS101_? This may help, if we will start to receive the similar reports _soon_.

P.S. Thank you for keeping in touch, it's a rare thing around here in the issue tracker. But if you don't mind, I would like to update the title since we established this as the fact that luckily your OLED **is not corrupted**. :)

showier-drastic, 2025-02-04:
> Oh, I have another theory: they could keep the same _OLED_ model, but **may change** _PCB design_ a bit with modification to traces between, let's say, _MCU_ and _OLED_, so in this case a new pinout [...]

A quick find is that the PCB revision number of mine and whitequarks' (in your disassemble guide) are the same (V1.51A). So the possibility of this should be somewhat low.

> Another question! If you have opportunity and mood to keep experiments with (re-)flashing, or maybe you have this information from the previous _run_ of _IronOS_ on your device: even without _OLED_ [...]

Roughly 3 weeks ago.

> 
> P.S. Thank you for keeping in touch, it's a rare thing around here in the issue tracker. But if you don't mind, I would like to update the title since we established this as the fact that luckily your OLED **is not corrupted**. :)

Thanks for your help too. It's very important to get in touch.

showier-drastic, 2025-02-04:
The rest of IronOS's functionality seems to be working. If I press the + button, the tip heats up, and I can tell that the temperature numbers are changing from the corrupted image. If I press the - button, some animation happens. Screen rotation works too, as shown in my video.

ia, 2025-02-04:
> A quick find is that the PCB revision number of mine and whitequarks' (in your disassemble guide) are the same (V1.51A).

Good catch. Didn't pay attention to this myself.

> So the possibility of this should be somewhat low.

Yes... unless they did it without renaming the version on the PCB. Not sure how much this probable though. :/

> It's very important to get in touch.

Although at this point I don't have much to comment/provide anymore, unfortunately. :(

Ralim, 2025-02-04:
Hia,

This definitely screams of a bad OLED init sequence.
Are you able to build IronOS yourself?

It's most likely one of these : https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/Drivers/OLED.cpp#L36

I would be suspect its lines 39-42 (inclusive); as those do change with different clones of the OLED's.

If your able to build locally, changing those +-1 to see results can often be telling?

resistancelion, 2025-02-06:
> I had to ask, because recently there was a [very similar comment](https://github.com/Ralim/IronOS/issues/583#issuecomment-2615434966). Here is [the statement that it did happen only because of [...]

Yeah, but that was due to constant voltage fluctuation (got a PSU that trades voltage for amps, not the opposite as it should...)

> So we just eliminated the theory about OLED corruption by bad power source, in your case. I'm sincerely happy to know that _physically_ your screen works and that it hasn't been damaged.
>
> Now, could [...]

This thing with PSU and voltage spikes provokes only converter part of the circuit to fail, in my case it was the same capacitor and transistor over and over again, i have no idea why something else didn't fail on PCB, but ok

oliverpool, 2025-02-07:
In the meantime, maybe an hint could be added to the README in the "Miniware TS101" notes, linking to this issue:

> [OLED issues with DFU:1.06](https://github.com/Ralim/IronOS/issues/2063)

showier-drastic, 2025-02-24:
Just to add a note here, when I am trying to debug my TS101 with SWD today (I want to extract the correct OLED sequence by reverse-engineer their DFU firmware), I found that the MCU is secured, so the DFU firmware cannot be read out. Obviously they do not want us to read it... It should still be possible, though, through a modified App, but will be much more difficult to do. Sad.

[image]

showier-drastic, 2025-02-24:
Based on some experiments, I think the main difference of the new unknown OLED model and the good-old SSD1307 are the following:

* Some offset difference (so that the current initialization code needs some tweaks that I'm unable to figure out)
* Most importantly, the new unknown OLED model does not support "Horizontal addressing mode" (register 20h => A[1:0]=00b) that we use now. After transmission of one page, it always goes back to the same page, instead of going to the next page. So we cannot send the whole screen at once. Frame buffer need to be sent page-by-page.

I did some reverse-engineering of the official App V210. The official App does not initialize the OLED screen at all; it relies on the bootloader to set up the screen. In the app, it just directly writes data to the OLED.

Based on the philosophy of doing things in the same way as the official firmware, I created a dirty experimental version (functions such as screen rotation won't work for now):

https://github.com/showier-drastic/IronOS/tree/oled-experiment-ts101

[TS101_EN-2.hex.zip](https://github.com/user-attachments/files/18939333/TS101_EN-2.hex.zip)

The main changes are:

1. Remove anything related to the OLED reset pin (PA7)
2. Removed OLED init code
3. Rewrote the `refresh` method, based on function 0x0800db54 (I named it `oled_bulk_write` in my code) in official App.
4. Temporarily removed other places that touches the OLED config.

[image]

Please test if this works on your TS101 (especially with the "old" OLED model). If it works both on the old and new version, I think we can continue to see how to improve it, and finally integrate to the main code base.

showier-drastic, 2025-03-02:
Just a humble request here that someone can test my firmware build, and if it works on older TS101, I will try to optimize the code quality.

paulfertser, 2025-03-02:
It's great progress indeed, thank you for sharing!

My personal opinion though is that IronOS should rather not depend on hardware state inherited from bootloader, especially when it's miniware trash code.

XaocuHKa, 2025-03-02:
Hey, thanks a lot for your help!

I have an issue after installing you FW branch on the new OLED, has some artifacts on the right side of the screen. Is that normal>

[image]

Luckily I have an old TS101 with broken heating element, so I can test the screen, but not heating. So, just installed your FW and it gets stuck in "calibrating". In PD Debug mode it says State 3 No VBus. Not sure if that helps.
Please let me know if there is anything I can do to test it further?

showier-drastic, 2025-03-02:
> Hey, thanks a lot for your help!

Thanks for your testing!

> 
> I have an issue after installing you FW branch on the new OLED, has some artifacts on the right side of the screen. Is that normal>
> 
> [image]

This is normal, as I did not implement the code related to the scrollbar area.

> 
> Luckily I have an old TS101 with broken heating element, so I can test the screen, but not heating. So, just installed your FW and it gets stuck in "calibrating".  In PD Debug mode it says State 3 No VBus. Not sure if that helps. Please let me know if there is anything I can do to test it further?

Is the screen working? If yes, it might be unrelated to my change.

showier-drastic, 2025-03-02:
> It's great progress indeed, thank you for sharing!
> 
> My personal opinion though is that IronOS should rather not depend on hardware state inherited from bootloader, especially when it's miniware trash code.

I would say if we can get it to work, it does not matter much. Or we can keep two ways to initialize the screen? On old hardware we use the old way, on new hardware we use the new way. I'm not sure.

paulfertser, 2025-03-03:
> I would say if we can get it to work, it does not matter much. 

Until someone finally adds support for proper DFU to https://github.com/Ralim/IronOS-dfu which should be actually quite easy and provide numerous advantages.

I do not have TS101 at hand for testing or else I'd just connected SWD and tried that...

> Or we can keep two ways to initialize the screen? On old hardware we use the old way, on new hardware we use the new way. I'm not sure.

Not a clean solution IMHO; usually it's not too hard to find a suitable initialisation sequence for any display. I wonder if it really lacks any markings at all?

Again, not my call to judge here, I'm talking from a potential user point of view.

oliverpool, 2025-03-06:
> https://github.com/showier-drastic/IronOS/tree/oled-experiment-ts101
> 
> [TS101_EN-2.hex.zip](https://github.com/user-attachments/files/18939333/TS101_EN-2.hex.zip)

I did flash this on my "older" device (DFU 1.05) and the screen looked good (didn't test heating).

(the 2nd line of the debug menu read 20250224)

showier-drastic, 2025-03-06:
To @oliverpool and @XaocuHKa :

Thanks for testing! It seems that this approach works. I'll try to add all the missing functionality to my code to make it really usable.

To @paulfertser:

> > I would say if we can get it to work, it does not matter much.
> 
> Until someone finally adds support for proper DFU to https://github.com/Ralim/IronOS-dfu which should be actually quite easy and provide numerous advantages.

With, or without, the potential IronOS-dfu support, I would say it's okay to rely on the DFU to initialize the OLED.

> 
> I do not have TS101 at hand for testing or else I'd just connected SWD and tried that...

The device is locked. You cannot debug anything (even dumping the DFU firmware is impossible via SWD), until you erase the whole device, which I would prefer not to do right now.

> 
> > Or we can keep two ways to initialize the screen? On old hardware we use the old way, on new hardware we use the new way. I'm not sure.
> 
> Not a clean solution IMHO; usually it's not too hard to find a suitable initialisation sequence for any display. I wonder if it really lacks any markings at all?

It will be possible to find the sequence via reverse engineering, if anyone can dump the DFU firmware. I'm happy to do the reverse-engineering of DFU (I've already reverse-engineered the App part, so it should be easy to do it again on DFU). Dumping the DFU firmware should be possible utilizing the USB device peripheral of the MCU, but I currently do not have time to write this dumping part.

> 
> Again, not my call to judge here, I'm talking from a potential user point of view.

Thanks for your comment!

paulfertser, 2025-03-06:
> It will be possible to find the sequence via reverse engineering, if anyone can
> dump the DFU firmware. I'm happy to do the reverse-engineering of DFU (I've
> already reverse-engineered the App part, [...]

I think it's essentially already written in the form of that
IronOS-DFU code, and it supports running from the offset normally used
for "application", so it's safe to experiment with it by just
uploading it via the vendor bootloader and trying to use for reading
the flash out I guess. From a cursory look it seems like it's really
a minimal modification needed to get it running on TS101, I just do
not have a device for testing that.

I do not feel like display init belongs to a bootloader, it might be
probably better to have it than not but I do not see much value in
doing it there either, hence my idea about IronOS display init being
independent.

Woodsnaps, 2025-03-31:
Very wild take, but! I think I might have found the issue:

When I used stock firmware (straight from the factory), I was able to change DisplayDir to `Left`. All worked fine. Then, I uploaded a new firmware via my macbook. After updating I seemed to get a black screen after the boot image. Very strange.
Then I tried reverting back to v1.11 and v2.01: issue persisted.
When changing the DisplayDir back to `Right` via the config file, the TS101 boots up normally. When changing back to "Left": issue returns.
Then I tried installing 2.23 IronOS and I get the exact same issue as @showier-drastic 

Hypothesis: the macbook places the .Trashes, .fseventd and ._.Trashes on the Volume which are impossible to get rid of. Not via Windows, not via macOS.
Due to these files being present, the memory gets issues and can't initialize booting properly.

Can anyone confirm with a Windows PC (& having the TS101 never connected to macOS) that these files are not present?

330Speedy, 2025-04-27:
> Based on some experiments, I think the main difference of the new unknown OLED model and the good-old SSD1307 are the following:
> 
> * Some offset difference (so that the current initialization code needs some tweaks that I'm unable to figure out)
> * Most importantly, the new unknown OLED model does not support "Horizontal addressing mode" (register 20h => A[1:0]=00b) that we use now. After transmission of one page, it always goes back to the same page, instead of going to the next page. So we cannot send the whole screen at once. Frame buffer need to be sent page-by-page.
> 
> I did some reverse-engineering of the official App V210. The official App does not initialize the OLED screen at all; it relies on the bootloader to set up the screen. In the app, it just directly writes data to the OLED.
> 
> Based on the philosophy of doing things in the same way as the official firmware, I created a dirty experimental version (functions such as screen rotation won't work for now):
> 
> https://github.com/showier-drastic/IronOS/tree/oled-experiment-ts101
> 
> [TS101_EN-2.hex.zip](https://github.com/user-attachments/files/18939333/TS101_EN-2.hex.zip)
> 
> The main changes are:
> 
> 1. Remove anything related to the OLED reset pin (PA7)
> 2. Removed OLED init code
> 3. Rewrote the `refresh` method, based on function 0x0800db54 (I named it `oled_bulk_write` in my code) in official App.
> 4. Temporarily removed other places that touches the OLED config.
> 
> <img alt="Image" width="592" src="https://private-user-images.githubusercontent.com/168159404/416130516-b38950b8-0637-4d52-b23c-451b6849aac2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU3NjMwMTMsIm5iZiI6MTc0NTc2MjcxMywicGF0aCI6Ii8xNjgxNTk0MDQvNDE2MTMwNTE2LWIzODk1MGI4LTA2MzctNGQ1Mi1iMjNjLTQ1MWI2ODQ5YWFjMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQyN1QxNDA1MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNTVlYjhmMzEwYzBmMmVjZWYzZGEzODQwMzEwZmI5NjJhYzE0MTgxNjkyNzUwMTEyNTY5MGUzZDdmNTAyNTkyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hREzKy3yi9XyWb566dmPCLDaiBdkt332sXv7y_LpL1M">
> Please test if this works on your TS101 (especially with the "old" OLED model). If it works both on the old and new version, I think we can continue to see how to improve it, and finally integrate to the main code base.

> Based on some experiments, I think the main difference of the new unknown OLED model and the good-old SSD1307 are the following:
> 
> * Some offset difference (so that the current initialization code needs some tweaks that I'm unable to figure out)
> * Most importantly, the new unknown OLED model does not support "Horizontal addressing mode" (register 20h => A[1:0]=00b) that we use now. After transmission of one page, it always goes back to the same page, instead of going to the next page. So we cannot send the whole screen at once. Frame buffer need to be sent page-by-page.
> 
> I did some reverse-engineering of the official App V210. The official App does not initialize the OLED screen at all; it relies on the bootloader to set up the screen. In the app, it just directly writes data to the OLED.
> 
> Based on the philosophy of doing things in the same way as the official firmware, I created a dirty experimental version (functions such as screen rotation won't work for now):
> 
> https://github.com/showier-drastic/IronOS/tree/oled-experiment-ts101
> 
> [TS101_EN-2.hex.zip](https://github.com/user-attachments/files/18939333/TS101_EN-2.hex.zip)
> 
> The main changes are:
> 
> 1. Remove anything related to the OLED reset pin (PA7)
> 2. Removed OLED init code
> 3. Rewrote the `refresh` method, based on function 0x0800db54 (I named it `oled_bulk_write` in my code) in official App.
> 4. Temporarily removed other places that touches the OLED config.
> 
> <img alt="Image" width="592" src="https://private-user-images.githubusercontent.com/168159404/416130516-b38950b8-0637-4d52-b23c-451b6849aac2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU3NjMxMjcsIm5iZiI6MTc0NTc2MjgyNywicGF0aCI6Ii8xNjgxNTk0MDQvNDE2MTMwNTE2LWIzODk1MGI4LTA2MzctNGQ1Mi1iMjNjLTQ1MWI2ODQ5YWFjMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDQyN1QxNDA3MDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NzlhODc4MzE4ZjIyYmI0Yzk2MzczYWZlNDJjMjliZjg3Y2Y1ODFhOWI4NjNkOTM1MDg5ZDBlYjQwZmJiN2QxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.iu7XjZdaYeSq40lRuQbbqF3G6xCEe89owUXTC4Cb_6g">
> Please test if this works on your TS101 (especially with the "old" OLED model). If it works both on the old and new version, I think we can continue to see how to improve it, and finally integrate to the main code base.

Hi everyone.
This is perfect news!
I have exactly the same problem with my TS101 and this modified version TS101_EN-2.hex.zip solved the problem perfectly. But I have a huge question. Could someone help me compile exactly this firmware version, only in a different language version? (TS101_CS) I am not familiar with these completely unknown things like compiling firmware xD
Thank you all in advance for your help :)

ttelford, 2025-05-02:
> Based on the philosophy of doing things in the same way as the official firmware, I created a dirty experimental version (functions such as screen rotation won't work for now): [...]

I have an old(ish) TS101 purchased in late 2023: The OLED works! (It doesn't swap orientations, but I believe you mentioned it wouldn't/it wasn't a goal of the test).

showier-drastic, 2025-05-03:
Thanks for all your testing! I'll probably find some time to integrate this method into the current code base. I'd like to do as this approach: use a macro to check whether it's TS101. if it is, then use the very minimal initialization code, and use "new way" to draw on screen; if it is not TS101, then use old code.

@ia @Ralim I'd like to hear about your opinion on the above; if it sounds OK, I'll start do it.

Ralim, 2025-05-04:
_I_ would vote against not initialising the display.

My logic here is that this will break things potentially when using other bootloaders if they don't initialise the display before the main firmware starts. And this can be a nightmare to debug for the user if they dont understand. 

Additionally when I'm debugging issues on devices I normally dont run the bootloader (but just jump in directly with SWD debugger), so this would be annoying. Though I don't really support the TS101 so 🤷🏼.

I _prefer_ to just reverse engineer the bootloader (I can send you this if you dont have it). As its more likely this means that we are missing a step (wrong oled reset pin? an enable pin?).

However having said that, I _am_ ok with accepting this as a patch _for now_ but I wouldn't want it to stick around forever if possible.

MyHealthSheets, 2025-05-04:
Well, I just spent about the last 6hrs with what sounds like the same issue on a new TS101 I got earlier today :( . This thread looks like it came in pretty handy...  

My device was working fine on the stock Miniware f/w, but of course having the device for a couple hours already, figured it was time to try out ironOS. After my first successful flash of IronOS, my OLED got scrambled. However, in my case, regardless of what f/w I tried flashing after (stock, ironOS v2.23rc4, v2.22 and even your recent patch), I couldn't get my OLED to display correctly. 

I was about to give up, figured I could return my iron to Amazon, claim defective, but oddly I saw the comment a few above about macOS files getting added to the mounted 'TS101_DFU' drive. I honestly didn't think much of it, and still don't exactly see how these stupid auto MacOS files could cause any issues, but, I kinda think they did!? 

While my DCU was mounted, I deleted to folders from my Terminal (rm -R) '.Trashes' and '.fseventsd'. After, I simply flashed the stock f/w again, and all of a sudden, my OLED was working! After, and without doing anything else, I quickly flashed the ironOS v2.23rc4, and my OLED was still working....

From there, I flashed a few more back and forth, and eventually my OLED get messed up again. I tested the theory and again deleted both folders from DCU mount, and after flashing both f/w's again, the OLED was working #shrug !?

Anyway, I think I can probably reproduce it again if you want any more info. Just as a heads up, as I was figuring out how to flash the f/w to begin with, I was going back and forth between a Ubuntu Server and my Macbook. It just happened that I was on my Mac when reading through this thread regarding the "extra" folders. I still don't quite get how they could have caused issues, I'd like to understand that better. It's really late for me right now, but perhaps I can take another look in the next couple days.

Thank you all though for this thread, as a Software Support Engineer by trade, this was nice to follow!

Unb0rn, 2025-05-24:
Hey! Is there any chance it will be merged in final 2.23? Have the same issue and in IronOS DFU too (it's not that critical in DFU, yet it's nice to have a working screen)

Woodsnaps, 2025-06-18:
@edlf it's useless trying to flash the original bootloader if the AppleDoubles are present. You need to get rid of those first.

Woodsnaps, 2025-06-19:
@edlf if you've managed to get a clean bootloader, could you hit me up? I'd love to have that as well❤️

Woodsnaps, 2025-06-19:
@edlf not all heroes wear capes!

edlf, 2025-06-23:
@Woodsnaps check your email, I've dumped the bootloader and fixed my other iron.

Woodsnaps, 2025-06-23:
Thank you! I will try to get to this during the week:D

0Karakurt0, 2025-07-17:
> I deleted to folders from my Terminal (rm -R)

That's an easy one to add, to a flashing script, I'll do that right now.

0Karakurt0, 2025-07-17:
Now this is curios... Removing anything from mounted config disk before writing a new firmware led to failing flashing :/
On Linux!

If I'm not touching anything, I can consistently flash firmware on the second try.

Whitelist, maybe..?
On my iron I see those files before flashing:
```
 .   ..   .fseventsd  'System Volume Information'   ._.Trashes   .Trashes
```

josefdahari, 2025-08-04:
So am I assuming that the ts101 cannot use the IronOS firmware and not to attempt it ?    I was trying to read through the previous conversations and I have a brand new iron I wanted to flash since I like to use iron OS but from what I gather it will screw up the OLED Is that the case?

Woodsnaps, 2025-08-05:
> [@zeroClearAmerican](https://github.com/zeroClearAmerican)  I need to find the stock hex hopefully.

If you find a stock hex, could you please share it with me as well? :)

zeroClearAmerican, 2025-08-05:
The IronOS documentation for the TS100 has the link to the official backup firmware listed on [this page](https://ralim.github.io/IronOS/Flashing/TS100/) in case something goes wrong. This also includes the TS101 stock firmware. I'll copy the link here as well:

https://e-design.com.cn/en/NewsDetail/4203645.html

josefdahari, 2025-08-05:
> The IronOS documentation for the TS100 has the link to the official backup firmware listed on [this page](https://ralim.github.io/IronOS/Flashing/TS100/) in case something goes wrong. This also includes the TS101 stock firmware. I'll add the relevant quote from the docs below:
> 
> > This is completely safe, but if it goes wrong just put the corresponding .hex file from [the official website](https://e-design.com.cn/en/NewsDetail/4203645.html) ([mirror backup](https://github.com/Ralim/IronOS-Meta/tree/main/Firmware/Miniware)) onto the unit and you're back to the old firmware.

I appreciate you dude.   I'm going to try and flash a new  iron or 2 later with the OLED firmware and hopefully all goes well 🙂    wish me luck 🤞

josefdahari, 2025-08-12:
So the firmware for IronOS flashed and working but it's not working on any of my PD power banks or bricks.    The original does.  Am I missing something ?

josefdahari, 2025-08-12:
> [@josefdahari](https://github.com/josefdahari) I was able to use the build from [@showier-drastic](https://github.com/showier-drastic)'s reply [here](https://github.com/Ralim/IronOS/issues/2063#issuecomment-2677874723) for mine, and I even added a custom logo to it using the instructions in the wiki. It works great but doesn't appear to have been merged.
> 
> [@showier-drastic](https://github.com/showier-drastic) what's the over-under on getting this fix merged into the main branch for a release?

Have you got your iron working on all PD power bank or power bricks.  I just get a on and off type of behavior. Starts up and then resets

josefdahari, 2025-08-12:
Ok so I tried a 65w brick and it works well. But my 100w brick and my anker pd bank just reset it.  Is this a settings issue or firmware issue

zeroClearAmerican, 2025-08-13:
> Have you got your iron working on all PD power bank or power bricks. I just get a on and off type of behavior. Starts up and then resets

I've used a few different types of USB-C sources to power it, with success in all 3. Two different bricks, one power bank.

OfficialAudite, 2025-08-28:
I have a weird issue, when uploading the firmware it did it on the second upload. The first just made the .ERR. Now I'm trying to restore to original one and there is big issue. When i upload the second time nothing happens it just stays with .err and .hex file and nothing happens...

josefdahari, 2025-08-28:
> I have a weird issue, when uploading the firmware it did it on the second upload. The first just made the .ERR. Now I'm trying to restore to original one and there is big issue. When i upload the second time nothing happens it just stays with .err and .hex file and nothing happens...

This also happened to me on the first upload.  Are you sure you're actually extracting the file and not using the zip that possibly downloaded?   I don't know what you're on Windows or Mac but it happened to me the first time I tried to do it. Actually extracted the file somewhere else and then re flashed and it worked

OfficialAudite, 2025-08-28:
> This also happened to me on the first upload. Are you sure you're actually extracting the file and not using the zip that possibly downloaded? I don't know what you're on Windows or Mac but it [...]

I have the extracted file on desktop, the hex itself. So it should work. But yet first time .err which happened to me too when flashing the oled one. And the second time the oled one flashed but the original from official website just stays there, while nothing is happening.

karageorg, 2025-08-30:
Hi everyone!
Yesterday, I received my new TS101 soldering iron and, of course, decided to update the firmware to IronOS 2.22. As a result, the display does not work, as shown at the beginning of this thread. After many attempts to restore the display, I found a way to flash the correct firmware and restore the display.

**Details:**
The basic firmware that came with the soldering iron is `2.11`
The DFU version is `1.06`
Operating system: `Ubuntu 24.04`
For the disk to work correctly, it must be mounted with the following command as root:
`mount /dev/sd? /mnt/ts101 -t msdos`
After that, make sure that the ddrescue command is present in the system. You can do this by typing ddre and pressing TAB several times in `Bash`, `Zsh`, etc.
If the command is not present in the system, install it using apt, dnf, yum, or another command, depending on your OS.
Once the command appears in the system, run:
`ddrescue -D ./TS101AppV211.hex /mnt/ts101/ts211.hex`
After a few seconds, the previously mounted disk will disconnect. Mount it back and check which file appeared (.err or .rdy).
In my case, the first time the file had the .err extension, but after repeating the command:
`ddrescue -D ./TS101AppV211.hex /mnt/ts101/ts211.hex`
The disk disconnected, and after remounting it, I see the following lines in the console
```
...
drwxr-xr-x 2 root root    512 nov 25  2014 trashe~1
-rwxr-xr-x 1 root root 125162 aug 30 03:33 ts211.err
-rwxr-xr-x 1 root root 125162 aug 30  2025 ts211.rdy
```
Okay, after that, I was able to restore the original soldering iron firmware and got the display working again.
I hope this helps someone in the future.

[16 more replies]