Ralim/IronOS #2069: Pinecil V2 Kills my Anker Nano Charger 100W PD Powersupply

Pukesmiley, 2025-02-03:
My version is v2.23E.DD4A5500 2024-12-22
IDD f42d01fb 0000e1b142cf0eb4
every other charger I try works fine, 100W Powerbanks, 45W PD and 12V barreljack
The first charger died before updating and the second one, that anker luckily replaced died after updating.
They work fine for a couple minutes, heating, cooling down unplugging while hot, replugging, but after leaving the pinecil on the stand and while Im away something happens that kills the charger. I cant even get 5V out of them.
A

ia, 2025-02-03:
> The first charger died before updating and the second one, that anker luckily replaced died after updating.

Could you, please, provide the exact & full model name of the charger? You may include even the direct link to the charger page. Very curious to see _the exact specifications & supported power protocols_.

> They work fine for a couple minutes, heating, cooling down unplugging while hot, replugging, but after leaving the pinecil on the stand and while Im away something happens that kills the charger

Did you leave _Pinecil_ two times in a row unattended with power supply, even though the first one died? Was there something after all, maybe, that you could see on the display of your _Pinecil_ right before the second charger died?

However, unfortunately, it will be probably nearly impossible to detect the root cause. Not without capturing USB PD traffic between _Pinecil_ and the charger in question, which means burning at least one more with high probability. :(

Pukesmiley, 2025-02-23:
The model of the charger is [B2679, Anker Nano Charger (100W)](https://www.anker.com/eu-de/products/b2679-nano-100w-usb-c-ladegeraet)-.
yes, unfortunately and I dont know why I did it, I left it a second time without looking at it(the third one anker send me is still unpacked). How would you suggest capturing USB PD traffic? Do I need a special device for this, is there a proven DIY version?

Ralim, 2025-02-28:
You will need a custom device to capture the traffic sadly.

Thinking more on this, I dont think its going to be a PD issue to brick the supply as its unlikely PPS is being used, so its likely the PD traffic just stops after negotiation.

The only other cause I can think of is the very short pulses of power often used to maintain a temperature are somehow causing issues with the supply :(

20andredias02, 2025-03-04:
I’d like to add my case to this issue. I have an Anker Prime 67W GaN charger (3 ports) (model A2669) and experienced a similar problem. It was working fine until, during use, I got a "thermal runaway" message. After that, whenever I tried to draw more than 20W, the charger would restart (pinecil and other devices).

I sent the charger for RMA, and with the replacement unit, after some testing, I encountered the same issue (this time no thermal runaway message). Now, it fails when drawing more than ~35W (I haven’t tested extensively this time), causing it to restart—not just with the Pinecil, but with other devices as well.

20andredias02, 2025-03-07:
> Your pinecil still working afterwards ? I mean it's only the charger that breaks ?

Yeah, the pinecil still works perfectly, it's just the charger that failed.

GreyXor, 2025-03-10:
> > Your pinecil still working afterwards ? I mean it's only the charger that breaks ?
> 
> Yeah, the pinecil still works perfectly, it's just the charger that failed.

I have the A2669, now i'm just afraid to use it.

Pukesmiley, 2025-03-17:
> > > Your pinecil still working afterwards ? I mean it's only the charger that breaks ?
> > 
> > 
> > Yeah, the pinecil still works perfectly, it's just the charger that failed.
> 
> I have the A2669, now i'm just afraid to use it.

I wouldnt use it, but if it breaks after using it, reach out to Anker, my experience with them was VERY nice, after opening the ticket, they were supportive and send me two replacements (i broke the first replacement with the pinecil)

ali1234, 2025-10-23:
I have two Anker Nano II 100W (B2145) chargers that failed after three months of just charging tablets and phones (nothing over 20W). Now they shut off if you try to draw more than 0.5W from them. Even from the USB A port. It seems Anker forgot how to do quality control. I very much doubt the Pinecil is to blame here.