Ralim/IronOS #2071: Unused objects and the expressions without side effects

safocl, 2025-02-04:
https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L40C26-L40C47
`buffer_remaining_size` is not used anywhere except in a non-significant expression:
https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L47

and the following expression without side effects:
https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L48

resistancelion, 2025-02-06:
> https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L40C26-L40C47 `buffer_remaining_size` is not used anywhere except in a non-significant expression:
> 
> [IronOS/source/Core/LangSupport/lang_multi.cpp](https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L47)
> 
> Line 47 in [4ce63fa](/Ralim/IronOS/commit/4ce63fab47fae2b2f76eab366edd4d449053188f)
> 
>  buffer_remaining_size -= outsize; 
> and the following expression without side effects:
> 
> [IronOS/source/Core/LangSupport/lang_multi.cpp](https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/LangSupport/lang_multi.cpp#L48)
> 
> Line 48 in [4ce63fa](/Ralim/IronOS/commit/4ce63fab47fae2b2f76eab366edd4d449053188f)
> 
>  buffer_next_ptr += outsize;

Debugging artifacts, clearly

safocl, 2025-02-07:
> Debugging artifacts, clearly

ok.

closed 2025-02-07 by safocl