Ralim/IronOS #2072: Potentially undefined behavior when accessing objects out of alignment and beyond their lifetime

safocl, 2025-02-04:
https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/Drivers/BootLogo.cpp#L21
https://github.com/Ralim/IronOS/blob/4ce63fab47fae2b2f76eab366edd4d449053188f/source/Core/BSP/Pinecilv2/BSP.cpp#L283
```cpp
  uint8_t scratch[1024];
  flash_read(FLASH_LOGOADDR - 0x23000000, scratch, 1024);

  BootLogo::handleShowingLogo(scratch);
```
and similar other code may produce undefined behavior:
`*(reinterpret_cast<const uint32_t *>(ptrLogoArea))` from `uint8_t scratch[1024];` -- `scratch` does not contain `uint32_t` objects -- their lifetime has not yet begun (the lifetime of `uint8_t` elements has begun).
And potentially unaligned access is possible here.

[[basic.compound#4]](https://eel.is/c++draft/basic.compound#4)

> A pointer value P is [valid in the context of](https://eel.is/c++draft/basic.compound#def:valid_in_the_context_of) an evaluation E if P is a pointer to function or a null pointer value, or if it is a pointer to or past the end of an object O and E happens before the end of the duration of the region of storage for O[.](https://eel.is/c++draft/basic.compound#4.sentence-1) If a pointer value P is used in an evaluation E and P is not valid in the context of E, then the behavior is undefined if E is an indirection ([[expr.unary.op]](https://eel.is/c++draft/expr.unary.op)) or an invocation of a deallocation function ([[basic.stc.dynamic.deallocation]](https://eel.is/c++draft/basic.stc.dynamic.deallocation)), and implementation-defined otherwise[.](https://eel.is/c++draft/basic.compound#4.sentence-2)

[[basic.compound#5]](https://eel.is/c++draft/basic.compound#5)
> Two objects a and b are [pointer-interconvertible](https://eel.is/c++draft/basic.compound#def:pointer-interconvertible) if
> -- they are the same object, or
> -- one is a union object and the other is a non-static data member of that object ([[class.union]](https://eel.is/c++draft/class.union)), or
> -- one is a standard-layout class object and the other is the first non-static data member of that object or any base class subobject of that object ([[class.mem]](https://eel.is/c++draft/class.mem)), or
> -- there exists an object c such that a and c are pointer-interconvertible, and c and b are pointer-interconvertible[.](https://eel.is/c++draft/basic.compound#5.sentence-1)

[[basic.memobj#intro.object-1]](https://eel.is/c++draft/basic.memobj#intro.object-1)

> The constructs in a C++ program create, destroy, refer to, access, and manipulate objects[.](https://eel.is/c++draft/basic.memobj#intro.object-1.sentence-1) An [object](https://eel.is/c++draft/basic.memobj#def:object) is created by a [definition](https://eel.is/c++draft/basic.def), by a [new-expression](https://eel.is/c++draft/expr.new#nt:new-expression) ([[expr.new]](https://eel.is/c++draft/expr.new)), by an operation that implicitly creates objects (see below)

_[[basic.memobj#intro.object-example-1]](https://eel.is/c++draft/basic.memobj#intro.object-example-1)_

[[basic.memobj#basic.life-2]](https://eel.is/c++draft/basic.memobj#basic.life-2)

> [2](https://eel.is/c++draft/basic.memobj#basic.life-2) The [lifetime](https://eel.is/c++draft/basic.memobj#def:lifetime) of an object or reference is a runtime property of the object or reference[.](https://eel.is/c++draft/basic.memobj#basic.life-2.sentence-1) A variable is said to have [vacuous initialization](https://eel.is/c++draft/basic.memobj#def:initialization,vacuous) if it is default-initialized, no other initialization is performed, and, if it is of class type or a (possibly multidimensional) array thereof, a trivial constructor of that class type is selected for the default-initialization[.](https://eel.is/c++draft/basic.memobj#basic.life-2.sentence-2) The lifetime of an object of type T begins when:
> [(2.1)](https://eel.is/c++draft/basic.memobj#basic.life-2.1) storage with the proper alignment and size for type T is obtained, and
> [(2.2)](https://eel.is/c++draft/basic.memobj#basic.life-2.2) its initialization (if any) is complete (including vacuous initialization) ([[dcl.init]](https://eel.is/c++draft/dcl.init)),

[[expr.unary#op-1]](https://eel.is/c++draft/expr.unary#op-1)

> [1](https://eel.is/c++draft/expr.unary#op-1) The unary * operator performs [indirection](https://eel.is/c++draft/expr.unary#def:indirection)[.](https://eel.is/c++draft/expr.unary#op-1.sentence-1) Its operand shall be a prvalue of type “pointer to T”, where T is an object or function type[.](https://eel.is/c++draft/expr.unary#op-1.sentence-2) The operator yields an lvalue of type T[.](https://eel.is/c++draft/expr.unary#op-1.sentence-3) If the operand points to an object or function, the result denotes that object or function; otherwise, the behavior is undefined except as specified in [[expr.typeid]](https://eel.is/c++draft/expr.typeid)[.](https://eel.is/c++draft/expr.unary#op-1.sentence-4)
> [Note [1](https://eel.is/c++draft/expr.unary#op-note-1): Indirection through a pointer to an incomplete type (other than cv void) is valid[.](https://eel.is/c++draft/expr.unary#op-1.sentence-5) The lvalue thus obtained can be used in limited ways (to initialize a reference, for example); this lvalue must not be converted to a prvalue, see [[conv.lval]](https://eel.is/c++draft/conv.lval)[.](https://eel.is/c++draft/expr.unary#op-1.sentence-6) — end note]

https://learn.microsoft.com/en-us/cpp/cpp/indirection-operator-star?view=msvc-170#remarks
https://en.cppreference.com/w/cpp/language/reinterpret_cast#Type_accessibility
https://cplusplus.github.io/CWG/issues/658.html

safocl, 2025-02-07:
```cpp
void BootLogo::handleShowingLogo(const uint8_t *ptrLogoArea) {
  OLED::clearScreen();
  // Read the first few bytes and figure out what format we are looking at
  if (OLD_LOGO_HEADER_VALUE == *(reinterpret_cast<const uint32_t *>(ptrLogoArea))) {
    showOldFormat(ptrLogoArea);
  } else if (ptrLogoArea[0] == 0xAA) {
    showNewFormat(ptrLogoArea + 1);
  }

  OLED::clearScreen();
}
```
can replace with:
```cpp
void BootLogo::handleShowingLogo(const uint8_t *ptrLogoArea) {
  OLED::clearScreen();
  // Read the first few bytes and figure out what format we are looking at
  uint32_t header_v;
  std::memcpy( &header_v, ptrLogoArea, sizeof(header_v) );
  if (OLD_LOGO_HEADER_VALUE == header_v) {
    showOldFormat(ptrLogoArea);
  } else if (ptrLogoArea[0] == 0xAA) {
    showNewFormat(ptrLogoArea + 1);
  }

  OLED::clearScreen();
}
```
***but ENDIANNESS is not checket here.***

safocl, 2025-02-07:
> _**but ENDIANNESS is not checket here.**_ _(maybe need byteswap for this value)_

but if `OLD_LOGO_HEADER_VALUE` is created as a `constexpr std::array` (or _C-style array_), then these tricks are not needed.

Ralim, 2025-02-07:
@safocl Please also understand that _ALL_ IronOS targets are Little-Endian. Endian is assumed to always be little-endian project wide unless denoted otherwise.

safocl, 2025-02-07:
> [@safocl](https://github.com/safocl) Please also understand that _ALL_ IronOS targets are Little-Endian. Endian is assumed to always be little-endian project wide unless denoted otherwise.

but `flash_read` gives the result in RAW-binary format?

Ralim, 2025-02-07:
`flash_read` is basically just a memcpy. So it will be in native (LE) endianess.

safocl, 2025-02-07:
> As far as I know, all targets also allow un-aligned reads without any issue. Though logoArea shall always be aligned anyway as it has to be flash aligned.

but alignment is not the only difficulty there -- uint32_t TYPE has not started its lifetime...

safocl, 2025-02-07:
> `flash_read` is basically just a memcpy. So it will be in native (LE) endianess.

byte order for registers only -- memory is RAW
CPU read to register from memory in itself byte-ordering (endianness) -- either direct or reverse.

Ralim, 2025-02-07:
>uint32_t TYPE has not started its lifetime

I guess this is the limit of my understanding; to me the generated assembly appears to do what I expect, where we load in the raw encoded bytes from the start of the uint8_t buffer into the register using the native load instruction. Then we interpret that as a constant we compare to the check value? Like, in this case uint32_t or not doesn't realllly matter, as we are just comparing to know constant values?

safocl, 2025-02-07:
> I guess this is the limit of my understanding; to me the generated assembly appears to do what I expect, where we load in the raw encoded bytes from the start of the uint8_t buffer into the register [...]

https://godbolt.org/z/dqMb1GGhK

Ralim, 2025-02-07:
All your example shows to me is it failing on x86 as its not LE which is expected behaviour?

safocl, 2025-02-07:
> All your example shows to me is it failing on x86 as its not LE which is expected behaviour?

this is because when perceiving memory as uint32, it is put into the register with the reverse byte order (LE) - it is necessary to compare in the reverse order - I provided an example with this option

Ralim, 2025-02-07:
Yeah, reading it as LE is _expected_ and by design. This code WILL NOT WORK on BE systems and that is by design.

Any support for BE is not a concern to this code at all; BE systems are out-of-scope.

So far your example has shown what is already known? That it only works on the systems its designed for?

safocl, 2025-02-07:
https://godbolt.org/z/WMnvPdf69

safocl, 2025-02-07:
> > All your example shows to me is it failing on x86 as its not LE which is expected behaviour?
> 
> this is because when perceiving memory as uint32, it is put into the register with the reverse byte order (LE) - it is necessary to compare in the reverse order - I provided an example with this option

this only concerns the byte order, but other aspects of the problem remain -- regarding the byte order, there was an additional clarification.

safocl, 2025-02-07:
> this code is valid

the simple answer that is possible UB:

[[basic.memobj#basic.life-2]](https://eel.is/c++draft/basic.memobj#basic.life-2)

> [2](https://eel.is/c++draft/basic.memobj#basic.life-2) The [lifetime](https://eel.is/c++draft/basic.memobj#def:lifetime) of an object or reference is a runtime property of the object or reference[.](https://eel.is/c++draft/basic.memobj#basic.life-2.sentence-1) [...]
> The lifetime of an object of type T begins when:
> [(2.1)](https://eel.is/c++draft/basic.memobj#basic.life-2.1) storage with the proper alignment and size for type T is obtained, and
> [(2.2)](https://eel.is/c++draft/basic.memobj#basic.life-2.2) its initialization (if any) is complete (including vacuous initialization) ([[dcl.init]](https://eel.is/c++draft/dcl.init)),

https://en.cppreference.com/w/cpp/memory/start_lifetime_as#Example

Ralim, 2025-02-07:
Originally this code used the C-Style casts to cause the reinterpretation without allocation of any r-values; I believed that when I updated this to use the C++17 style casting, that `reinterpret_cast` would also do the same, where there is no temporary value created; and thus there is no lifetime to be concerned over?

My confusion is that I'm not sure what lifetime we are talking about here; that would not-be-valid. The storage is already allocated and sized; and its initialised at the time of use?

To move forward; what would you propose as the _correct_ solution to the end goal of re-interpreting the given memory bytes as a uint32_t without the invocation of extra functions at runtime?

safocl, 2025-02-07:
> Originally this code used the C-Style casts to cause the reinterpretation without allocation of any r-values; I believed that when I updated this to use the C++17 style casting, that [...]

C-style cast is:

> [4](https://eel.is/c++draft/expr.cast#4) The conversions performed by
> [(4.1)](https://eel.is/c++draft/expr.cast#4.1) a const_cast ([[expr.const.cast]](https://eel.is/c++draft/expr.const.cast)),
> [(4.2)](https://eel.is/c++draft/expr.cast#4.2) a static_cast ([[expr.static.cast]](https://eel.is/c++draft/expr.static.cast)),
> [(4.3)](https://eel.is/c++draft/expr.cast#4.3) a static_cast followed by a const_cast,
> [(4.4)](https://eel.is/c++draft/expr.cast#4.4) a reinterpret_cast ([[expr.reinterpret.cast]](https://eel.is/c++draft/expr.reinterpret.cast)), or
> [(4.5)](https://eel.is/c++draft/expr.cast#4.5) a reinterpret_cast followed by a const_cast,

safocl, 2025-02-07:
> To move forward; what would you propose as the _correct_ solution to the end goal of re-interpreting the given memory bytes as a uint32_t without the invocation of extra functions at runtime?

here it all depends on what the flash_read function does -- if it implicitly creates an object that can be represented as a `uint32` -- then everything will be fine. However, since the buffer passed to the function itself is not aligned according to the `uint32` alignment requirements -- then it is still undefined behavior when dereferencing a pointer as a `uint32*`

safocl, 2025-02-07:
o -- I think perhaps the last example will be very clear for our discussion:
https://en.cppreference.com/w/cpp/utility/launder#Example
```cpp
// Case 2: access to a new object whose storage is provided
    // by a byte array through a pointer to the array.
    struct Y { int z; };
    alignas(Y) std::byte s[sizeof(Y)];
    Y* q = new(&s) Y{2};
    const int f = reinterpret_cast<Y*>(&s)->z; // Class member access is undefined
                                               // behavior: reinterpret_cast<Y*>(&s)
                                               // has value "pointer to s" and does
                                               // not point to a Y object
    const int g = q->z; // OK
    const int h = std::launder(reinterpret_cast<Y*>(&s))->z; // OK
```

safocl, 2025-02-07:
https://en.cppreference.com/w/cpp/numeric/bit_cast also maybe use as source of information for this discussion

Ralim, 2025-02-07:
Note: `bit_cast` isn't available in cpp17

We can change the buffer decl to be a `uint32_t buffer[256]` potentially to ensure its aligned as you want it?

Its going to be loaded either byte by byte or in uint32_t chunks, thats not in control of the reader function impl. The loading method shouldn't matter though, as its only filling in the on-stack array (it doesnt make anything otherwise)

safocl, 2025-02-07:
> We can change the buffer decl to be a `uint32_t buffer[256]` potentially to ensure its aligned as you want it?
>
> Its going to be loaded either byte by byte or in uint32_t chunks, thats not in control [...]

or 
`alignas(uint32_t) uint8_t scratch[1024];`

safocl, 2025-02-07:
I think I found an example straight from the C++ standard that shows what I meant -- in what situation everything will be fine:

[[intro.object#example-3]](https://eel.is/c++draft/intro.object#example-3)
> [Example [3](https://eel.is/c++draft/intro.object#example-3): 
> #include <cstdlib>
> struct X { int a, b; };
> X *make_x() {
>   // The call to std​::​malloc implicitly creates an object of type X
>   // and its subobjects a and b, and returns a pointer to that X object
>   // (or an object that is pointer-interconvertible ([[basic.compound]](https://eel.is/c++draft/basic.compound)) with it),
>   // in order to give the subsequent class member access operations
>   // defined behavior.
>   X *p = (X*)std::malloc(sizeof(struct X));
>   p->a = 1;
>   p->b = 2;
>   return p;
> }
> — end example]

here lifetime of `struct X` starts _[(2.2)](https://eel.is/c++draft/basic.life#2.2) its initialization (if any) is complete (including vacuous initialization) ([[dcl.init]](https://eel.is/c++draft/dcl.init))_

Ralim, 2025-02-27:
The alignas has been added PR.

>I think I found an example straight from the C++ standard that shows what I meant 

I think one of my gaps is that for a lot of these things I expected them to only really matter for things like structs, rather then raw primitive arrays.

closed 2025-02-27 by Ralim