Ralim/IronOS #2076: Scroll animation

Siutas, 2025-02-14:
Scroll animation should be removed if both idle and operating detailed screens are used

Ralim, 2025-02-15:
Hello,

Can you please provide more detail on this request, do you not want the settings scroll animation or the sliding transition between screens?

ia, 2025-02-15:
> Scroll animation should be removed if both idle and operating detailed screens are used

Hello, @Siutas.

I'm happily do use _IronOS_ with _Soldering detailed screen_ & _Idle detailed screen_ options enabled always all the time, and I do not have any issues with scroll animation at all.

So, could you, please, describe what is _**the issue**_ exactly? Thanks.

discip, 2025-02-25:
@Siutas
I see your point.
If I'm not mistaken, that's exactly what I initially wasn't fond of either.
Eventually I got used to it, but now that you mentioned it, my memories have been revived.
I also liked activating or stopping the heating without the sliding transition more, while both the **`idle-`** and the **`soldering screen`** were set to detailed.

Ralim, 2025-02-28:
I think this is the transitions rather than the scroll animation (in settings).
We could add a settings option to just disable these fairly easily.

ia, 2025-03-04:
Dear friends, _it seems_ I got a working prototype. I just will have to add a _setting_ routine and double check everything, but _**"I take this case"**_. ;)

@discip, out of curiosity:

> If I'm not mistaken, that's exactly what I initially wasn't fond of either.
> Eventually I got used to it, but now that you mentioned it, my memories have been revived.
> I also liked activating or [...]

But what's wrong with the slide transition animation for you? Just asking, because although it really looks pretty cool and fancy, I think I tend to agree that _in some cases_ it's a bit more distracting from the process of getting information on the display, than entertaining.

And what's your _"rationale"_, so to speak? Just would like to hear. :)

ia, 2025-03-05:
@discip, while you seem interested in this option as well, could you, please, try and test something from [these builds](https://github.com/ia/IronOSf/actions/runs/13666961111#artifacts).

I added a new checkbox option _Anim.sliding_ in _User interface_ submenu. I tested it briefly. It seems like it's doing exactly what you & @Siutas were talking about. Except... _**it works too good**_: even smooth scrolling animation inside settings menu (between menu items) is disabled, once this option is disabled as well.

Now, could you tell me, please, how do you think - is this expected behavior (I mean, we disable slide-side-scroll animation, right?)? Or should be there some exceptions, and if yes, then for which situations/modes?

Just let me know, what you think, because I can't wait to hear any feedback.

Ralim, 2025-03-05:
We could expand it to a Transition setting of "All, Only Vertical, Only Horizonal, None" ? to give users the choice

discip, 2025-03-05:
I'm currently unable to spend time on this (work), but will let you know as soon as possible.

ia, 2025-03-05:
> We could expand it to a Transition setting of "All, Only Vertical, Only Horizontal, None" ? to give users the choice

I just love _boolean_-like checkbox options since they are sooo easy to implement, maintain & support. But if we have to do multiple choice here, then I think that something like _"fully off, on modes change only, in settings only, fully on"_ sounds more reasonable than a specific effect, wouldn't you agree?

What is the exact reason to disable a particular animation effect regardless the mode, while some users may would like to have animation scrolling in settings, but not between modes and vice versa. I can say it from my perspective: I tested my patch, and I must admit, that without full animation the response from the iron subjectively (for obvious reasons) looks way much more faster, but not having animation on scrolling settings gives a subjective feeling like _something is wrong_.

@discip, I don't rush you at all, so only when you have time, but, please, try to test my builds first, and then let me know what you think and how do you think we should proceed, since you're interested in this option. Thanks.

ia, 2025-03-05:
@Siutas, are you still interested in this option? Just asking.

closed 2025-03-09 by ia