Ralim/IronOS #2078: TS101 boot logo orientation does not respect display orientation

FPSUsername, 2025-02-15:
**Describe the bug**

The original TS101 firmware allows the display orientation to be left or right. When adjusting the orientation to left (as right is the standard), the boot logo will also be mirrored.

I noticed that the miniware startup logo remains in the right orientation even when the display orientation is set to left.

**To Reproduce**

5. 
Go to the user interface settings, set the display orientation to left.
Power down the device (unplug)
Plug in the device and you'll see the boot logo in the right rotation instead of left.

**Expected behavior**

The boot logo should take over the left orientation.

**Details of your device:**
 
 - Device: MiniWare TS101
 - Release: [V2.23 RC1](https://github.com/Ralim/IronOS/releases/tag/v2.23-rc1)
 - Power adapter being used: Computer USB port
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2025-02-15:
The miniware startup logo is outside the control of this firmware. 
They show that as part of their bootloader. IronOS doesn't get control until after this logo is shown.

FPSUsername, 2025-02-15:
I didn't know that. Then I think there's no default ironos boot logo. Is that normal?

Ralim, 2025-02-15:
Correct, boot logo is opt in

closed 2025-02-15 by FPSUsername

ia, 2025-02-15:
Hello, @FPSUsername.

> Then I think there's no default ironos boot logo.

Yes, there is no, because there is no such thing as _"default ironos boot logo"_.

> Is that normal?

Yes, that's normal, because _custom logo_ (which a user would like to set with _IronOS firmware_) and _IronOS firmware_ are technically two different binary files which are flashed separately.