Ralim/IronOS #2089: Black screen after installing v2.23-rc2 on Sequre S60P and T55

StevenSeifried, 2025-02-28:
**Describe the bug**
After installing v2.23-rc2 onto Sequre S60P and T55 I get a black screen. Heating _seems_ also not working, due to nothing happens after a short press on the A button.

Flashing v2.23-rc1 brings them "back to life" (sorry, I didn't find a better wording)

**To Reproduce**
1. Put S60P and/or T55 into UPDATE mode by pressing the A button while insert the USB-C cable
2. Copy the `S60P_DE.hex` or `S60P_EN.hex`/`T55_DE.hex` or `T55_EN.hex` of rc2 into the UPDATE drive
    - I just only tested them 
3. When "[READY]" get changed to "[YES]" reinsert the USB-C cable
4. See error --> black screen

**Expected behavior**
"Certain settings changed!" message

**Details of your device:**
 - Device: Sequre S60P and T55
 - Release: v2.23-rc2, I assume v2.23E.7EC1ECE1 (that's the version on my Pinecil v2)

**Additional context**
My first assumption was, that it's caused by something which has to do with "Full OLED resolution support".
But there are no changes mentioned for "Full OLED resolution support" in the RC2 changelog.

Unfortunately I can't test on the TS101 because I can no longer install firmware on it. Not even the original, but that's not a topic for here.

v2.23-rc2 works fine on my Pinecil v2

Ralim, 2025-03-01:
Hia, can you please test with assets from this build https://github.com/Ralim/IronOS/actions/runs/13603097007

Working for me on S60P but want cross-check

StevenSeifried, 2025-03-01:
Hi,
tested and screen is working on both.

PR #2091 (2025-03-01): Fix RToS Update

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Closes #2090
Closes #2089

Fixes task max priorities due to update to rtos related code earlier.

* **What is the current behavior?**

* **Other information**:

Tested on
- [x] TS80
- [x] MHP30 
- [x] TS80P
- [x] S60P
- [x] PinecilV2
closed 2025-03-01 by Ralim