Ralim/IronOS #2090: Bootloop with TS101 and v2.23-rc2

Achduka, 2025-02-28:
After installing v2.23-rc2 on my TS101 I've stuck in a bootloop (miniware logo) on different power devices. Installing 2.22 solved this issue.

FPSUsername, 2025-02-28:
Can confirm using the TS101_EN.hex file.
rc1 works fine.

Ralim, 2025-03-01:
Hia, can you please test with assets from this build https://github.com/Ralim/IronOS/actions/runs/13603097007

I do not have a TS101 to hand to test, but working on my other miniware devices.

indeyets, 2025-03-01:
@Ralim works for me on ts80p.

Achduka, 2025-03-01:
Unfortunately same issue but different behaviour. With the rc2 the bootloop was really fast, with the test build it's starting into the menu (heat or settings) and then the logo appears again.

Ralim, 2025-03-01:
>with the test build it's starting into the menu (heat or settings) and then the logo appears again.

Likely this is a different issue (power source?)

closed 2025-03-01 by Ralim

Achduka, 2025-03-02:
I tested now with rc3 and different power sources. I could find out the issue only exists with apple hardware. Bootloop with Macbook Pro M1 67W Charger (worked with 2.22) but no bootloop with Dell Charger or Makita Battery.

Ralim, 2025-03-04:
Eugh Apple regressions are the worst 😢 For 2.23 I added in the missing feature for the other device to be able to ask the pinecil what it can support. This fixed compatability with some newer laptops so unsure what broke with the apple charger.
Can you comment on the charger model and info? I guess I'll need to see if I can find one to borrow :D

Achduka, 2025-03-04:
Ah and maybe it's important to know, it's from Germany.

Ralim, 2025-03-04:
Hmm, I wonder if its anything weird to do with the 20.3V not being 20V.

Could you check if this issue occurs on all the options for the PD mode?

FPSUsername, 2025-03-09:
I can confirm that RC3 solved the bootloop issue

PR #2091 (2025-03-01): Fix RToS Update

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Closes #2090
Closes #2089

Fixes task max priorities due to update to rtos related code earlier.

* **What is the current behavior?**

* **Other information**:

Tested on
- [x] TS80
- [x] MHP30 
- [x] TS80P
- [x] S60P
- [x] PinecilV2