Ralim/IronOS #2101: Extend range of values for display brightness setting

ia, 2025-03-06:
**Describe the bug**
Range of values for display brightness setting may be too short for some situations.

**To Reproduce**
Open OLED brightness in settings: it has values from 0/1 to 5 (from 0/1 to 100 with step 25 _under the hood_).

**Expected behavior**
Display brightness may have more wide range of values, i.e., from 1 to 10 (from 0/1 to 100 with step 10).

**Details of your device:**
Applicable to any supported device.

**Additional context**
I would like to fill this bug report to track further possible work & to track the feedback from other users. Considering how much _IronOS_ users like to tweak their irons and customize every possible setting and button, I'm very surprised that no one didn't ask for this before. I've made a couple of test builds already, _hacking around_ `MIN` / `MAX` / `STEP` brightness in `configuration.h`, and considering that users may have to work in different light conditions, I start to think that brightness should have a bit more wider range of values. It seems that from 1 to 10 with 10 step looks pretty reasonable, but at the same time it gives pretty distinguishable light for every step.

@Ralim & @discip, what do you think? Would you agree to accept this as new global defaults?

discip, 2025-03-06:
@ia
This is funny — some time ago, there was a request to reduce the increments from **10** to **5** (see https://github.com/Ralim/IronOS/pull/1622). 🤣
So that's what I ended up with, and every one seemed to be happy.

Another helpful reference in this regard: https://github.com/Ralim/IronOS/pull/1087

discip, 2025-03-07:
Don't get me wrong—this isn't set in stone! 😊
If people need finer increments, then that's the way to go.
I personally use all my devices on the lowest setting anyway.
Maybe @Ralim has a definitive opinion on this?

Ralim, 2025-03-10:
Err I dont have strong feelings.
@ia I'm curious why you use the extra steps, how often are you adjusting this? 🤣