Ralim/IronOS #2109: HP GaN 65W charger incompatible with Pinecil v2

Fironet, 2025-03-21:
Originally submitted this as a [discussion](https://github.com/Ralim/IronOS/discussions/2105) but unsure how frequently that section gets seen and was hoping for any insight into whether this is an IronOS issue or charger issue (for refund purposes).

I bought a HP GaN 65W USB-C PD charger ([datasheet](https://www8.hp.com/h20195/v2/GetDocument.aspx?docname=4aa8-1882enw)) that supports the following:

- Supports fast charging protocols: PD/PPS/QC5.0/AFC
- 20V/3.25A max for single port output.
- Has grounded power socket and cable (the key reason I bought it so the iron tip can be properly grounded without need for separate grounding via the lug, from what I've read)

When the Pinecil v2 (bought from pine64.com) is plugged into it, with or without the short iron tip attached, via either the official Pine64 1.5m USB-C cable or a separate USB 3.x cable the iron doesn't display anything on the screen.

However it works for say charging phones. While if attached to a separate Anker 65W charger (45W max single port output) the Pinecil displays and works.

---

Things I've tried without any change:

- Attaching the cable to the Pinecil while the '-' button is held.
- Attaching the cable while the '+' is held.
- Both of the above with / without the tip attached.
- Upgrading the firmware to the latest stable release listed via pineflash (2.23 (2025-03-01)).
- Checking that Bluetooth is disabled in the settings (read a user saying BT caused issues with some of their chargers)

I believe the official Pine64 cable is USB 2.0 but can't confirm. Read Ralim saying the Pinecil exposes extra data over USB 3.0 pins that may conflict with some chargers and that USB 2.0 USB-C-to-C cables may resolve the issue but don't have any other known USB 2.0 USB-C-to-C cables (my others are all known USB 3.x or USB-A to USB-C).

---

Any idea if the issue is on the IronOS side or the charger side? As on paper it should be compatible.

Ralim, 2025-03-24:
The discussions tab is about as monitored as issues. Though this can be opened in either location.

When booting holding (-) the screen **is meant to stay black** so you will need to use a usb power monitor or a cable with an inbuilt led indication to see if the power supply is providing 5V. This test is done as it causes no code to run on the processor, so no PD negotiation/init is done. And the FUSB that controls the PD protocol stays in its default mode of providing the pull-down to turn on a PD compliant supply.

If the supply does not provide 5V with the PD spec current sink's in the pinecil (the ones built into the FUSB), then its going to be an issue with the adaptor not playing well with the PD spec.

I should probably write up a bit of a debugging guide 😓 

Can you confirm if you boot holding (+) on the working charger you correctly can see the PD information about the charger? (To confirm PD hardware is working in the Pinecil)

Fironet, 2025-03-24:
> When booting holding (-) the screen is meant to stay black so you will need to use a usb power monitor or a cable with an inbuilt led indication to see if the power supply is providing 5V

Oh, I see. I thought it was related to the documented debug menu, which had me confused. Yeah unfortunately I lack one of those intermediate USB power readers.

> Can you confirm if you boot holding (+) on the working charger you correctly can see the PD information about the charger? (To confirm PD hardware is working in the Pinecil)

So on the Anker charger the Pinecil states `PD Debug State 3 No Vbus`.

In the meantime I bought a fixed negotiation 20V/3A USB-C (female) to DC5525 (male) adapter. Should arrive sometime in the next week. Wondering if that will provide a workaround and still keep the tip earth grounded.

Fironet, 2025-03-29:
@Ralim so good news about the 20V adapter: it works with the HP charger in making the display appear and shows it's 20V (I realize now that's what the vertical numbers are on the Pinecil's home screen, since with the Anker adapter it only shows '15').

I also measured the Pinecil's short tip resistance to the mains ground socket, with the HP charger + 20V adapter, and its 0 (ie: grounded), while when using the Anker that lacks the grounding prong it measures 30 MΩ.

Haven't yet tested soldering with it yet, since just wanted to check the adapter idea worked first.

---

Does this suggest that it's something IronOS could resolve with USB-C PD negotiation, given the 20V USB-C adapter is seemingly negotiating with the charger in a way that works?

Ralim, 2025-03-31:
>Does this suggest that it's something IronOS could resolve with USB-C PD negotiation, given the 20V USB-C adapter is seemingly negotiating with the charger in a way that works?

Yes, but I would need PD packet captures to know what is breaking. This is the main issue with PD at the moment is that its not trivial to debug.

Fironet, 2025-03-31:
> Yes, but I would need PD packet captures to know what is breaking. This is the main issue with PD at the moment is that its not trivial to debug.

Is there a list of PD sniffers that are recommended? While I'm not currently planning on getting one as yet I did look up some and there's a DIY one that uses Wireshark and some others that use proprietary software. Does the style of output matter? Also some are USB 2.0 only and others support USB 3.0.

Ralim, 2025-04-02:
Nothing really _reccomended_. Whatever is the easiest to get / use, so long as we can get either a capture file or even a list of messages sent (with their contents) will work.