Ralim/IronOS #2115: ts101 gets stuck in boost mode when lock is on

thatonehhdev, 2025-04-01:
**Describe the bug**
with the lock mode set to B, when soldering with lock on, the boost mode stays on even after i release the button.

**To Reproduce**

thatonehhdev, 2025-04-01:
**Describe the bug**
with the lock mode set to B, when soldering with lock on, the boost mode stays on even after i release the button.

**To Reproduce**

dinwiggy, 2025-04-28:
I'm seeing the same on my Pinecil V2. Issue is not present in 2.22. Issue is present in 2.23rc1, 2.23rc3, and 2.23rc4.

dinwiggy, 2025-04-28:
It looks like this was lost in the refactor:

https://github.com/Ralim/IronOS/compare/v2.22...v2.23-rc4#diff-8c50a175baa140c41981bf7da2a97e85d5ac6b25a88b9ee3c2b89fe055ae7e16L40

https://github.com/Ralim/IronOS/compare/v2.22...v2.23-rc4#diff-9e19aca0dc4c0156c58464c504e2c526c6ea2e85155efd23c5876fd2f456c36fR35

Adding that logic back in fixed it for me:

[image]

This is the only code I've ever looked at in this repo (got my Pinecil last week and noticed this behavior while trying out settings), so I don't feel I'm familiar enough to make a PR at the moment. Hopefully this will help someone with an implementation 😃.

Ralim, 2025-07-19:
This should be fixed in #2143

commit 1056f0cf (2025-07-19): Fix #2115

Refactor lost line
PR #2143 (2025-07-23): Fix 2115 2131

* **Please check if the PR fulfils these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Bug fixes

* **What is the current behaviour?**

Fixes #2115 
Fixes #2131 

* **What is the new behaviour (if this is a feature change)?**

* **Other information**:
commit b4bb6d54 (2025-07-23): Fix 2115 2131 (#2143)

* Mark Pinecilv1 NRND

Closes #2131

* Fix #2115

Refactor lost line
closed 2025-07-23 by Ralim