Ralim/IronOS #2116: Better CJK font rendering

OctopusET, 2025-04-02:
Currently CJK font has no Korean font, so we need to add extra Korean font to make it work.
Rather than that I suggest other way.

For latin language it's 6x8 pixel size.

1. Use 8x8 font for default CJK font. Smaller, but we can write longer messages.
 - Japanese: https://littlelimit.net/font.htm#license
 - Korean: https://github.com/RanolP/dalmoori-font/releases

Since it's using free software license, we can merge them into one file.

2. Use bigger 12x12 or 16x16 font when the text is short enough

I will try to add PoC only for Korean language.

Reference:
 - https://chinese.stackexchange.com/questions/16669/lowest-pixel-resolution-needed-to-support-chinese

Related issues #1397 #791 #847

OctopusET, 2025-04-02:
I think some fonts support 6x8 for CJK font, but since most of CJK characters are square shape. It would be better to stick with nxn size.

OctopusET, 2025-04-02:
https://diaowinner.itch.io/galmuri-extended
This would be one of the best font for our case, but it's OFL1.1. Seems like it's not compatible with this project.

OctopusET, 2025-04-03:
Please share more fonts if you find more.

CoelacanthusHex, 2025-05-10:
Unifont has suitable license but unfortunately it use 16x16 size for CJK.
To be honest, use 8x8 point for Chinese characters is a terrible experience...