Ralim/IronOS #2122: Miniware TS21 iron support

CoelacanthusHex, 2025-05-10:
New profuct of Miniware, similar to S60(P), it uses C210 tips and with 128x32 OLED screen. But it has high power (100W instead of 60W)
User manual: https://forum.minidso.com/forumdownload/TS21%20Precision%20Soldering%20Iron%20User%20Manual.zip
Firmware: https://forum.minidso.com/forumdownload/TS21APPV1.5.0.zip

Ralim, 2025-07-19:
When schematics are open, I'm happy to look at support. Until such a time I dont have a good reason to add support. I will happily help review/merge/support PR's to add support if someone else wants to lead this effort.

iontodirel, 2026-10-13:
I am playing around with TS21 here: https://github.com/iontodirel/IronOS/commits/ts21-bringup, nearly there.